micropython/micropython #4720: IP Address Assignment Issue on LAN

hetvishah08, 2019-04-25:
I am using ESP32 DevkitC with Waveshare LAN8720 board and Micropython version downloaded on 10th April v1.10. I am trying to connect LAN with ESP32 but it takes a lot of time to assign an IP Address to the board. It takes around 30 seconds to 1 minute to assign an IP to the board.

Here is the log:

rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:4928
ho 0 tail 12 room 4
load:0x40078000,len:9332
load:0x40080400,len:6216
entry 0x400806e8
I (433) cpu_start: Pro cpu up.
I (433) cpu_start: Application information:
I (433) cpu_start: Compile time:     00:51:09
I (435) cpu_start: Compile date:     Apr 10 2019
I (440) cpu_start: ESP-IDF:          v3.3-beta1-268-g5c88c5996
I (447) cpu_start: Starting app cpu, entry point is 0x40082950
I (0) cpu_start: App cpu up.
I (457) heap_init: Initializing. RAM available for dynamic allocation:
I (464) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (470) heap_init: At 3FFB9B90 len 00026470 (153 KiB): DRAM
I (477) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (483) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (489) heap_init: At 40093434 len 0000CBCC (50 KiB): IRAM
I (496) cpu_start: Pro cpu start user code
I (66) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
OSError: [Errno 2] ENOENT
MicroPython v1.10-272-g358364b45 on 2019-04-10; ESP32 module with ESP32
Type "help()" for more information.
>>> from machine import Pin
>>> import network
>>> l = network.LAN(mdc = Pin(23), mdio = Pin(18), power = Pin(17), phy_type =network.PHY_LAN8720, phy_addr=1)
I (22927) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
>>> l.active(1)
I (61287) emac: emac reset done
True
>>> I (61287) ethernet: start
I (65287) ethernet: LAN cable connected
I (96407) event: eth ip: 192.168.29.59, mask: 255.255.255.0, gw: 192.168.29.1
I (96407) ethernet: Got IP

I receive the last two lines after 30 seconds. Why is it taking so long to respond?
Earlier it used to assign IP in 5 seconds.
I am developing a real time application where i need IP address to be assigned within 3-5 seconds.

petrkr, 2019-04-25:
As you do not uses different clock source (mine PR from earlier). Can you try older versions of uPy?

`esp32-20190425-v1.10-293-g56f6ceba7.bin`
`esp32-20190125-v1.10.bin`
`esp32-20180511-v1.9.4.bin`

Also can not be this issue with router? As system and network administrator I know there are lot of devices which assign IP faster than others. Did not refactor why, just took it as fact.

Petr

dpgeorge, 2019-04-26:
If you need fast Ethernet connection it's best to use a static IP address (see `lan.ifconfig(...)`).

Otherwise, there can be lots of reasons why the DHCP takes a while.  Eg if the ETH is not correctly configured to use the correct half/full duplex mode then packets can easily be lost and so DHCP retries many times.  The best thing to do is analyse the network packets (eg via Wireshark) to see exactly what is going on.  And use tools to measure performance to see if there are any other issues, eg using https://pypi.org/project/uiperf3/

hetvishah08, 2019-04-26:
@petrkr I tried it with the older versions of firmware as well and I get the same issue.

hetvishah08, 2019-04-26:
@dpgeorge Thank you for your help.
I'll try using lan.ifconfig().
Also I'll look into analysing the signals and using tools to check the issues.

hetvishah08, 2019-04-26:
@dpgeorge I used the lan.ifconfig() and according to my router I assigned an Static IP address to my device but when I use ntptime function to receive time it doesn't respond to it.It gave me an "Index Error".Even after assigning the IP address the other nework functions don't work. 
When I obtained Dyanmic IP address with lan.active(), it worked after 40 seconds and provided me with a time tuple.
I don't get the issue even after providing an IP Address,the network functions don't work.

dpgeorge, 2019-04-30:
@hetvishah08 to make progress here you'll need to debug it yourself because it's most likely related to your particular set up (hardware, wiring, network connection, etc).  You can try building an example from ESP IDF that uses Ethernet to see if that works.

hetvishah08, 2019-05-03:
@dpgeorge  Okay.Thank you for your help.

closed 2019-05-03 by hetvishah08