micropython/micropython #4722: sdcard esp32 does not work TTGO T8 V1.8

newbie71, 2019-04-25:
Hi, 

i am using a TTGO T8 V1.8 devboard which has an onboard card reader. i installed the newest micropyhton Version on the board. 

The Board has on the bottom side following description:
SD
D2=12
D3=13
CMD=15
CLK=14
D0=2
D1=4

i used the following Code to use the SD-Card:

import machine, sdcard, os
from machine import Pin, SPI
spisd = machine.SPI(2, baudrate=10000000, polarity=0, phase=0, sck=machine.Pin(14), miso=machine.Pin(2), mosi=machine.Pin(15))
sd = sdcard.SDCard(spisd, machine.Pin(13))
vfs = os.VfsFat(sd)
os.mount(vfs, "/sd")

When the esp32 starts i get the following error:

WebREPL daemon started on ws://XXX.XXX.XXX.XXX:8266
Started webrepl in normal mode
I (8681) gpio: GPIO[15]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (8681) gpio: GPIO[2]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (8681) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (8961) gpio: GPIO[15]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pullown: 0| Intr:0
I (8961) gpio: GPIO[2]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (8961) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
Traceback (most recent call last):
  File "main.py", line 6, in <module>
OSError: [Errno 19] ENODEV
MicroPython v1.10 on 2019-01-25; ESP32 module with ESP32
Type "help()" for more information.

robert-hh, 2019-04-25:
Using the above connections, the following script works for me:
```
import machine, sdcard, os
sd = sdcard.SDCard(machine.SPI(1, sck=machine.Pin(14), mosi=machine.Pin(15),
                   miso=machine.Pin(2)), machine.Pin(13))
os.mount(sd, "/sd")
```
I have a different board and just wired a SD card module accordingly. 
Please note also, that questions like these are better posted in the forum at forum.microypytheon.org.

dpgeorge, 2019-04-26:
You may need to format the SD card first before it can be mounted.  Either on a PC or with `os.VfsFat.mkfs(sd)`.

Apart from that, SD cards can be unreliable with the SPI interface.  It's more robust to use SDIO/SDMMC, which is currently not supported on the ESP32.

newbie71, 2019-04-26:
Hello,
i solved the problem. I erased the flash and installed a pre-compiled micropython version from loboris.
then i had only to add the following lines to the main.py

import machine, network, utime, uos
uos.sdconfig(uos.SDMODE_4LINE, clk=14, mosi=15, miso=2, cs=13)
uos.mountsd(True)
uos.listdir()

after reboot of the TTGO T8 V1.8 i get now the following output:

Internal FS (SPIFFS): Mounted on partition 'internalfs' [size: 1048576; Flash ad                                                                                       dress: 0x230000]
----------------
Filesystem size: 956416 B
           Used: 1792 B
           Free: 954624 B
----------------
---------------------
 Mode:  SD (4bit)
     Name: SU08G
     Type: SDHC/SDXC
    Speed: high speed (50 MHz)
     Size: 7580 MB
      CSD: ver=1, sector_size=512, capacity=15523840 read_bl_len=9
      SCR: sd_spec=2, bus_width=5

Maybe it helps someone else

thx & kind regards

nickovs, 2019-06-05:
If you want to get back to the current, main-line Micropython instead of the LoBoris fork you should try recompiling from the current master branch. Since the merger of #4772 into these is support for the built-in SD card driver hardware, which is much faster than using an SPI interface and the Python `sdcard` module. On the TTGO T8 all you should need is:
```
uos.mount(machine.SDCard(), "/sd")
```

justinmoon, 2019-07-26:
@nickovs Any ideas how to get this working on an ESP32 m5stack? 

```
>>> uos.mount(machine.SDCard(), "/sd")
E (946554) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: 16
```

nickovs, 2019-07-27:
Cross-referencing the [answer](https://github.com/micropython/micropython/issues/4848#issuecomment-515275447) I gave over here.

dukeduck1984, 2019-08-11:
@nickovs Hi Nicko

I had an error on my TTGO T8 with the built-in sdcard slot
```
>>> sd = machine.SDCard()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: 259
```

Any idea?

Thanks

nickovs, 2019-08-11:
The `OSError: 259` is from the ESP IDF library and means "Invalid state". The most common cause for this is that you're already using one of the components of the system that is needed by the SD/MMC engine; in particular you need to not be using any of the pins that are used by the interface. Do you get this error if you try it immediately after a hard reset of the card with no `main.py` or `boot.py` code being run?

The other time that I've seen this error come up during card initialisation is when the ESP32 has had trouble negotiating with the card. I've seen this happen when the card is not properly seated in the microSD slot and when the card is broken. I also have one or two cards that for some reason don't like the ESP32 when communicating using the hardware SD/MMC interface but work fine using the SPI interface. If the card works fine in other machines and you are sure that it is not a problem with the card being loose in the slot then try communicating with it over SPI rather than using the pure hardware implementation. On the TTGO T8 you can do this with:
```
sd = machine.SDCard(slot=2, mosi=15, miso=2, sck=14, cs=13)
```
With those settings you'll be using the hardware SPI engine, DMA access for the data transfer and the C implementation of the SD Card protocol so it should still be about 10x to 12x faster than the pure software version.

dukeduck1984, 2019-08-11:
After I have freed pins 2, 4, 12, 13, 14, 15, the sd card can be initialized, but it gave me another error when mounting the sd card.  Pls see below.
```
>>> sd = machine.SDCard()
>>> import uos
>>> uos.mount(sd, '/sd')
E (76117) sdmmc_sd: sdmmc_init_sd_scr: send_scr (1) returned 0x109
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: 16
```

Communicating with the sd card over the SPI interface worked fine though.

dukeduck1984, 2019-08-11:
The pins 5, 18, 19, 22, 23, 25 are being used by the peripherals just in case the above error has anything to do with the pins.

dukeduck1984, 2019-08-18:
@nickovs it seems that 32gb sd cards are not supported?  I have tried 3 different brands (including SanDisk and Kingston) but none worked.

The 16gb sd card works, but I need to execute ```uos.mount``` twice in order to successfully mount it - as the first time always failed.  Pls see below.  Any idea?  BTW, this card was formatted by ```uos.VfsFat.mkfs(sd)``` as pointed out by Damien.
```
>>> import machine, uos
>>> sd = machine.SDCard()
>>> uos.mount(sd, '/sd')
E (21800) sdmmc_sd: sdmmc_init_sd_scr: send_scr (1) returned 0x109
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: 16
>>> uos.mount(sd, '/sd')
>>> uos.listdir('/sd')
['www']
```

Thank you!

grbd, 2019-08-29:
I've recently been trying this out on a TTGO-T8-ESP32 board
Using a 32Gb micro SD Card

  * https://github.com/micropython/micropython/blob/master/docs/library/machine.SDCard.rst
  * https://github.com/LilyGO/TTGO-T8-ESP32

Based on the wiring, it does appear to be wired correctly for 4 line Hardware SD Card Access

Using the latest master firmware (since the pull request for ESP32 SDIO went through after 1.11)
esp32spiram-20190829-v1.11-255-g400a128e1.bin

I seem to be getting the following results
```
import machine
import os
uos.mount(machine.SDCard(slot=1, width=4), "/sd")
```

```
E (13680) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
```

I've not tried software SPI yet using this API <br>
but one thought is if there's someway of influencing the speed of the access to the SD Card
I don't think that setting is in this API though

I might try formatting the SD card to 16Gb next and or trying out the SPI method to see if that works

nickovs, 2019-08-29:
@grbd I've not checked with a multimeter or oscilloscope but I *think* that the documentation page is incorrect and the T8 only supports 1-bit wide access. The back of the card (pictured below) lists the connections as only having the connections for 1-bit mode.
[image]

 Can you try it with width=1? If you have a multimeter you could check which pins are actually connected.

grbd, 2019-08-30:
I've already tried
```
import machine
import os
uos.mount(machine.SDCard(width=1), "/sd")
```

and
```
import machine
import os
uos.mount(machine.SDCard(slot=1, width=1), "/sd")
```

but both give the same error
```
E (29380) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
```

after that you need to reset the board otherwise attempting anything will just result in a
```
OSError: 259
```

I've also tried
```
import machine
x1 = machine.SDCard()
x1.info()
```
but same error
```
E (29380) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
```

Managed to find a 16Gb sd card, but that had the same result
Next I'm going to check the wiring on the pins as you've mentioned above and try the SPI option

grbd, 2019-08-30:
I've ordered some newer multimeter probes with smaller tips since my existing ones are too big to reach the pins
I've also ordered a ESP32-WROVER-KIT V4.1 board to test on

so far SPI seems to work okay
```
uos.mount(machine.SDCard(slot=2, sck=14, miso=2, mosi=15, cs=13), "/sd")
```
although it's a bit slow for what I want (I'm hoping to run a SPA webpage using micropython)
when those bits arrive I'll try and work out if those other Data lines are wired up

nickovs, 2019-08-30:
Since you are compiling from source, If you want to try to diagnose this further you could change line 39 of `ports/esp32/machine_sdcard.c` to read `#define DEBUG 1` and see if you get anything enlightening from the debug output.

grbd, 2019-08-31:
I've actually been using a download from the daily snapshot over at https://micropython.org/download since the pull request for SDIO was added in after 1.11
But I might try that debug method if I don't get anywhere with the official SDK board (which should be arriving Thursday)

cwt, 2019-08-31:
TTGO T8 schematic show that the SD DAT1 and DAT2 are not connected.
[image]

grbd, 2019-08-31:
I thought that might be the case when I was probing around but I was going to verify once I had some smaller probes
If Dat0 is wired up then 1 bit / 1 line mode should work.

I just tried the lobo fork using 1 line mode and that seems to work
https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/wiki/filesystems
https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/tree/master/MicroPython_BUILD/firmware
```
import os, uos
uos.sdconfig(uos.SDMODE_1LINE, maxspeed=40)
os.mountsd()
uos.listdir('/sd')
```

I've noticed with this api there's no maxspeed setting

nickovs, 2019-08-31:
The 1-bit mode works just fine for me on the two TTGO T8's that I have.

I will look at adding a maximum speed setting to the API but the default in the ESP IDF is 40MHz.

grbd, 2019-09-05:
Hi,
So I've now got a ESP-WROVER-KIT 4.1
The first bug I've found seems to be a difference between the esp32 and the esp32spiram versions of the firmware
in that with esp32spiram you can't specify a width parameter but with esp32 you can
I think this is a different issue from what's mentioned above so I've raised it as a seperate thing

Next knowing the above I did some testing of the ESP-WROVER-KIT 4.1 vs the TTGO board
making sure to use esp32 and not esp32spiram to avoid the first bug above
but using 1 line mode

This works fine on the ESP-WROVER-KIT for both SD cards I tried (16Gb and 32Gb)
```
import machine
uos.mount(machine.SDCard(width=1), "/sd")
uos.listdir('/sd')
uos.umount('/sd')
```

Trying the same firmware, same SD cards on the TTGO with width=1
results in the below
```
E (11610) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: 16
```

So this suggests a hardware problem maybe
I can't think of any way around that in software other than changing the speed to a lower speed
or looking if there's a pullup that needs to be set, which is already wired in on the ESP-WROVER-KIT
It's possible that there's different results with different SD cards due to the different speeds that they operate at

grbd, 2019-09-05:
Comparing the schematics side by side I've noticed with the ESP-WROVER-KIT there's 3.3V pullups on all the lines with 10K resistors
https://dl.espressif.com/dl/schematics/ESP-WROVER-KIT_SCH-2.pdf

With the TTGO it looks like they added in the protection diodes but didn't populate the pullup resistors
as they're labled as N/C (not connected)
https://github.com/LilyGO/TTGO-T8-ESP32/blob/master/Schematic.pdf

[image]

So one way to work around this might be to ether

  * try and set the pins manually to being pulled up from within micropython
  * have an option on the SDCard that enables pull ups

nickovs, 2019-09-05:
@grbd Thanks for the diagnostics on this. If you can build form source then can you try uncommenting [line 246 in `machine_sdcard.c`](https://github.com/micropython/micropython/blob/fb73bdae6256451838ef48565030f181077584fe/ports/esp32/machine_sdcard.c#L246) and see if it makes any difference? When I was writing the code found conflicting notes on if the internal pull-ups should be enabled and it was more reliable on my cards if I had them disabled. If you have a card that has either weaker pull-ups or no pull-ups then enabling the internal ones will likely help on your sort of hardware. If it does then I will add a parameter to the `SD` class to make this configurable.

grbd, 2019-09-05:
just finished building / trying but it didn't seem to make a difference
same error as before

grbd, 2019-09-05:
One way around this which might be more reliable as far as the TTGO board is concerned anyway is to just run some external pullup resistors from 3.3V to those IO pins since I think they're accessible on the pinout. I might try that later on.

grbd, 2019-09-06:
Yep just ran some tests and it does look like the lack of pull up resistors is the cause
I ran some external 10K resistors between 3.3V across to pins IO2, IO13, IO14, IO15 and then it worked fine. I even ran a web service from the sd card and that worked okay as well

I can only assume the Lobo firmware uses some other method to pull up the lines other than the expresif SDMMC_SLOT_FLAG_INTERNAL_PULLUP method
but given that it's kind of insufficient anyway it might explain why folks get random errors when trying to access the SD card

External pull up resistors for the TTGO looks like it's the way to go
although you will get more speed on the ESP-WROVER-KIT since it can do 4line mode whereas the TTGO only does 1 line mode

grbd, 2019-09-06:
For anyone else wanting to get this to work
[image]
[image]

nickovs, 2019-09-06:
OK. Glad you got it working. I'll have a look at the LoBo code and see if I can work out what it does that's different.

cwt, 2019-09-09:
I remember we had this issue on Lobo firmware long time ago, the solution before he make change to the firmware itself was manually pull-up the pins.
```python
>>> from machine import Pin
>>> p = Pin(2, Pin.IN, Pin.PULL_UP)
>>> p = Pin(4, Pin.IN, Pin.PULL_UP)
>>> p = Pin(12, Pin.IN, Pin.PULL_UP)
>>> p = Pin(13, Pin.IN, Pin.PULL_UP)
>>> p = Pin(14, Pin.IN, Pin.PULL_UP)
>>> p = Pin(15, Pin.IN, Pin.PULL_UP)
```
then Lobo make change to the firmware which I think it's in this file https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/blob/master/MicroPython_BUILD/components/micropython/extmod/vfs_native.c on function `_setPins` (line 724), and then we never need external pull-up or manually pull-up anymore.

My TTGO T8 V1.3 works fine with Lobo firmware using 1 bit mode.

pnrhub, 2019-11-05:
I noticed that this code line is commented out:
https://github.com/micropython/micropython/blob/d209f9ebe72aa2af0af72ed3bba737a47c45e567/ports/esp32/machine_sdcard.c#L246

It is the line that enables the pull ups on the MMC lines. Is there a reason that this line is commented out, or is it simply a typo???

dpgeorge, 2019-11-06:
> It is the line that enables the pull ups on the MMC lines. Is there a reason that this line is commented out, or is it simply a typo???

Apparently those internal pull ups are not enough (see above discussion).  Instead setting the pull up on a pin explicitly will get it working (see above).  Maybe that could be done automatically in the code.

pnrhub, 2019-11-06:
Ah, missed that earlier comment about  SDMMC_SLOT_FLAG_INTERNAL_PULLUP, sorry.

My setup is a LOLIN32 board + a micro-SD card hooked up. A friend is using the ULX3S FPGA board. Neither has external pull-ups. We're getting  the same result: using SPI access works just fine (slot 3). 4-wire MMC access does not. Enabling the pull-ups beforehand in Python does not make a difference. When testing with Arduino it does work.

Does 4-wire MMC work correctly on other boards?

pnrhub, 2019-11-06:
I'm reading up on some code, in particular the SD card example in the esp-idf tree. Following the code step-by-step I find that after calling [sdmmc_host_init_slot](https://github.com/espressif/esp-idf/blob/de9b0b1c56f59fc213586e4790f861c7109b4227/components/fatfs/vfs/vfs_fat_sdmmc.c#L77), it then continues to call [sdmmc_card_init](https://github.com/espressif/esp-idf/blob/de9b0b1c56f59fc213586e4790f861c7109b4227/components/fatfs/vfs/vfs_fat_sdmmc.c#L86). This second step does not seem to be present in the micropython driver.

Looking at the code of SDMMC_card_init itself, it seems necessary to call this to [switch the card into 4 bit mode](https://github.com/espressif/esp-idf/blob/a45e9985344575a80acb1dc9c12e7bec4d8af401/components/sdmmc/sdmmc_init.c#L107). Any thoughts?

peterschlem, 2020-04-26:
TTG0 board t8 - micropython 12 - SDCard(slot=3) worked for me - labels on back are: D2,D3,CMD,CLK,D0,D1 with corresponding pin #s 12, 13, 15, 14 , 2, 4 - looks like 12  and 13 are the misoa and misi pins

Pablo-No, 2020-05-12:
SDCard works fine with TTGO-T8 1.7,  TTGO-T8 1.8 in SPI mode with this code:
```
import machine, uos
uos.mount(machine.SDCard(slot=2, width=1, mosi=23, miso=19, sck=18, cs=5),"/sd")
```
[image]

Even with DOIT ESP32 DEVKIT V1 – 30 PINS in SPI mode with the default code:
```
import machine, uos
uos.mount(machine.SDCard(slot=2, width=1, mosi=23, miso=19, sck=18, cs=5),"/sd")
```
[image]
I want to try something like this with an eMMC device:
[image]

But I have a doubt. I have always a speed of 3 kB/s Is that a good speed?
It's the same speed that I get when I copy a file to the primary flash chip.
Can it be improved?

Thanks!

dizcza, 2021-09-03:
> I have always a speed between 15-18 kB/s Is that a good speed?
> The speed I get when I copy a file to the primary flash chip is 30 kB/s.

I've got ~500 kB/s and ~250 kB/s for the read & write speed respectively with the 1-bit width SD MMC driver flashed with Arduino IDE.

jonnor, 2024-09-08:
It seems that this now works? Mostly out-of-the-box, and if not, then people should try enabling pull-ups manually. Can this issue be closed?

closed 2024-09-09 by dpgeorge