micropython/micropython #4724: Merging BLE onto ESP32?

PieChiu, 2019-04-26:
Hi guys,
It is a pity to find that BLE module is not the supported on the port of ESP32 yet.
I try the port of nrf, and the BLE module of micropython seems good.
So, do we have a plan of merging BLE onto ESP32?
Or some progress has already made?
Thanks.

homeoflittlewolf, 2019-04-28:
Hello, Micropython community. 
Thank you very much for supporting ESP32 Ble function. If possibility  I think many markets and people will benefit from Micropython. 
Of course, the adaptation process Espressif can cooperate with the operation, only let us know the problem points and what interface support we need to provide.

dpgeorge, 2019-04-30:
> So, do we have a plan of merging BLE onto ESP32?
> Or some progress has already made?

BLE support is being worked on in #4589, which includes ESP32 BLE.

> Of course, the adaptation process Espressif can cooperate with the operation, only let us know the problem points and what interface support we need to provide.

Thank you, we will let you know if there are issues that you can help with.

closed 2019-04-30 by dpgeorge

PieChiu, 2019-05-06:
Great.
Hope it come soon.