micropython/micropython #4725: Javascript Test Failure Rate

itsMikeLowrey, 2019-04-27:
I ran micropython with emscripten and fastcomp built from source. I also made the patch stated in the read.me. My test suite ran with the following results:

> 682 tests performed (19502 individual testcases)
> 391 tests passed
> 59 tests skipped

 I wanted to check in that I did not do anything majorly wrong and these test results align with what is expected from this specific build at this time. Awesome project and documentation by the way.

dpgeorge, 2019-04-28:
Thanks for reporting this.  The failure rate you see is too high and it's mostly due to `new Buffer(1)` being deprecated so printing a deprecation warning during the test; now fixed by bd0bacb637fb11556e76cbd929ed833dcd3abe97.  Also a few other minor issues were fixed and I now get the following for `make test`:
```
647 tests performed (19226 individual testcases)
599 tests passed
94 tests skipped: ...
48 tests failed: ...
```
The remaining failures are mainly importing, file IO and recursive overflow errors.

itsMikeLowrey, 2019-04-28:
Thank you so much. Great project also.

closed 2019-04-28 by itsMikeLowrey