micropython/micropython #4726: ESP32 Can  support bluetooth ，But micropython tean no support  blue doc

panhuaguo, 2019-04-28:
ESP32 Can  support bluetooth ，But micropython tean no support  blue doc

ThomasChr, 2019-04-29:
Please try to use whole sentences, I have absolutely no idea what you are talking about.

JulianJose, 2019-04-29:
I think he is talking about the implementation of bluetooth without documentation. 
It is a very new feature, so maybe the documentation about it will take some months.

AidenRay, 2019-05-01:
I'm assuming by this question that [this issue](https://github.com/micropython/micropython/issues/3809) is nearly resolved?

Or is [the bluetooth branch](https://github.com/aykevl/micropython/tree/bluetooth) still in need of merging?

JulianJose, 2019-05-05:
Hi AidenRay, thanks for your answer. Would you know an example more complete about how we can use this feature?

AidenRay, 2019-05-09:
If you follow the issue I linked you'll see an example posted by the author of the branch:
```
import bluetooth
bt = bluetooth.Bluetooth()               # singleton
bt.active(True)                          # activate BT stack
bt.advertise(100, 'hello MicroPython!')  # start advertisement (interval=100ms, name=..., connectable=True)
# [...]
bt.advertise(0)                          # stop advertisement (interval=0)
```

Currently, the functionality is limited.
See the commits on the bluetooth branch for the latest updated [here](https://github.com/aykevl/micropython/commits/bluetooth)

jacklinquan, 2019-05-09:
BLE UART from aykevl

`
import bluetooth
bt = bluetooth.Bluetooth()
bt.active(1)
bt.advertise(100, 'MicroPython')
print('----')
tx = bluetooth.Characteristic('6E400002-B5A3-F393-E0A9-E50E24DCCA9E', bluetooth.FLAG_READ|bluetooth.FLAG_NOTIFY)
rx = bluetooth.Characteristic('6E400003-B5A3-F393-E0A9-E50E24DCCA9E', bluetooth.FLAG_WRITE)
s = bt.add_service('6E400001-B5A3-F393-E0A9-E50E24DCCA9E', [tx, rx])

def callback(char, data):
    print('data:', data)

rx.on_update(callback)

`

closed 2019-05-20 by dpgeorge