micropython/micropython #4728: NRF non-fatal build warning (mp_qstr_const_pool)

pacmac, 2019-04-29:
Latest still builds, but I thought that I would report it anyway.

```
...
CC modules/music/musictunes.c
CC modules/ble/modble.c
CC modules/random/modrandom.c
CC ../../lib/nrfx/mdk/system_nrf52840.c
CC build-pca10056-s140/pins_gen.c
LINK build-pca10056-s140/firmware.elf
build-pca10056-s140/frozen_mpy.c:74:26: warning: size of 'mp_qstr_const_pool' differ from the size of original declaration [-Wlto-type-mismatch]
 extern const qstr_pool_t mp_qstr_const_pool;
                          ^
../../py/qstr.c:102:19: note: 'mp_qstr_const_pool' was previously declared here
 const qstr_pool_t mp_qstr_const_pool = {
                   ^
../../py/qstr.c:117:26: warning: size of 'mp_qstr_frozen_const_pool' differ from the size of original declaration [-Wlto-type-mismatch]
 extern const qstr_pool_t MICROPY_QSTR_EXTRA_POOL;
                          ^
build-pca10056-s140/frozen_mpy.c:75:19: note: 'mp_qstr_frozen_const_pool' was previously declared here
 const qstr_pool_t mp_qstr_frozen_const_pool = {
                   ^
   text	   data	    bss	    dec	    hex	filename
 159244	      8	   1712	 160964	  274c4	build-pca10056-s140/firmware.elf
arm-none-eabi-objcopy -O binary build-pca10056-s140/firmware.elf build-pca10056-s140/firmware.bin
arm-none-eabi-objcopy -O ihex build-pca10056-s140/firmware.elf build-pca10056-s140/firmware.hex
```

dpgeorge, 2019-04-30:
It builds without a warning for me, using `arm-none-eabi-gcc` version 8.3.0.  It's likely the compiler you use is older and getting confused by the variable sized array when doing LTO.

pacmac, 2019-04-30:
I am using ubuntu 18.04 LTS which is fully patched / updated:

```
arm-none-eabi-gcc --version
arm-none-eabi-gcc (15:6.3.1+svn253039-1build1) 6.3.1 20170620
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

I assume you are using ubuntu 18 too, so how are you able to get version 8.3.0 ? ?

pacmac, 2019-04-30:
OK, have updated and able to get 7.3.1:

```
apt-get remove gcc-arm-none-eabi binutils
apt install software-properties-common
apt update
apt upgrade
apt autoremove ## Important (ERROR: trying to overwrite '/usr/bin/arm-none-eabi-addr2line', which is also in package binutils-arm-none-eabi 2.27-9ubuntu1+9)
add-apt-repository ppa:team-gcc-arm-embedded/ppa
apt-get install gcc-arm-embedded
```

```
arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2018-q3-update) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

closed 2019-04-30 by pacmac

dpgeorge, 2019-04-30:
> I assume you are using ubuntu 18 too, so how are you able to get version 8.3.0 ?

I use Arch Linux.

pacmac, 2019-04-30:
I've been using arch linux for al least 10 years, and switched all new servers to ubuntu LTS recently, I found that unless I updated arch linux VERY often, it would fail frequently with keyring import and other related errors.

Even running the pacman keyring fixes in some cases did not work, agreed that it's lighter and faster but everyone else seems to us ubuntu, probably because the ease of installation !