micropython/micropython #4729: machine module missing from build for STM32F4 Discovery board

TacosNChips, 2019-04-30:
I downloaded the last few builds for the  STM32F4 Discovery board from http://micropython.org/download

Unfortunately, all were missing the machine module, pyb module, or anything similar to that.

dpgeorge, 2019-04-30:
> all were missing the machine module, pyb module, or anything similar to that.

They should be included by default.  What is the output from entering `help('modules')` at the prompt?  Also, when you press ctrl-D what is the banner that is printed?

TacosNChips, 2019-04-30:
Unfortunately, I went home for the day.  I will post the output for you tomorrow.

I did check help('modules').  I found lots of modules, but not pyb or machine.

TacosNChips, 2019-04-30:
help("modules")

Please wait a moment while I gather a list of all available modules...

C:\Program Files\Mu\pkgs\IPython\kernel\__init__.py:13: ShimWarning: The `IPython.kernel` package has been deprecated since IPython 4.0.You should import from ipykernel or jupyter_client instead.
  "You should import from ipykernel or jupyter_client instead.", ShimWarning)
C:\Program Files\Mu\pkgs\gpiozero\devices.py:452: PinFactoryFallback: Falling back from rpigpio: No module named 'RPi'
  'Falling back from %s: %s' % (name, str(e))))
C:\Program Files\Mu\pkgs\gpiozero\devices.py:452: PinFactoryFallback: Falling back from rpio: No module named 'RPIO'
  'Falling back from %s: %s' % (name, str(e))))
%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
Can't connect to pigpio at localhost(8888)

Did you start the pigpio daemon? E.g. sudo pigpiod

Did you specify the correct Pi host/port in the environment
variables PIGPIO_ADDR/PIGPIO_PORT?
E.g. export PIGPIO_ADDR=soft, export PIGPIO_PORT=8888

Did you specify the correct Pi host/port in the
pigpio.pi() function? E.g. pigpio.pi('soft', 8888)
%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
C:\Program Files\Mu\pkgs\gpiozero\devices.py:452: PinFactoryFallback: Falling back from pigpio: failed to connect to localhost:8888
  'Falling back from %s: %s' % (name, str(e))))
C:\Program Files\Mu\pkgs\gpiozero\devices.py:452: PinFactoryFallback: Falling back from native: module 'os' has no attribute 'O_SYNC'
  'Falling back from %s: %s' % (name, str(e))))
IPython             audioop             itertools           select
PIL                 autoreload          jedi                selectors
PyQt5               base64              json                semver
__future__          bdb                 jupyter             serial
_ast                binascii            jupyter_client      setuptools
_asyncio            binhex              jupyter_core        shelve
_bisect             bisect              keyword             shlex
_blake2             builtins            lib2to3             shutil
_bootlocale         bz2                 linecache           signal
_bz2                cProfile            locale              simplegeneric
_codecs             calendar            logging             sip
_codecs_cn          certifi             lzma                site
_codecs_hk          cgi                 macpath             six
_codecs_iso2022     cgitb               macurl2path         smtpd
_codecs_jp          chardet             mailbox             smtplib
_codecs_kr          chunk               mailcap             sndhdr
_codecs_tw          cmath               marshal             socket
_collections        cmd                 math                socketserver
_collections_abc    code                matplotlib          sqlite3
_compat_pickle      codecs              mimetypes           sre_compile
_compression        codeop              mmap                sre_constants
_csv                collections         modulefinder        sre_parse
_ctypes             colorsys            msilib              ssl
_datetime           compileall          msvcrt              stat
_decimal            concurrent          mu                  statistics
_dummy_thread       configparser        multiprocessing     storemagic
_elementtree        contextlib          netrc               string
_findvs             copy                nntplib             stringprep
_functools          copyreg             nt                  struct
_hashlib            crypt               ntpath              subprocess
_heapq              csv                 nturl2path          sunau
_imp                ctypes              nudatus             symbol
_io                 curses              numbers             sympyprinting
_json               cycler              numpy               symtable
_locale             cythonmagic         opcode              sys
_lsprof             datetime            operator            sysconfig
_lzma               dateutil            optparse            tabnanny
_markupbase         dbm                 os                  tarfile
_md5                decimal             parser              telnetlib
_msi                decorator           parso               tempfile
_multibytecodec     difflib             pathlib             tests
_multiprocessing    dis                 pdb                 textwrap
_opcode             distutils           pexpect             this
_operator           doctest             pgzero              threading
[... 21 lines cut ...]
_thread             gpiozerocli         pyexpat             wave
_threading_local    guizero             pyflakes            wcwidth
_tkinter            gzip                pygame              weakref
_tracemalloc        hashlib             pygments            webbrowser
_warnings           heapq               pylab               winreg
_weakref            hmac                pyparsing           winsound
_weakrefset         html                pytz                wsgiref
_winapi             http                qtconsole           xdrlib
abc                 idna                queue               xml
aifc                imaplib             quopri              xmlrpc
antigravity         imghdr              random              xxsubtype
appdirs             imp                 re                  zipapp
argparse            importlib           reprlib             zipfile
array               inspect             requests            zipimport
[... cut ...]
asyncio             ipykernel           runpy               
asyncore            ipykernel_launcher  sched               
atexit              ipython_genutils    secrets             

Enter any module name to get more help.  Or, type "modules spam" to search
for modules whose name or summary contain the string "spam".

C:\Program Files\Mu\Python\python36.zip\pkgutil.py:107: VisibleDeprecationWarning: zmq.eventloop.minitornado is deprecated in pyzmq 14.0 and will be removed.
    Install tornado itself to use zmq with the tornado IOLoop.

TacosNChips, 2019-04-30:
Ctrl-D on the REPL in Mu does not have any noticeable effect.  This is the banner at the start of the REPL session though.

Jupyter QtConsole 4.3.1
Python 3.6.3 (v3.6.3:2c5fed8, Oct  3 2017, 18:11:49) [MSC v.1900 64 bit (AMD64)]
Type 'copyright', 'credits' or 'license' for more information
IPython 6.2.1 -- An enhanced Interactive Python. Type '?' for help.

dhylands, 2019-04-30:
It looks like you're not talking to the STM32F4DISC board, and instead . I just flashed this image:
http://micropython.org/resources/firmware/STM32F4DISC-20190430-v1.10-308-g7b5400134.dfu

and help('modules') shows
```
MicroPython v1.10-308-g7b5400134 on 2019-04-30; F4DISC with STM32F407
Type "help()" for more information.
>>> help('modules')
__main__          io                select            ujson
_onewire          json              socket            umachine
array             lcd160cr          stm               uos
binascii          lcd160cr_test     struct            urandom
builtins          machine           sys               ure
cmath             math              time              uselect
collections       micropython       ubinascii         usocket
dht               network           ucollections      ustruct
errno             onewire           uctypes           utime
framebuf          os                uerrno            utimeq
gc                pyb               uhashlib          uzlib
hashlib           random            uheapq            zlib
heapq             re                uio
Plus any modules on the filesystem
>>> 
```

Your output looks like the output from CPython, and not MicroPython.

closed 2019-05-14 by dpgeorge