micropython/micropython #4732: <WLAN>.isconnected returns false when ESP32 connected to Wifi

DevAerial, 2019-04-30:
In `boot.py` I have following code that connects ESP32 NodeMCU to WiFi network.
```python
import network
import ujson
from time import sleep
from machine import Pin

def read_networks_from_config():
    with open('config.json', 'r') as file:
        data = ujson.load(file)
    return data['Networks']

networks = read_networks_from_config()

sta_if = network.WLAN(network.STA_IF)
sta_if.active(True)

print('Scanning network...')

discovered_networks = sta_if.scan()
for network in discovered_networks:
    n_ssid = network[0].decode('utf-8')
    print('Checking network {}'.format(n_ssid))
    net_info = list(filter(lambda x: x['SSID'] == n_ssid, networks))
    if net_info:
        net_info = net_info[0]
        sta_if.connect(net_info['SSID'], net_info['Password'])
        if sta_if.isconnected():
            led = Pin(2, Pin.OUT)
            led.value(True)
            print('Connected...')
            break
        else:
            print('Failed to connect...')
  ```
If device successfully connected to WiFi network then onboard led should turn on, but for some reason `if sta_if.isconnected():` statement always returns `False` even though microcontroller is visible in network and is pingable. Weird thing I noticed that when connecting to device using `rshell` utility like this:
```shell
$  rshell --buffer-size=30 -p /dev/ttyUSB0 
```
`if sta_if.isconnected():` does not return `False` value and led turns on. Also in `repl` `isconnected` returns `True` too.

Firmware information:
```python
>>> os.uname()
(sysname='esp32', nodename='esp32', release='1.10.0', version='v1.10-298-g47e76b527 on 2019-04-18', machine='ESP32 module with ESP32')
```

dpgeorge, 2019-05-01:
Please try using the latest firmware, not the bluetooth one, and see if that resolves the issue.

closed 2019-06-02 by DevAerial

fengkaijia, 2021-05-29:
@deepaerial Did you managed to solve it?

I have a very similar situation. I can connect to WIFI router when REPL is on (e.g. via `rshell` or `picocom`); but without REPL, the WIFI connection status will always be `STAT_CONNECTING` and `isconnected()==False`. So far, the ESP32 board failed to connect to all my routers without REPL, but it did manage to connect to a hotspot I created from my Android phone.

What's different in my case, is that my device actually didn't connect to the router, i.e. not pingable. So `isconnected()==False` might actually be correct.

I can confirm this has nothing to do with BLE, as I tried on a custom firmware with:

```
#define MICROPY_PY_BLUETOOTH (0)
#define MICROPY_HW_ENABLE_SDCARD (0)
```

@dpgeorge Can you advice how I can debug this issue? Thanks!