micropython/micropython #4733: PYB11 Build with WIZNET5K & LWIP Fails

pacmac, 2019-05-01:
```
make -j4 BOARD=PYBV11 MICROPY_PY_WIZNET5K=5500 MICROPY_PY_LWIP=1
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
make: Warning: File 'mpconfigport.mk' has modification time 0.25 s in the future
make[1]: Entering directory '/nfs/qnap/dev/uPython/build/micropython/micropython/mpy-cross'
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
GEN build-PYBV11/genhdr/qstr.i.last
GEN build/genhdr/qstr.i.last
In file included from ../lib/lwip/src/include/lwip/init.h:40:0,
                 from ../extmod/modlwip.c:40:
../lib/lwip/src/include/lwip/opt.h:51:10: fatal error: lwipopts.h: No such file or directory
 #include "lwipopts.h"
          ^~~~~~~~~~~~
compilation terminated.
In file included from ../lib/lwip/src/core/def.c:61:0:
../lib/lwip/src/include/lwip/opt.h:51:10: fatal error: lwipopts.h: No such file or directory
 #include "lwipopts.h"
          ^~~~~~~~~~~~
compilation terminated.
In file included from ../lib/lwip/src/core/dns.c:85:0:
../lib/lwip/src/include/lwip/opt.h:51:10: fatal error: lwipopts.h: No such file or directory
 #include "lwipopts.h"
          ^~~~~~~~~~~~
compilation terminated.
In file included from ../lib/lwip/src/core/inet_chksum.c:48:0:
../lib/lwip/src/include/lwip/opt.h:51:10: fatal error: lwipopts.h: No such file or directory
 #include "lwipopts.h"
          ^~~~~~~~~~~~
compilation terminated.
In file included from ../lib/lwip/src/core/init.c:38:0:
../lib/lwip/src/include/lwip/opt.h:51:10: fatal error: lwipopts.h: No such file or directory
 #include "lwipopts.h"
          ^~~~~~~~~~~~
compilation terminated.
In file included from ../lib/lwip/src/core/ip.c:56:0:
../lib/lwip/src/include/lwip/opt.h:51:10: fatal error: lwipopts.h: No such file or directory
 #include "lwipopts.h"
          ^~~~~~~~~~~~
compilation terminated.
In file included from ../lib/lwip/src/core/mem.c:56:0:
../lib/lwip/src/include/lwip/opt.h:51:10: fatal error: lwipopts.h: No such file or directory
 #include "lwipopts.h"
          ^~~~~~~~~~~~
compilation terminated.
In file included from ../lib/lwip/src/core/memp.c:46:0:
../lib/lwip/src/include/lwip/opt.h:51:10: fatal error: lwipopts.h: No such file or directory
 #include "lwipopts.h"
          ^~~~~~~~~~~~
compilation terminated.
In file included from ../lib/lwip/src/core/netif.c:51:0:
../lib/lwip/src/include/lwip/opt.h:51:10: fatal error: lwipopts.h: No such file or directory
 #include "lwipopts.h"
          ^~~~~~~~~~~~
compilation terminated.
In file included from ../lib/lwip/src/core/pbuf.c:112:0:
../lib/lwip/src/include/lwip/opt.h:51:10: fatal error: lwipopts.h: No such file or directory
 #include "lwipopts.h"
          ^~~~~~~~~~~~
compilation terminated.
In file included from ../lib/lwip/src/core/raw.c:48:0:
../lib/lwip/src/include/lwip/opt.h:51:10: fatal error: lwipopts.h: No such file or directory
 #include "lwipopts.h"
          ^~~~~~~~~~~~
compilation terminated.
In file included from ../lib/lwip/src/core/stats.c:39:0:
../lib/lwip/src/include/lwip/opt.h:51:10: fatal error: lwipopts.h: No such file or directory
 #include "lwipopts.h"
          ^~~~~~~~~~~~
compilation terminated.
In file included from ../lib/lwip/src/core/sys.c:77:0:
../lib/lwip/src/include/lwip/opt.h:51:10: fatal error: lwipopts.h: No such file or directory
 #include "lwipopts.h"
          ^~~~~~~~~~~~
compilation terminated.
In file included from ../lib/lwip/src/core/tcp.c:48:0:
../lib/lwip/src/include/lwip/opt.h:51:10: fatal error: lwipopts.h: No such file or directory
 #include "lwipopts.h"
          ^~~~~~~~~~~~
compilation terminated.
In file included from ../lib/lwip/src/core/tcp_in.c:44:0:
../lib/lwip/src/include/lwip/opt.h:51:10: fatal error: lwipopts.h: No such file or directory
 #include "lwipopts.h"
          ^~~~~~~~~~~~
compilation terminated.
In file included from ../lib/lwip/src/core/tcp_out.c:41:0:
../lib/lwip/src/include/lwip/opt.h:51:10: fatal error: lwipopts.h: No such file or directory
 #include "lwipopts.h"
          ^~~~~~~~~~~~
compilation terminated.
In file included from ../lib/lwip/src/core/timeouts.c:42:0:
../lib/lwip/src/include/lwip/opt.h:51:10: fatal error: lwipopts.h: No such file or directory
 #include "lwipopts.h"
          ^~~~~~~~~~~~
compilation terminated.
In file included from ../lib/lwip/src/core/udp.c:48:0:
../lib/lwip/src/include/lwip/opt.h:51:10: fatal error: lwipopts.h: No such file or directory
 #include "lwipopts.h"
          ^~~~~~~~~~~~
compilation terminated.
In file included from ../lib/lwip/src/core/ipv4/autoip.c:59:0:
../lib/lwip/src/include/lwip/opt.h:51:10: fatal error: lwipopts.h: No such file or directory
 #include "lwipopts.h"
          ^~~~~~~~~~~~
compilation terminated.
In file included from ../lib/lwip/src/core/ipv4/dhcp.c:63:0:
../lib/lwip/src/include/lwip/opt.h:51:10: fatal error: lwipopts.h: No such file or directory
 #include "lwipopts.h"
          ^~~~~~~~~~~~
compilation terminated.
In file included from ../lib/lwip/src/core/ipv4/etharp.c:46:0:
../lib/lwip/src/include/lwip/opt.h:51:10: fatal error: lwipopts.h: No such file or directory
 #include "lwipopts.h"
          ^~~~~~~~~~~~
compilation terminated.
In file included from ../lib/lwip/src/core/ipv4/icmp.c:42:0:
../lib/lwip/src/include/lwip/opt.h:51:10: fatal error: lwipopts.h: No such file or directory
 #include "lwipopts.h"
          ^~~~~~~~~~~~
compilation terminated.
In file included from ../lib/lwip/src/core/ipv4/igmp.c:83:0:
../lib/lwip/src/include/lwip/opt.h:51:10: fatal error: lwipopts.h: No such file or directory
 #include "lwipopts.h"
          ^~~~~~~~~~~~
compilation terminated.
In file included from ../lib/lwip/src/core/ipv4/ip4_addr.c:39:0:
../lib/lwip/src/include/lwip/opt.h:51:10: fatal error: lwipopts.h: No such file or directory
 #include "lwipopts.h"
          ^~~~~~~~~~~~
compilation terminated.
In file included from ../lib/lwip/src/core/ipv4/ip4.c:41:0:
../lib/lwip/src/include/lwip/opt.h:51:10: fatal error: lwipopts.h: No such file or directory
 #include "lwipopts.h"
          ^~~~~~~~~~~~
compilation terminated.
In file included from ../lib/lwip/src/core/ipv4/ip4_frag.c:41:0:
../lib/lwip/src/include/lwip/opt.h:51:10: fatal error: lwipopts.h: No such file or directory
 #include "lwipopts.h"
          ^~~~~~~~~~~~
compilation terminated.
In file included from ../lib/lwip/src/core/ipv6/dhcp6.c:42:0:
../lib/lwip/src/include/lwip/opt.h:51:10: fatal error: lwipopts.h: No such file or directory
 #include "lwipopts.h"
          ^~~~~~~~~~~~
compilation terminated.
In file included from ../lib/lwip/src/core/ipv6/ethip6.c:42:0:
../lib/lwip/src/include/lwip/opt.h:51:10: fatal error: lwipopts.h: No such file or directory
 #include "lwipopts.h"
          ^~~~~~~~~~~~
compilation terminated.
In file included from ../lib/lwip/src/core/ipv6/icmp6.c:42:0:
../lib/lwip/src/include/lwip/opt.h:51:10: fatal error: lwipopts.h: No such file or directory
 #include "lwipopts.h"
          ^~~~~~~~~~~~
compilation terminated.
In file included from ../lib/lwip/src/core/ipv6/inet6.c:42:0:
../lib/lwip/src/include/lwip/opt.h:51:10: fatal error: lwipopts.h: No such file or directory
 #include "lwipopts.h"
          ^~~~~~~~~~~~
compilation terminated.
In file included from ../lib/lwip/src/core/ipv6/ip6_addr.c:43:0:
../lib/lwip/src/include/lwip/opt.h:51:10: fatal error: lwipopts.h: No such file or directory
 #include "lwipopts.h"
          ^~~~~~~~~~~~
compilation terminated.
In file included from ../lib/lwip/src/core/ipv6/ip6.c:42:0:
../lib/lwip/src/include/lwip/opt.h:51:10: fatal error: lwipopts.h: No such file or directory
 #include "lwipopts.h"
          ^~~~~~~~~~~~
compilation terminated.
In file included from ../lib/lwip/src/core/ipv6/ip6_frag.c:42:0:
../lib/lwip/src/include/lwip/opt.h:51:10: fatal error: lwipopts.h: No such file or directory
 #include "lwipopts.h"
          ^~~~~~~~~~~~
compilation terminated.
In file included from ../lib/lwip/src/core/ipv6/mld6.c:49:0:
../lib/lwip/src/include/lwip/opt.h:51:10: fatal error: lwipopts.h: No such file or directory
 #include "lwipopts.h"
          ^~~~~~~~~~~~
compilation terminated.
In file included from ../lib/lwip/src/core/ipv6/nd6.c:44:0:
../lib/lwip/src/include/lwip/opt.h:51:10: fatal error: lwipopts.h: No such file or directory
 #include "lwipopts.h"
          ^~~~~~~~~~~~
compilation terminated.
In file included from ../lib/lwip/src/netif/ethernet.c:41:0:
../lib/lwip/src/include/lwip/opt.h:51:10: fatal error: lwipopts.h: No such file or directory
 #include "lwipopts.h"
          ^~~~~~~~~~~~
compilation terminated.
../py/mkrules.mk:74: recipe for target 'build/genhdr/qstr.i.last' failed
make[1]: *** [build/genhdr/qstr.i.last] Error 1
make[1]: *** Deleting file 'build/genhdr/qstr.i.last'
make[1]: Leaving directory '/nfs/qnap/dev/uPython/build/micropython/micropython/mpy-cross'
../../py/mkrules.mk:108: recipe for target '../../mpy-cross/mpy-cross' failed
make: *** [../../mpy-cross/mpy-cross] Error 2
make: *** Waiting for unfinished jobs....
```

dpgeorge, 2019-05-01:
This is another case of environment variables contaminating the mpy-cross build.  To workaround the issue explicitly build mpy-cross first, eg `make -C ../../mpy-cross` (from the stm32 dir).

dpgeorge, 2019-06-03:
Should be fixed by 41739506589ec8397613c86d8f682fb7f86c0a9f

closed 2019-06-03 by dpgeorge