micropython/micropython #4734: esp8266, 16MB

apollo2mond, 2019-05-01:
Hi,
the latest mpy version can't be flashed to support 16MB flash.
the esp will be end up in a reset -loop.
with the version before it work as expected!

robert-hh, 2019-05-01:
On flashing please specify explicitly a flash size of 4 MB

apollo2mond, 2019-05-01:
With 4MB it works BUT I want to use the 16MB for the file system AND the version before has no problem if I flash like:
esptool.py --port COMxx --baud 460800 write_flash -fm dio -fs 16MB 0 esp8266-20180511-v1.9.4.bin 0xffc000 esp_init_data_default.bin

robert-hh, 2019-05-01:
Interestingly that worked. I took an old ESP01 with the 512k flash replaced by an 16MB flash,
- erased it
- flashed 1.9.4 with the command you used
- tested it. I had 16 MB file system
- flash a new version 1.10-315 using the "make deploy" approach
that still had the 16 MB file system, I created 3 files of 5MByt each, and the device still boots and resets fine.

apollo2mond, 2019-05-05:
I want to reproduce/follow your approach and need help, what means "make deploy" in detail?

robert-hh, 2019-05-05:
"make deploy" is the command which I use to falsh the firmware to the device after I have build the firmware myself. But if simply calls esptool.py. SO if you use esptool.py to flash an image from the download page, but without the '0xffc000 esp_init_data_default.bin', the you get the same. So the command is at my system for instance:

esptool.py --port /dev/ttyUSB0 --baud 230400 write_flash --verify --flash_size=detect --flash_mode=qio 0 esp8266-20190505-v1.10-323-g906fb89fd.bin

apollo2mond, 2019-05-05:
Hi, I have anticipated your approach and flashed again with last build without success. I still end up in a endless loop with resets. My hw is a wemos D1 mini  module with 16MB.

peterhinch, 2019-05-05:
Try `--flash_mode=dio`.

apollo2mond, 2019-05-05:
Unfortunately the same result - endless loop with resets.
Have you any other reasons in mind for this issue?
As mentioned with v1.9.4 it works!

robert-hh, 2019-05-05:
In this thread: https://github.com/micropython/micropython/issues/2335  there is some discussion about the topic. One of these tells that you have to use your own build. So I made one and placed it here:
https://github.com/robert-hh/Shared-Stuff
It is called "firmware-combined.bin"
It has my own convenience package, meaning, that it excludes btree, but includes upysh.py (unix like shell commands like ls, ...), pye.py (Editor) and uftpd.py (ftp server).

apollo2mond, 2019-05-05:
This is also not the right medicine - again endless loop with resets and v1.9.4 still works.

robert-hh, 2019-05-05:
Did you also write again esp_init_data_default.bin, like recommended in the other thread?

apollo2mond, 2019-05-05:
With your fw build AND esp_init_data_default.bin it works! 
flash mode was still qio.
I will install now all tools to make own builds. :-)

Big thanks!