micropython/micropython #4736: esp8622, mpy-cross

apollo2mond, 2019-05-01:
hi,
with the latest mpy AND myp-cross a incompatible issue exists!
And NO, I have not mixed up the different versions - for sure after countless trails!

robert-hh, 2019-05-01:
Please give some details on:
- the firmware version of the ESP8266
- the version or  source of the mpy-cross executable-
- how you created the mpy file.

apollo2mond, 2019-05-01:
firmware version of the ESP8266: 
v1.10-261-gd89ce2ed1 on 2019-04-04
version or source of the mpy-cross executable:
v1.10
how created the mpy file: ... just picked up here
https://pypi.org/project/mpy-cross/

robert-hh, 2019-05-01:
> how created the mpy file: ... just picked up here
> https://pypi.org/project/mpy-cross/

My question was more about the .mpy  files created with mpy-cross. So you made them using the mpy-cross.exe downloaded from that site. I just started the WIN32 version, and it looks like an old version. So better create an up-to-date version by yourself.
For convenience, I have placed a win32 version of mpy-cross here: https://github.com/robert-hh/Shared-Stuff/blob/master/mpy-cross.exe
I tested it with the latest esp8266 firmware and it works, at least no complaint about versions.

apollo2mond, 2019-05-01:
Robert, many thanks! 
I can confirm that it works with this version.

closed 2019-05-01 by apollo2mond