micropython/micropython #4738: esp8266, log10(x), log2(x) have wrong syntax

apollo2mond, 2019-05-01:
log10(x), log2(x) are not implemented as documented.
instead the syntax is log(x, 10) and log(x, 2)!

dpgeorge, 2019-05-01:
In general, not all functions/classes/methods are available on all ports, due to space restrictions (this is mentioned in the docs).  On the esp8266 the extended math functions are not available, which includes `math.log2` and `math.log10`.  Instead use `math.log`.

closed 2019-05-01 by dpgeorge

t-m-cor, 2024-09-09:
This will make it more "portable" for different boards:
```
def log10(x)
    return log(x) / 4.60517 # actually: log(x) / log(10)
```
Or:
```
def log10(x)
    return log(x, 10)
```

but according to micropython docs, _math.log_ takes exactly one argument. So last suggesting violates docs, and might stop working with no notice (?).

(yes I know it's an old post, but I switched to a ESP8266, and got no-such-function error)

dpgeorge, 2024-09-10:
> but according to micropython docs, _math.log_ takes exactly one argument

You are right.  The docs need updating.