micropython/micropython #4741: stm32f7: Hard Fault from SDCard Interrupt Handler

andrewleech, 2019-05-02:
We've just started to increase the amount of data being logged to SD/MMC card and think I've come across an issue in the HAL drivers.
Not sure whether it's feasible to make a reproduction case, so just want to document what I've found here while in the middle of a debugging session. I'm running master with MMC enabled, but it looks like the same code path exists in SD card too.

Starting from: https://github.com/micropython/micropython/blob/master/ports/stm32/sdcard.c#L608
we start a new `HAL_MMC_WriteBlocks_DMA()` then `sdcard_wait_finished()`

sdcard_wait_finished basically monitors flags set in the SD/MMC interrupt handler.

While it's waiting, the DMA does its thing and the interrupt handler fires to monitor state, etc:
https://github.com/micropython/stm32lib/blob/a5e93de18479879dd129cf6272cfd75e0c794bd4/STM32F7xx_HAL_Driver/Src/stm32f7xx_hal_mmc.c#L1364

The problem comes when it gets to https://github.com/micropython/stm32lib/blob/a5e93de18479879dd129cf6272cfd75e0c794bd4/STM32F7xx_HAL_Driver/Src/stm32f7xx_hal_mmc.c#L1425
```
if(__HAL_MMC_GET_FLAG(hmmc, SDMMC_IT_TXFIFOHE) != RESET)
  {
    __HAL_MMC_CLEAR_FLAG(hmmc, SDMMC_FLAG_TXFIFOHE);

    MMC_Write_IT(hmmc);
  }
```
Note: `SDMMC_IT_TXFIFOHE` = Transmit FIFO Half Empty flag
That flag seems to get set pretty regularly during DMA usage, just as the fifo is used normally internally.
This mean `MMC_Write_IT()` gets run, which to paraphase basically does 
```
SDMMC_WriteFIFO(hmmc->pTxBuffPtr++)
```
`hmmc->pTxBuffPtr` is only set in the `HAL_MMC_WriteBlocks_IT` version of usage, so it starts at 0 and is incremented each interrupt that triggers by this.

I hit this issue when I get a hard fault on the `SDMMC_WriteFIFO()` line when `hmmc->pTxBuffPtr` == 0x4000 

### Solution
It seems clear to me that `MMC_Write_IT()` should not be run when writing is being done by `_DMA` version of functionality. I can't see anything being done wrong by micropython in the setup so far, so I'm going to test some patches to the HAL functions.

andrewleech, 2019-05-03:
The following change fixes the issue for us
```
stm32f7xx_hal_mmc.c @ -1422,14 +1422,14 @@ void HAL_MMC_IRQHandler(MMC_HandleTypeDef *hmmc)

    }
  }

 - else if(__HAL_MMC_GET_FLAG(hmmc, SDMMC_IT_TXFIFOHE) != RESET)
 + else if(__HAL_MMC_GET_FLAG(hmmc, SDMMC_IT_TXFIFOHE) != RESET && hmmc->Instance->MASK & SDMMC_IT_TXFIFOHE)
  {
    __HAL_MMC_CLEAR_FLAG(hmmc, SDMMC_FLAG_TXFIFOHE);

    MMC_Write_IT(hmmc);
  }

-  else if(__HAL_MMC_GET_FLAG(hmmc, SDMMC_IT_RXFIFOHF) != RESET)
+  else if(__HAL_MMC_GET_FLAG(hmmc, SDMMC_IT_RXFIFOHF) != RESET && hmmc->Instance->MASK & SDMMC_IT_RXFIFOHF)
  {
    __HAL_MMC_CLEAR_FLAG(hmmc, SDMMC_FLAG_RXFIFOHF);
```

dpgeorge, 2019-05-03:
Nice debugging, but I have a few questions/points:
- Why does the IRQ handler get to that TXFIFOHE line when doing DMA?  With DMA write the only IRQs enabled are `(SDMMC_IT_DCRCFAIL | SDMMC_IT_DTIMEOUT | SDMMC_IT_TXUNDERR)` which are all errors.  So there must be either an error with the write, or some other spurious IRQ??
- It looks like the SD driver has the same issue, and all MCU variants.  Interesting that this was never seen before... maybe related to the first point?
- Other code in that IRQ handler uses a test of the `Context` variable, so an alternative (but not necessarily better) fix is:
```c
__HAL_MMC_CLEAR_FLAG(hmmc, SDMMC_FLAG_TXFIFOHE);

if ((hmmc->Context & MMC_CONTEXT_IT) != RESET) {
    MMC_Write_IT(hmmc);
}
```

andrewleech, 2019-05-03:
Ah that's an interesting point, I didn't think to check exactly which interrupt is being fired.

Later on today I confirmed the problem has only occurred from the last rebase of our tree up master, our previous pin was before MMC support so I assumed the bug has crept in from a libstm32 update but didn't actually look for specific changes.

dpgeorge, 2019-05-17:
@andrewleech did you make any further progress investigating the root cause of this?

andrewleech, 2019-05-17:
Not yet, though I've literally got a breakpoint set in the right place to find out, just haven't triggered it yet this afternoon.

dpgeorge, 2019-05-17:
Ok thanks.  Just wondering if this can be fixed before #4784 (v1.11 release).  Even if it's not the stm32lib submodule still needs updating.

andrewleech, 2019-05-21:
The problem I'm seeing appears to be related to `MICROPY_HW_ENABLE_MMCARD` which I had set to `1`, in that it disappears when I set it to `0`.

How different is the mmc driver to the sd one?

With MMCARD enabled again I'm trying to determine what's firing the interrupt, but it appears nothing is... I didn't believe the debugger report of the peripheral values so added a couple of variables to check at the top of the interrupt handler:
``` c
/**
  * @brief  This function handles MMC card interrupt request.
  * @param  hmmc: Pointer to MMC handle
  * @retval None
  */
void HAL_MMC_IRQHandler(MMC_HandleTypeDef *hmmc)
{
  uint32_t errorstate = HAL_MMC_ERROR_NONE;

  volatile uint32_t flags = hmmc->Instance->STA;
  UNUSED(flags);
  volatile uint32_t enabled = hmmc->Instance->MASK;
  UNUSED(enabled);

  /* Check for SDMMC interrupt flags */
  if(__HAL_MMC_GET_FLAG(hmmc, SDMMC_IT_DATAEND) != RESET)
```
Assuming I've read the HAL code and the STM32F76xxx reference manual correctly, you should have to enable an interrupt in the `MASK` register before it'll trigger the interrupt. `MASK` is still zero at this point, but I'm in the interrupt. The flags register `STA` is 0x400 which I believe is just  
`SDMMC_STA_DBCKEND` : Data block sent/received (CRC check passed)
So nothing of any particular concern.

Just to be clear, this interrupt HAL function is being called via `SDMMC2_IRQHandler` (confirmed with debugger), we're using the 2nd interface on the chip and it appears to be functioning fine generally other than this issue.

andrewleech, 2019-05-23:
Ok, I think I've found the root cause... it's running too many interrupt handlers

andrewleech, 2019-05-23:
I still think the logic of the interrupt handlers is incorrect, it should not be handling a sdmmc flag that isn't enabled as an interrupt. I guess that's an issue to take up with stm though really!

dpgeorge, 2019-05-23:
> I still think the logic of the interrupt handlers is incorrect,

I agree.  But I'd rather not try to fix what's not technically broken, and this code (the SD card path) has been working for a long time.

Let's keep it in mind in case a similar bug pops up in the future, but for now I think this issue can be closed.

andrewleech, 2019-05-23:
Yes, that code path doesn't seem to get hit without the other issue present, so it's possibly safest to leave it as is, especially as it's in external code.

Thanks for raising the question about how it was getting there though, I would have just left my original hacky change in place and moved on!

closed 2019-05-23 by andrewleech