micropython/micropython #4742: nrf52832 does not respond to control-C. I can not stop my program

Afantor, 2019-05-02:
While testing the loop print string , I found that once I start my program running, I cannot use control-C to break out. 

Control-C does work under the REPL -- have not tried other sensors yet.

EPS32 -- control C works properly running the same code.

No exeption handling in the program.

my code :
`import machine
import time

machine.info()
while True:
  print('abc')
  time.sleep_ms(100)`

dpgeorge, 2019-05-02:
It looks like the nrf port doesn't capture ctrl-C at a UART REPL, only at a BLE REPL.

dpgeorge, 2019-05-03:
See #4745 for a fix for this.

Afantor, 2019-05-03:
sorry, I still didn't succeed. my board is feather52, and I need run bluetooth stack.
Flash board command is : make BOARD=feather52 FROZEN_MPY_DIR=freeze V=1 SD=s132 sd

and I check same loop print string, still did not succeed. Crtl - C still can not stop my program.

please check again,Thanks.

dpgeorge, 2019-05-03:
How are you accessing the REPL on the board, in other words how does the ctrl-C character get input to the board (eg via UART0)?

Afantor, 2019-05-04:
I tried minicom under Ubuntu 16.04 and it did not work for me. :-(
and tried putty again under windows 10 did not work to break(Ctrl-C).

Afantor, 2019-05-05:
I made a mistake, must be git branch to nrf-uart-nonblocking. Before check it in master branch.
Now, Ctrl + C was worked.  Thanks @dpgeorge

dpgeorge, 2019-05-22:
Ctrl-C support was implemented in 302ffdba7f06e8827532834b5902406cd7bf4ea0 and 50d5114fcd5f3b81d400c1297d0ce754c1dcd2e6

closed 2019-05-22 by dpgeorge