micropython/micropython #4744: SoftDevice update for nRF51?

thirtythreeforty, 2019-05-03:
I have a need for a Central Bluetooth device role for my nRF51-based WT51822-S4AT modules.  I see that the current implementation only uses SoftDevice S110 and thus is only compatible with the Peripheral role; however, [SoftDevice S130](https://www.nordicsemi.com/Software-and-Tools/Software/S130) is available and compatible with this SoC.

I'm willing to take a crack at getting S130 working on the nRF51, but before I do, I wanted to ask: has this been tried before?  Are there known issues as to why it hasn't already been done, or is it simply a lack of volunteer work?

If the latter, I'd appreciate pointers about getting started.  I'm familiar with embedded development, but only passingly familiar with the MicroPython project.

Thanks!

glennrub, 2019-05-03:
@thirtythreeforty , the quick answer is that there has been some work on it before. Removed by several reasons; not to much interest around s130 on nrf51, and API was kind of outdated (v.2.0.1). 

There are traces of the "support" not to far back in the history. Most of the s130 support can be seen in the squashed commit when the port was merged with mainline master (linker scripts/ble_drv.c etc). Just search for s130 or 130: https://github.com/micropython/micropython/commit/e22b94350889af32527e4ebf3a8323e4eba9280e#diff-ee59b54ccf29ec7974bda2e27b863fa6 

The limited support back in the days was targeting the ubluepy API for micropython (note that s130 was never well maintained). The future the ubluepy module is that it will be replaced by the work in #4589 in time. My recommendation is that you put your effort into supporting s130 by contributing to the work in this PR. Possible using the old commits to get some help on how to integrate s130. 

If contributing to the new API is out of question, i do plan to keep a copy of ubluepy for my own scripts as an external c-module to micropython. But, i'm not planning to maintain this much, more to keep it for not loosing it. https://github.com/glennrub/ubluepy

If you need a few more pointers please let me know. I just wanted to reply a bit short telling that there might be something to start looking through, to see if its worth anything :)

thirtythreeforty, 2019-05-04:
@glennrub, awesome thank you.  I decided to base my work on that PR you mentioned.

I have some initial work in [my fork's nrf51-s130](https://github.com/thirtythreeforty/micropython/tree/nrf51-s130) branch.  It is based on the latest revision @aykevl's PR, but it will probably rebase fine against any future versions.

This is completely untested!  However, it does build cleanly (which obviously means it is bug-free and ready for production).  I feel weird opening a PR since the work it's based on isn't merged, although I can if you want.  Regardless, comments are welcome.

thirtythreeforty, 2019-05-04:
I don't have a development board for my little modules (or the nRF51) yet -- they're SMT and I will need to fab a breakout board.  So it may be some time before I can really make some tested progress on this.

aykevl, 2019-05-05:
Yes, s130 should be possible. However, note that the current PR only supports peripheral mode, not central mode.

Just one warning: the nrf51822 has very little RAM and is barely usable together with a SoftDevice. In my experience, you need about 3kB of stack to avoid stack overflows leaving just about 4kB for the heap. You may want to look for one with 32kB of RAM or even better a nrf52832 which is actually usable. They're a bit more expensive but well worth it (says someone who always tries the cheapest boards, including the wt51822-s4at). I have good experience with the GT832E-01.

thirtythreeforty, 2019-05-06:
Do you have any plans about Central mode?

That amount of RAM sounds okay for my purpose actually: I am simply building a keyfob.  It only has to poke a characteristic on a BLE-enabled gate.  It is the epitome of this relevant xkcd:
[image]

jonnor, 2024-09-19:
Is this relevant for latest MicroPython?

closed 2025-03-03 by jonnor