micropython/micropython #4747: Problems building for esp8266

brzez, 2019-05-03:
I'm trying to build micropython for esp8266.
I wrote a micropython framework for mqtt+http config+network persistency, and currently it eats a lot of RAM, so I'm trying to build it into modules, so it runs from flash.

Currently, I have 2 problems:

1. When building, using ```make``` I don't get modules, which I need (eg. ```uasyncio utimeq```)
I would like to include all the modules, which are also avaliable in regular micropython build for esp8266. Is there a list somewhere, or a makefile?

2. My module didn't fit in the build size(?). 

I've tried #2700
esp8266.ld - ```irom0_0_seg :  org = 0x40209000, len = 0xa7000```

But I get:
```
FAT filesystem appears to be corrupted. If you had important data there, you
may want to make a flash snapshot to try to recover it. Otherwise, perform
factory reprogramming of MicroPython firmware (completely erase flash, followed
by firmware programming).
```
Am I doing something wrong? Should building a .bin solve the low RAM (on import) issue? or am I better of just rewriting the framework with less features?

btw. Micropython is amazing. I've been toying with it since I've learned about it, already have a 3D printer IoT power/led controller and remote humidity sensor which controls humidity via mqtt.

dpgeorge, 2019-05-03:
> I would like to include all the modules, which are also avaliable in regular micropython build for esp8266. Is there a list somewhere, or a makefile?

Have a look at the `esp-extra-scripts` branch, which contains symlinks for these extra modules.

> Am I doing something wrong?

Everything is correct, just do a `make erase` first before deploying the firmware.  That will reset the filesystem.

Or at the REPL you can format the filesystem with `os.VfsFat.mkfs(bdev)`

brzez, 2019-05-04:
Thanks for the help

It looks like I was just using an old, outdated docker image for building. I wrote my own docker image, and now the build runs fine. 

It looks like I've saved ~8k of RAM which is very nice.

If needed - I can share the dockerfile for building the project.

closed 2019-05-04 by brzez

dpgeorge, 2019-05-05:
> If needed - I can share the dockerfile for building the project.

If you found a docker file that was out of date you can ask the maintainer/host of it to update it.

brzez, 2019-05-05:
The dockerfile I've used was https://hub.docker.com/r/mattyt/esp8266-micropython-build

I've tried, and I'm not sure if it's possible to message an user on dockerhub.

dpgeorge, 2019-05-06:
Ping @mattytrentini , re dockerfile perhaps needing updating

mattytrentini, 2019-05-06:
Thanks for the heads-up @dpgeorge, I'll update the image when time allows. 

Actually, I'd like to change the way the container operates - and @brzez this may be useful to you too. I I belibe that the container should actually contain only the dependencies required to build the system. The code should be shared into the container from the host. This will mean the container needs updating much less often and is more consistent with common practices.

brzez, 2019-05-07:
@mattytrentini 
I've created a dockerfile, which just sets up the build environment.
Then I have a volume with my code mounted to ```ports/esp8266/modules/MY_MODULE```.
I also have a script, which pulls micropython and micropython-lib, builds micropython-lib, copies the libs I've specified to ```ports/esp8266/modules``` and finally builds the .bin file.

mattytrentini, 2019-05-08:
That sounds about perfect! Is the container available?

We should probably build up a collection of these build containers for each of the ports...

brzez, 2019-05-11:
@mattytrentini 
Ok, I pushed initial version of the build thing.
https://github.com/brzez/micropython-esp8266-buildtools

You can configure it inside docker-compose.yml
It allows adding custom libs from micropython-lib via `MICROPYTHON_LIB` (space delimited list of libraries). Your code is added from `./my_module` dir - this can be also changed inside docker-compose.yml