micropython/micropython #4748: [ESP32] DHCP Hostname not working correctly

milkshakespoon, 2019-05-04:
I can't put my finger on which version this stopped working or if it's just my code that's the issue.

```
def ConnectNetwork(NetworkName, Password):
    import network

    wlan = network.WLAN()
    wlan.active(True)
    wlan.config(dhcp_hostname="testname")
    wlan.ifconfig(('192.168.0.210', '255.255.255.0', '192.168.0.1', '8.8.8.8'))

    if not wlan.isconnected():
        print('connecting to network...')
        wlan.connect(NetworkName, Password)

    print('Network Config:', wlan.ifconfig())
    print('DHCP Hostname:', wlan.config('dhcp_hostname'))

ConnectNetwork("SSID", "Password")
```

When the device is connected to my router it is showing as UNKNOWN name. I am unable to ping the device using its name. Sometimes the name does show up on the router but disappears quickly after. 

I have tested this in another home with a different board (same model) and a different router/network. Same things seem to happen.

This is part of some larger code, but pulled this snippet out for testing this issue. I have attempted to use different MicroPython versions below...

- esp32-bluetooth.bin (latest)
- esp32-20190504-v1.10-323-g906fb89fd.bin
- esp32-20180511-v1.9.4.bin

I have also tried removing this line but the same issue happens.
`wlan.ifconfig(('192.168.0.210', '255.255.255.0', '192.168.0.1', '8.8.8.8'))`

Thanks in advance!

dpgeorge, 2019-05-05:
If you set the IP settings manually using `wlan.ifconfig(...)` then DHCP won't be used and so there's no way for the esp32 device to inform the router about its name.

>  I am unable to ping the device using its name

Unless mDNS is active I don't see how it's possible to ping the device using its name.  Did this ever work for you?  What name did you pnig?

I tried using `dhcp_hostname` with latest master and it works ok for me: the router correctly reports the esp32's IP address and hostname.

milkshakespoon, 2019-05-06:
It may well be that I've misunderstood this and that it does work. I have been able to ping the name a couple of times and I'm not using mDNS as far as I'm aware. 

I'd appreciate it very much if you could post some code I can test?

Apologies as I am still learning all about this!

Thank you!!

dpgeorge, 2019-05-06:
>  I have been able to ping the name a couple of times

What is the exact name that you ping?

> I'd appreciate it very much if you could post some code I can test?

The code is simple: just connect to wifi as usual, eg:
```python
import network
wl = network.WLAN()
wl.active(1)
wl.connect(SSID, PASSWORD)
```
Then on your router you should see the name "espressif" corresponding to the IP address it was assigned.

milkshakespoon, 2019-05-06:
Hello, 

Thanks for your help. Using your code I was able to replicate the issue I'm having.

Pinging from my laptop: 

[image]

For a few seconds I see the device connected to my router and showing the correct name. After a few seconds the device will disappear from the router and come back as UNKNOWN name. I have tried this at a friends on the same model device also and the same thing happened.

For the record the device is a WEMOS LOLIN32.

I'm sure I had this working fine before with not disconnection... Do you have anything else I can try to provide you with more information?

dpgeorge, 2019-05-07:
If you can ping `espressif.Home` then it's most likely your router that is providing a local DNS service for your home (W)LAN.  And instead of pinging `espressif` explicitly ping the full name `espressif.Home`.  So in this case there's nothing to fix on the ESP32 side of things, rather it's the configuration of the router.

milkshakespoon, 2019-05-07:
Hello,

Yes you're right, my router provides my DNS for my WLAN. The .home is just a DNS suffix I have and I don't think that should make a difference. Think I've proved this here...

[image]

Something is making the device disconnect after a few seconds causing... Like I say previously I was able to replicate this issue at a friends house using a different router and a different board but the same model. I had this working previously.

Thanks for your help!

dpgeorge, 2019-05-08:
> Something is making the device disconnect after a few seconds

In that case you should see a log on the ESP32's REPL, and a reason for the disconnect.

nedoskiv, 2019-05-08:
I got some different network issue, but tested only with 2 kind of routers, issue is this:

my esp32 connect normal to wifi, it connection with the router is flawless, but sometimes other devices connected on router lost ping/arp on it. Router ping works all the time, but when arp table of other devices was cleared they need some time until discover it (few minutes). That is caused only when is used by AP/LAN router bridge, if i made them in another network and traffic is routed by the router it works fine.

Tested with mikrotik and an DSL modem.

milkshakespoon, 2019-05-08:
> > Something is making the device disconnect after a few seconds
> 
> In that case you should see a log on the ESP32's REPL, and a reason for the disconnect.

Hmm it is strange. I don't think a full disconnection happens hence I can still ping via IP. Although the DNS name seems to disappear. 

Going back to what you said about a static IP meaning the DNS name wont be registered is that true? I have servers that have static IPs and the hostname of those are registered on my router. No other devices seem to drop off my DNS...

Thanks!

peterhinch, 2019-05-09:
Perhaps one of you guys might explain to someone with little networking expertise. I never see names for ESP8266 or ESP32 devices on my network and access them by IP. I thought you needed an mDNS service running on the device for named access. What is the mechanism by which you might expect this to work?

FWIW my router is a fairly professional one: a Draytek Vigor.

milkshakespoon, 2019-05-13:
> Perhaps one of you guys might explain to someone with little networking expertise. I never see names for ESP8266 or ESP32 devices on my network and access them by IP. I thought you needed an mDNS [...]

The way I understood this to work like any other device connecting to your network a hostname would be registered with the router which is used in DNS.

If this isn't the way it works I'd love someone to correct me. And if not maybe it should be?

milkshakespoon, 2019-06-02:
Would like to see if anyone has any ideas on this still... been doing a lot of testing and packet analyzing but not able to find the issue. 

When booting I see this...

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:4928
ho 0 tail 12 room 4
load:0x40078000,len:9332
load:0x40080400,len:6216
entry 0x400806e8
I (442) cpu_start: Pro cpu up.
I (442) cpu_start: Application information:
I (442) cpu_start: Compile time:     00:40:19
I (445) cpu_start: Compile date:     May  4 2019
I (450) cpu_start: ESP-IDF:          v3.3-beta1-268-g5c88c5996
I (456) cpu_start: Starting app cpu, entry point is 0x40082950
I (0) cpu_start: App cpu up.
I (467) heap_init: Initializing. RAM available for dynamic allocation:
I (474) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (480) heap_init: At 3FFB9B88 len 00026478 (153 KiB): DRAM
I (486) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (492) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (499) heap_init: At 40093434 len 0000CBCC (50 KiB): IRAM
I (505) cpu_start: Pro cpu start user code
I (76) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (50) modsocket: Initializing
I (140) wifi: wifi driver task: 3ffe3268, prio:23, stack:3584, core=0
I (368) wifi: wifi firmware version: 38e2484
I (368) wifi: config NVS flash: enabled
I (368) wifi: config nano formating: disabled
I (368) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (378) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (418) wifi: Init dynamic tx buffer num: 32
I (418) wifi: Init data frame dynamic rx buffer num: 32
I (418) wifi: Init management frame dynamic rx buffer num: 32
I (418) wifi: Init static rx buffer size: 1600
I (428) wifi: Init static rx buffer num: 10
I (428) wifi: Init dynamic rx buffer num: 32
Started webrepl in normal mode
MicroPython v1.10-323-g906fb89fd on 2019-05-04; ESP32 module with ESP32
```

I'm intrigued if these two lines have anything to do with the issues...
```
I (368) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (378) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
```

Thank you!

milkshakespoon, 2019-06-04:
I just flashed the newest version of MicroPython in hope this would be fixed. Just wanted to add that it does not seem to be. Still getting the same issues on esp32-20190604-v1.11-25-gce8262a16

Thanks

milkshakespoon, 2019-06-07:
Just some further info on this...

By running these in the REPL I am able to get the device to register a hostname after I have already connected to my WiFi and lost the hostname after a few seconds.
```
sta_if = network.WLAN(network.STA_IF)
sta_if.disconnect()
sta_if.connect()
```
[image]

After a few seconds the hostname disappears and I am no longer able to access through any device. I can still access via the IP.

[image]

street-grease-coder, 2021-01-10:
Having the same issue, described [>here<] https://github.com/micropython/micropython/pull/4951#issuecomment-757511852) with some info that might help to debug this easier