micropython/micropython #4754: I2C communication, restart system over time.

nedoskiv, 2019-05-06:
Well I hardly can reproduce that issue, but at  least let me explain.
In my project (home security) I use esp32, I got started 4 threads:
one is communicating over SPI with MFRC 522
another is listening for incoming TCP connections
other is reading ADC to determina if there is alarms
one is remotelly logging data.
and main loop which watch over _threads and restart system if any of them fails, also read some data over i2c every 2 seconds.

Device was working fine before I add i2c part, after that it restart without any trace after few hours of working, all my attempts to debug was failing. I started it recentrly without connecting i2c bus and noticed it restart in shorted period of time 20-40 min.
When I remove i2c part it works without a problem. Here is a part of my code:

```
i={}
i["i2cd"]=bytearray(20)
def ir2(d,r):		# read 2 patameters from i2c device (d),address (r)  with add checksum.   "," is the delimiter, return -1 on first var if error
	global i
	try:
		i["i2c"].readfrom_mem_into(d, r, i["i2cd"])
	except:
		return -1,0	# i2c comminication error
	t=0
	t1=""
	while True:
		try:
			if i["i2cd"][t]==255:
				break
			else:
				t1=t1+str(chr(i["i2cd"][t]))
				t+=1
		except:
			break
	t2=t1.split(",")
	try:
		t=int(t2[0])
		t1=int(t2[1])
		t2=int(t2[2])
	except:
		return -2,0		# parse error
	if (t2 != (t1+t)):	# checksum error
		return -3,0
	return t,t1
```
And the initialization part/reading part:

```
i["i2c"] = machine.I2C(scl=machine.Pin(22), sda=machine.Pin(21),freq=10000)

i["bs"],z["bv"]=ir2(41,0x10)	# read battery status, handled in rlog0
z["ev"],z["2r"]=ir2(41,0x11)

```

Hope you got the idea of how I use it, one time I managed to catch the error via serial monitor:

```
Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception)
Debug exception reason: BREAK instr
Core 0 register dump:
PC      : 0x400803c0  PS      : 0x00060836  A0      : 0x00000000  A1      : 0x3ffc7de0
A2      : 0x00000000  A3      : 0x00000001  A4      : 0x8008b1cc  A5      : 0x3ffd5fa0
A6      : 0x00000003  A7      : 0x00060023  A8      : 0x00060023  A9      : 0x3ffd6060
A10     : 0x3ffb09e8  A11     : 0xffffffff  A12     : 0x8008b1cc  A13     : 0x3ffd5f80
A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x00000016  EXCCAUSE: 0x00000001
EXCVADDR: 0xffffffd0  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff

Backtrace: 0x400803c0:0x3ffc7de0 0x7ffffffd:0x3ffc79e0 0x400e92da:0x3ffc7a00 0x400e91c3:0x3ffc7a30 0x400df848:0x3ffc7ad0 0x400dbb25:0x3ffc7b00 0x400e8c69:0x3ffc7b20 0x400df848:0x3ffc7bc0 0x400dbb25:0x3ffc7c40 0x400dbb52:0x3ffc7c60 0x400f6e76:0x3ffc7c80 0x400f70d9:0x3ffb1ade

Rebooting...
ets Jun  8 2016 00:22:57
```

I do not know what cause it, one is sure when there is lot of I2C errors, system restart often.

dpgeorge, 2019-05-06:
Do you have the exact firmware binary that was used when the crash occurred, the one reported above?  If so please provide it so the backtrace can be decoded.

nedoskiv, 2019-05-06:
[esp32-20190125-v1.10_work_fine!.zip](https://github.com/micropython/micropython/files/3148473/esp32-20190125-v1.10_work_fine.zip)

dpgeorge, 2019-05-07:
Thanks, but actually I'd need the corresponding .elf file, it's too hard to reverse engineer the .bin file.  Do you have it?

nedoskiv, 2019-05-07:
Nope, i never see an elf file when download micropython for esp32 there is only .bin
If you can provide me an binary file of lastest build and keep .elf file on your side, I probabbly gonna reproduce the issue after few hours.

dpgeorge, 2019-05-08:
> Nope, i never see an elf file when download micropython for esp32 there is only .bin

The elf files live alongside the bin files, so you can download them just by changing the filename.

I see you're using official releases so I could get the correct elf file and disassemble it.  The backtrace is:
```
400803c0 <_DoubleExceptionVector>:
400803c0:  **   004140          break   1, 4

400e92d0 <mp_decode_uint_value>:
400e92d0:       006136          entry   a1, 48
400e92d3:       01ad            mov.n   a10, a1
400e92d5:       0129            s32i.n  a2, a1, 0
400e92d7:       9cd881          l32r    a8, 400d0638 <_flash_cache_start+0x620> 0x40184468=mp_decode_uint
400e92da:  **   0008e0          callx8  a8
400e92dd:       0a2d            mov.n   a2, a10
400e92df:       f01d            retw.n
400e92e1:       000000          ill

...
```

So it's crashing when it tries to call a function (`mp_decode_uint`) but since there's nothing wrong with calling this function the problem most likely lies in a stack overflow.

In fact, looking at the stack values at the start and end of the backtrace it uses over 90k of stack, which is way over the limit.

By default threads created with `_thread.start_new_thread()` have 5k of stack space (the main thread has 16k).  I would try increasing that by using `_thread.stack_size(8 * 1024)` before calling `_thread.start_new_thread()`.

nedoskiv, 2019-05-08:
thank you, not sure what's go wrong, i use board without SPIRAM so my memory is very limited. 90K is 3/4 of total free heap when device reboot.
just a few questions about threads, I guess when I create new theread memory for stack is reserved permanently until thread is closed?
And what about other memory that it need to process variables, read files etc, how is it provided to the _thread?

closed 2019-05-08 by nedoskiv

dpgeorge, 2019-05-08:
>  I guess when I create new theread memory for stack is reserved permanently until thread is closed?

Yes.  But it comes from the FreeRTOS heap, not the MicroPython heap.

> And what about other memory that it need to process variables, read files etc, how is it provided to the _thread?

That comes from the MicroPython heap and is shared among all threads.

dpgeorge, 2019-05-08:
>  found nothing about how to view FreeRTOS available heap from micropython

It's not currently possible to do that.

> Is it possible my code somehow to generate memory leak into freeRTOS heap, in this case used for stack?

Unlikely.  If you just create a fixed number of threads at start-up that is the only time it uses FreeRTOS heap memory (well, not really, the ESP O/S may at anytime allocate on the FreeRTOS heap, but I think an error would be raised if the heap memory was exhausted).

But you should really try increasing the stack size.