micropython/micropython #4755: raspberry pi fail to build micropython

zazinio, 2019-05-06:
hi I'm noob and I'm using rspberry pi b3+ V1.3
and rasbian version
`pi@raspberrypi:~ $ lsb_release -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 9.9 (stretch)
Release:	9.9
Codename:	stretch`
when I try to install micropython build crash with error
`pi@raspberrypi:~ $ git clone https://github.com/micropython/micropython.git
Клониране и създаване на хранилище в „micropython“…
remote: Enumerating objects: 40, done.
remote: Counting objects: 100% (40/40), done.
remote: Compressing objects: 100% (27/27), done.
remote: Total 69980 (delta 17), reused 18 (delta 12), pack-reused 69940
Получаване на обекти: 100% (69980/69980), 39.39 MiB | 3.76 MiB/s, действието завърши.
Откриване на съответните разлики: 100% (50525/50525), действието завърши.
pi@raspberrypi:~ $ cd micropython
pi@raspberrypi:~/micropython $ git pull
Already up-to-date.
pi@raspberrypi:~/micropython $ cd ports
pi@raspberrypi:~/micropython/ports $ cd unix
pi@raspberrypi:~/micropython/ports/unix $ make clean
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
rm -f micropython
rm -f micropython.map
rm -rf build 
pi@raspberrypi:~/micropython/ports/unix $ make axtls
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
You cloned without --recursive, fetching submodules for you.
(cd ../..; git submodule update --init --recursive)
Регистриран е подмодул „lib/axtls“ (https://github.com/pfalcon/axtls) за пътя към подмодул „lib/axtls“
Регистриран е подмодул „lib/berkeley-db-1.xx“ (https://github.com/pfalcon/berkeley-db-1.xx) за пътя към подмодул „lib/berkeley-db-1.xx“
Регистриран е подмодул „lib/libffi“ (https://github.com/atgreen/libffi) за пътя към подмодул „lib/libffi“
Регистриран е подмодул „lib/lwip“ (https://git.savannah.gnu.org/r/lwip.git) за пътя към подмодул „lib/lwip“
Регистриран е подмодул „lib/nrfx“ (https://github.com/NordicSemiconductor/nrfx.git) за пътя към подмодул „lib/nrfx“
Регистриран е подмодул „lib/stm32lib“ (https://github.com/micropython/stm32lib) за пътя към подмодул „lib/stm32lib“
Клониране и създаване на хранилище в „/home/pi/micropython/lib/axtls“…
Клониране и създаване на хранилище в „/home/pi/micropython/lib/berkeley-db-1.xx“…
Клониране и създаване на хранилище в „/home/pi/micropython/lib/libffi“…
Клониране и създаване на хранилище в „/home/pi/micropython/lib/lwip“…
Клониране и създаване на хранилище в „/home/pi/micropython/lib/nrfx“…
Клониране и създаване на хранилище в „/home/pi/micropython/lib/stm32lib“…
Път към подмодул „lib/axtls“: изтеглена е версия „43a6e6bd3bbc03dc501e16b89fba0ef042ed3ea0“
Път към подмодул „lib/berkeley-db-1.xx“: изтеглена е версия „35aaec4418ad78628a3b935885dd189d41ce779b“
Път към подмодул „lib/libffi“: изтеглена е версия „e9de7e35f2339598b16cbb375f9992643ed81209“
Път към подмодул „lib/lwip“: изтеглена е версия „92f23d6ca0971a32f2085b9480e738d34174417b“
Път към подмодул „lib/nrfx“: изтеглена е версия „d4ebe15f58de1442e3eed93b40d13930e7785903“
Път към подмодул „lib/stm32lib“: изтеглена е версия „a5e93de18479879dd129cf6272cfd75e0c794bd4“
pi@raspberrypi:~/micropython/ports/unix $ make
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
mkdir -p build/genhdr
GEN build/genhdr/mpversion.h
GEN build/genhdr/moduledefs.h
GEN build/genhdr/qstr.i.last
GEN build/genhdr/qstr.split
GEN build/genhdr/qstrdefs.collected.h
QSTR updated
GEN build/genhdr/qstrdefs.generated.h
mkdir -p build/build/
mkdir -p build/extmod/
mkdir -p build/lib/axtls/crypto/
mkdir -p build/lib/axtls/ssl/
mkdir -p build/lib/berkeley-db-1.xx/btree/
mkdir -p build/lib/berkeley-db-1.xx/mpool/
mkdir -p build/lib/embed/
mkdir -p build/lib/mp-readline/
mkdir -p build/lib/oofatfs/
mkdir -p build/lib/timeutils/
mkdir -p build/lib/utils/
[... 259 lines cut ...]
CC ../py/modmath.c
CC ../py/modcmath.c
CC ../py/modmicropython.c
CC ../py/modstruct.c
CC ../py/modsys.c
CC ../py/moduerrno.c
CC ../py/modthread.c
CC ../py/vm.c
CC ../py/bc.c
CC ../py/showbc.c
CC ../py/repl.c
CC ../py/smallint.c
CC ../py/frozenmod.c
CC main.c
CC gccollect.c
LINK mpy-cross
collect2: fatal error: ld terminated with signal 11 [Нарушение на разделянето(segfault)]
compilation terminated.
../py/mkrules.mk:132: recipe for target 'mpy-cross' failed
make[1]: *** [mpy-cross] Error 1
make[1]: *** Deleting file 'mpy-cross'
make[1]: Leaving directory '/home/pi/micropython/mpy-cross'
../../py/mkrules.mk:108: recipe for target '../../mpy-cross/mpy-cross' failed
make: *** [../../mpy-cross/mpy-cross] Error 2
pi@raspberrypi:~/micropython/ports/unix $ `

dhylands, 2019-05-06:
This type of issue is better dicussed on the forum: https://forum.micropython.org/

I tried this and it worked fine for me. I use a 3B+ and the 2019-04-08-raspbian-stretch-lite.zip version of raspbian on an 8 Gb SDCard.

After flasing and booting up, I did the following commands:
```
sudo apt update
sudo apt upgrade
sudo apt install git
sudo apt install libffi-dev
git clone https://github.com/micropython/micropython
cd micropython/ports/unix/
git submodule update --init
make
```
And it ran:
```
pi@gateway:~/micropython/ports/unix$ ./micropython 
MicroPython v1.10-326-g089c9b71d on 2019-05-06; linux version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> 
```

How much free space do you have on your sdcard?
Do you have a good power suppply?
Perhaps try a different sdcard.

zazinio, 2019-05-07:
Sorry if I am in the wrong place
I am noob but fixed the problem by mistake.
Did not conect to wifi and build succeeded.

"How much free space do you have on your sdcard?
Do you have a good power suppply?
Perhaps try a different sdcard." 
It's 32GB sdcard with fresh install so it has free space.
Power supply and sd card are ok

closed 2019-05-12 by zazinio