micropython/micropython #4758: pwm.deinit() can not close Pin if Duty Cycle set to 1023

hassankamran012, 2019-05-07:
Hi I was using the latest nightly built for esp8266 that is esp8266-20190430-v1.10-304-g8031b7a25. I setup GPIO Pin 5 as a PWM pin and set the frequency to 1000. the deinit() method works fine for all the other duty cycles except when the duty cycle is set to max 1023.

Code 
from machine import Pin, PWM
pwm = PWM(Pin(5))
pwm.freq(1000)
pwm.duty(1023)
pwm.deinit()##Does not work in this case

dpgeorge, 2019-05-07:
> the deinit() method works fine for all the other duty cycles except when the duty cycle is set to max 1023

What exactly happens to the pin when `deinit()` is called with duty set to 1023?  With maximum duty the pin should always be on (while PWM is active).

hassankamran012, 2019-05-08:
If I set the pwm to 1023 I can't turn it off with deinit() anymore. If it's any other pwm value it does work as expected by there this unintuitive behaviour at 1023.

dpgeorge, 2019-05-09:
> If I set the pwm to 1023 I can't turn it off with deinit() anymore.

It works ok for me.  Eg:
```python
p5 = machine.Pin(5)
pwm = machine.PWM(p5)
pwm.duty(1023)
pwm.deinit()
p5.init(p5.OUT) # make pin 5 an output pin
p5(0) # set pin 5 low
p5(1) # set pin 5 high
```

hassankamran012, 2019-05-13:
Maybe it's something wrong on my end. I will re-flash my ESP8266 and try again. I'll give you an update soon.

hassankamran012, 2019-05-26:
I tried it out again. There was no issue this time. I was using a nightly at the time that I posted it. Sorry for the delay I had exams.

closed 2019-05-26 by hassankamran012