micropython/micropython #4761: USB cdc->tx_buf_ptr_in & cdc->tx_buf_ptr_out error

dobotopensource, 2019-05-08:
If I re plug in the usb, in & out ptr will change like this blow

> 0> [INFO]usbd_cdc_interface.c:usbd_cdc_tx_ready:191: cdc->tx_buf_ptr_in is 88
>  0> [INFO]usbd_cdc_interface.c:usbd_cdc_tx_ready:192: cdc->tx_buf_ptr_out is 82
>  0> [INFO]usbd_cdc_interface.c:usbd_cdc_tx_ready:193: buffer data len is 6
>  0> [INFO]Dobot/command/command.c:ExecuteInstantCmd:1139: Message to ExtC:240
>  0> [INFO]Dobot/command/command.c:OnQueuedCmdStartExec:854: 
>  0> 0xAA 0xAA 0x02 0xF0 0x11 0xFF 
>  0> [INFO]usbd_cdc_interface.c:usbd_cdc_tx_ready:191: cdc->tx_buf_ptr_in is 94
>  0> [INFO]usbd_cdc_interface.c:usbd_cdc_tx_ready:192: cdc->tx_buf_ptr_out is 0
>  0> [INFO]usbd_cdc_interface.c:usbd_cdc_tx_ready:193: buffer data len is 94
>  0> [INFO]Dobot/command/command.c:ExecuteInstantCmd:1139: Message to ExtC:240
>  0> [INFO]Dobot/command/command.c:OnQueuedCmdStartExec:854: 
>  0> 0xAA 0xAA 0x02 0xF0 0x11 0xFF 

tx_buf_ptr_in changes from 88 to 94, and tx_buf_ptr_out changes from 82 to 0

dpgeorge, 2019-05-08:
When the USB is re-plugged in, `tx_buf_ptr_out` and `tx_buf_ptr_out_shadow` are reset to 0, but `tx_buf_ptr_in` is not.  This is to allow output data to be buffered before the USB is enabled.

So the issue reported here is a side effect of that behaviour.  It will retransmit some of the old data in the buffer when USB is reconnected.

It's relatively easy to fix: just remove the 2 lines in `usbd_cdc_init()` which clear those 2 variables.  I guess it makes sense to do this, since it's not possible to know (as the receiver) if the data coming in over USB is retransmitted, or new data.

dobotopensource, 2019-05-08:
I have find the solution, you need to add this line 
`cdc->tx_buf_ptr_in = 0;`
The whole code likes this blow
```
uint8_t *usbd_cdc_init(usbd_cdc_state_t *cdc_in) {
    usbd_cdc_itf_t *cdc = (usbd_cdc_itf_t*)cdc_in;

    // Reset all the CDC state
    // Note: we don't reset tx_buf_ptr_in in order to allow the output buffer to
    // be filled (by usbd_cdc_tx_always) before the USB device is connected.
    cdc->rx_buf_put = 0;
    cdc->rx_buf_get = 0;
    cdc->rx_buf_full = false;
    cdc->tx_buf_ptr_out = 0;
    // add 
    cdc->tx_buf_ptr_in = 0;
    cdc->tx_buf_ptr_out_shadow = 0;
    cdc->tx_need_empty_packet = 0;
    cdc->connect_state = USBD_CDC_CONNECT_STATE_DISCONNECTED;
    #if MICROPY_HW_USB_ENABLE_CDC2
    cdc->attached_to_repl = &cdc->base == cdc->base.usbd->cdc;
    #else
    cdc->attached_to_repl = 1;
    #endif

    // PrintInfo("cdc->tx_buf_ptr_in is %d", cdc->tx_buf_ptr_in);
    // PrintInfo("cdc->tx_buf_ptr_out is %d", cdc->tx_buf_ptr_out);

    // Return the buffer to place the first USB OUT packet
    return cdc->rx_packet_buf;
}
```

dobotopensource, 2019-05-08:
Thank you for your answer @dpgeorge

dpgeorge, 2019-05-08:
I'll reopen this issue because this behaviour should be fixed.

dpgeorge, 2019-05-08:
Fixed as discussed above in b8c74014e42331afca3db5911b30032a109f73f1

closed 2019-05-08 by dpgeorge