micropython/micropython #4762: USB cdc->connect_state != USBD_CDC_CONNECT_STATE_DISCONNECTED error

dobotopensource, 2019-05-08:
I use the xcom v2.0 terminal tool to communicate with the usb vcom, if I don't check the DTR radio button, the condition blow is satisfied, and the terminal tool can't receive any data.
`cdc->connect_state != USBD_CDC_CONNECT_STATE_DISCONNECTED`
If I checked the DTR button, the function blow will be called, and the cdc->connect_state will change into USBD_CDC_CONNECT_STATE_CONNECTED state
`void HAL_PCD_SOFCallback(PCD_HandleTypeDef *hpcd)`

I have tried the stm32cube for usb vcom, everything is normal.

dpgeorge, 2019-05-08:
@dobotopensource I can confirm the behaviour, I have experienced it before.  But do you consider this a bug?

It's a feature that the device can detect whether the serial port is open via `pyb.USB_VCP().isconnected()`.  Just because the USB is plugged in to a host PC doesn't mean that anything is listening on the other end of the VCP interface.

Well, there might be other ways to detect connection, eg via the `CDC_SET_LINE_CODING` event, which would work without DTR selected.

dobotopensource, 2019-05-08:
Now I will change the connect state when the USB received something to work around it ...
```
void usbd_cdc_tx_always(usbd_cdc_itf_t *cdc, const uint8_t *buf, uint32_t len) {
    // for work around 
    if (data_received){
        data_received = 0;
        cdc->connect_state = USBD_CDC_CONNECT_STATE_CONNECTED;
    }
    ...............   
}
```

dpgeorge, 2019-08-19:
Closing as "not a bug", as discussed above.

closed 2019-08-19 by dpgeorge