micropython/micropython #4769: FOTA upgrade for stm32 or arm devices via various channels like wifi,BLE etc

amrithmmh, 2019-05-09:
Will there be an FOTA upgrade feature for arm cortex microcontrollers connected to WIFI or BLE or other ways

dpgeorge, 2019-05-09:
For stm32 there is the custom mboot bootloader which supports reflashing the main firmware automatically from a file on the filesystem.  So this can be used to do OTA updates: download/copy/etc the new firmware to a file and then call into mboot to do the update

A utility script is provided to show how this works, see `ports/stm32/mboot/fwupdate.py`.  It allows to update the application firmware from a `.dfu.gz` file: `update_mpy(filename, fs_base, fs_len)`, where `fs_base` is the address of the start of the (FAT) filesystem, and `fs_len` is its byte length.

More examples how to use this will come in time.

jonnor, 2025-11-09:
This appears to have been answered. For reference, MBOOT is described here. https://github.com/micropython/micropython/tree/master/ports/stm32/mboot

closed 2025-11-09 by jonnor