micropython/micropython #4774: STM32H7 CAN FD partial and full support

TravisJoe, 2019-05-10:
Per a discussion with @dpgeorge CAN is not currently supported on H7 MCU. 

I am hoping to see CAN support on the H7 device, and eventually CAN-FD support. I think FD (Flexible Data-Rate) is still new enough that the need for full support is definitely not a requirement, basic support of 11bit and 29bit CAN would be more than enough to allow the H7 to replace the F7 in most use-cases. 

Unfortunately my time and skill on this is limited, and willing to provide financial support if needed, but this a gating item for me to switch over to the H7 and I image other might want/need this as well.

iabdalkader, 2019-05-12:
Just thought I should mention that I'll start working the H7 CAN driver.

TravisJoe, 2019-05-13:
This is also related to a conversation on OpenMV
http://forums.openmv.io/viewtopic.php?f=6&t=1301

TravisJoe, 2019-05-13:
Also related to
https://github.com/openmv/openmv/issues/526

Hello @iabdalkader @kwagyeman
Just tell me if there is something I might be able to help with.

iabdalkader, 2019-05-15:
@dpgeorge Do you prefer this implemented in a new driver (fdcan.c) ? There will be a lot of #ifdefs otherwise.

TravisJoe, 2019-05-15:
My less valuable thoughts, 
I would agree its own driver would make sense. In my head relating it like SPI to QSPI in that implementation is not coupled enough to justify the mess. Especially since CANFD would be coming on other ports eventually making it even more complex if all together.

dpgeorge, 2019-09-23:
FDCAN support on H7 was added in f7a07b36057d61264958e3577f7e4555e77d520c

closed 2019-09-23 by dpgeorge