micropython/micropython #4775: Micropython file truncate doesn't exist

morrison93, 2019-05-10:
Is there any alternative in MicroPython to truncate a file at the current seek position? I don't want to just empty the file, I want to truncate it at a position.

The truncate function doesn't exist ( https://docs.micropython.org/en/latest/esp8266/tutorial/filesystem.html?highlight=open#creating-and-reading-files https://docs.micropython.org/en/latest/library/builtins.html?highlight=open#open )

Any alternatives?

I thought about copying the whole contents to a new file, but I wanted to avoid that since I'm limited to memory and storage(working on a LoPy4)

servolan, 2020-02-04:
Hello,
Did-you find a solution for this problem. It does interest me because I have the same problem with a Lopy4

dane-if3, 2020-02-28:
I would use this if it was available

servolan, 2020-02-28:
I finally defined my own truncate function:

def Truncate(self, Num):
		with open("controle.txt","r") as FText:             # controle.txt is the file to truncate
			lignes = FText.readlines()			# Read the content of the file
		with open("controle.txt","w") as FText:           # Create a new empty file controle.txt
			for i in range(0,Num):				
				try:
					FText.write(lignes[i])		# Paste one line by one line until Num 
				except:
					pass

dpgeorge, 2023-04-11:
Both the FatFS and LittleFS drivers provide a file truncate function, so this would be straight forward to implement (and of course make it match CPython).

Auguxst-char, 2024-12-09:
I could also really use this for a mpy project that I'm working on