micropython/micropython #4776: deque object isn't iterable

spacemanspiff2007, 2019-05-10:
MicroPython v1.10 on 2019-01-25; ESP32 module with ESP32

```python
from ucollections import deque

samples = deque((), 64)
samples.append(1)
samples.append(2)
for i in samples:
    print(i)
```
fails with ```TypeError: 'deque' object isn't iterable```

```python
print(dir(samples))
```
prints
```['__class__', 'append', 'popleft']```

[Documentation](https://docs.micropython.org/en/latest/library/ucollections.html#ucollections.deque):
Deques (double-ended queues) are a **list-like** container, however lists behave different:
```python
test_list = [1,2,3]
print(dir(test_list))
for i in test_list:
    print(i)
```
```
['__class__', 'append', 'clear', 'copy', 'count', 'extend', 'index', 'insert', 'pop', 'remove', 'reverse', 'sort']
1
2
3
```

pfalcon, 2019-05-10:
If it's written that it's list-like, that doesn't mean that it behaves all like list, and docs are explicit in what functionality it provides. If you need list, use list. The purpose of objects like ucollections.deque is to do what lists can't do, in the minimal code size.

spacemanspiff2007, 2019-05-10:
But wouldn't it be useful to access the contents of the deque?
I would have expected that there is at least one possibility to access them.
The only possibility is to ```popleft``` all items and then ```append``` them again which makes no sense tbh.

pbtsrc, 2021-01-13:
The collections.deque from CPython makes it very easy to implement a fast moving average for real-time data (for example, for averaging data from some sensor):
```python
deq = collections.deque(window_size)
while true:
    sample = sensor.det_data()
    deq.append(sample)
    mov_aver = sum(deq) / len(deq)
```
So it would be very useful to make the ucollections.deque iterable also.

davewood, 2021-02-06:
I just ran into the exact same issue.
```
              return sum(self.queue)/len(self.queue)
```

Spinnaker-design, 2022-08-26:
This would make deque much more useful

peterhinch, 2022-08-27:
The version [here in micropython-lib](https://github.com/micropython/micropython-lib/blob/master/python-stdlib/collections.deque/collections/deque.py) is iterable. The class has an `__iter__` method, and it works as expected:
```python
>>> d = deque((5,1,9,2))
>>>  sum(d) 
17
>>> max(d)
9
>>> for n in d:
...     print(n)
... 
5
1
9
2
>>> 
```

mattytrentini, 2022-09-06:
For some background: There is a built-in version of deque [written in C](https://github.com/micropython/micropython/blob/master/py/objdeque.c) that supports `append` and `popleft`...and not much else. It's lean and fast but, obviously, doesn't have many features. 

There's a more full-featured version of [deque in micropython-lib](https://github.com/micropython/micropython-lib/tree/master/python-stdlib/collections.deque) that is written in Python and wraps up a list. micropython/micropython-lib#440 will add more features. This version is slower, takes more space but is becoming close to feature-complete.

The C built-in is packaged with each release of MicroPython but the MicroPython-lib version _needs to be manually installed_. 

Moving forward, I'd like to see the MicroPython-lib version reach feature parity with CPython as well as having tests added. In parallel we should judiciously add more features to the built-in; taking care since there is a cost in terms of flash for every build. 

Another alternative that has been discussed is to add deque as a [native module](https://docs.micropython.org/en/latest/develop/natmod.html). No work has started on this effort yet, to my knowledge.

plumonito, 2024-03-17:
Ran into this issue as well trying to implement a running average :(

peterhinch, 2024-03-17:
I would consider a circular buffer for this task: it avoids repeated allocation.

dpgeorge, 2024-03-18:
Link to #10724.

dpgeorge, 2024-03-18:
As of 7dff38fdc190d7b731fad8319d2ae8aa13fde18a, deque is iterable.

closed 2024-03-18 by dpgeorge