micropython/micropython #4784: Preparing for next release, v1.11

dpgeorge, 2019-05-14:
The last release was v1.10 on 26 Jan 2019.  Since then there were around 350 commits, and big changes to the .mpy file format, so the plan is to make the next release v1.11.

Remaining issues/PRs to resolve before the release are:
- #4020 / #4746 / #4763 - I2C supporting writes with a list of buffers 

Please mention any other issues/PRs that could fit in for this release.

peterhinch, 2019-05-14:
**upip** - Please ensure it can install library modules compatible with official MicroPython (either via an alternative version or a command line arg).

pfalcon, 2019-05-14:
Please include correct copyright information about files I created or made substantial contributions to, as well as to the whole project, pull requests https://github.com/micropython/micropython/pull/4787 & https://github.com/micropython/micropython/pull/4363 .

miketeachman, 2019-05-16:
It would be great to have v1.11 in a condition to facilitate productive development with the new PYBD boards.  I saw one PR #4669 related to this.  Perhaps there are other issues and PRs that fit into this category?  Are there step-by-step docs describing how to make a build?

What do you think?   There will be a lot PYBDs coming into developer hands - it would be cool to make it easy for people to get started and contribute.

nedoskiv, 2019-05-16:
It will be nice to have stable network.PPP module

dpgeorge, 2019-05-17:
> It would be great to have v1.11 in a condition to facilitate productive development with the new PYBD boards.

The reason PYBD support is not merged yet is because I wanted that to be the focus of the next-next release (v1.12).  The upcoming v1.11 already has enough new stuff in it, and merging big changes for PYBD will possibly lead to regressions to existing code which need time to be addressed.  So in the interest of cutting a stable release ASAP, I don't want to merge any PYBD related items yet.

dpgeorge, 2019-05-17:
> It will be nice to have stable network.PPP module

Unfortunately this is not possible right now.  It's a really difficult thing and needs someone who has the hardware, expertise and time to look into making it work.  If you can look into it yourself and make some progress improving it that would be really great.

mattytrentini, 2019-05-28:
I'm not sure what kind of timeframe we're looking at but I intend to make a few improvements to the [ESP32 docs](https://github.com/mattytrentini/micropython/issues) in the next week. Mostly issues we noticed shortly after the 1.10 release. Shouldn't be a showstopper for 1.11 though.

dpgeorge, 2019-05-28:
> I'm not sure what kind of timeframe we're looking at

A release is imminent.

dpgeorge, 2019-05-29:
Ok... v1.11 was tagged in 6f75c4f3cd393131579db70cdf0b35d1fe5b95ab

closed 2019-05-29 by dpgeorge

robert-hh, 2019-05-29:
When updating the Download pages, please update the Pyboard D binaries too.

dpgeorge, 2019-05-29:
> please update the Pyboard D binaries too.

Ok, that's now done.  Docs are updated (except for a PDF build error, to be fixed) and esp-extra-scripts branch is force pushed.

pfalcon, 2019-05-30:
> Please include correct copyright information about files I created or made substantial contributions to, as well as to the whole project, pull requests #4787 & #4363 .

What's the reason for not including this in the release?

bruno963852, 2019-05-31:
Sorry if the question is dumb, but it is not possible to just use the gsm module done by loboris on his fork?