micropython/micropython #4785: arm-none-eabi-gcc 9.1 build error (clobbering sp)

jimmo, 2019-05-14:
Looks today's updates got me GCC 9.1, which includes a new error for having "sp" in the clobber list for an `__asm` statement. This comes up in both `cmsis_gcc.h:__set_MSP` (and in the inlined copy in `modmachine.c:machine_bootloader`).

```
modmachine.c: In function 'machine_bootloader':
../../lib/cmsis/inc/core_cm4.h:93:28: error: listing the stack pointer register 'sp' in a clobber list is deprecated [-Werror=deprecated]
   93 |   #define __ASM            __asm                                      /*!< asm keyword for GNU Compiler */
      |                            ^~~~~
modmachine.c:306:5: note: in expansion of macro '__ASM'
  306 |     __ASM volatile ("movs r3, #0\nldr r3, [r3, #0]\nMSR msp, r3\n" : : : "r3", "sp");
      |     ^~~~~
../../lib/cmsis/inc/core_cm4.h:93:28: note: the value of the stack pointer after an 'asm' statement must be the same as it was before the statement
   93 |   #define __ASM            __asm                                      /*!< asm keyword for GNU Compiler */
      |                            ^~~~~
modmachine.c:306:5: note: in expansion of macro '__ASM'
  306 |     __ASM volatile ("movs r3, #0\nldr r3, [r3, #0]\nMSR msp, r3\n" : : : "r3", "sp");
      |     ^~~~~
cc1: all warnings being treated as errors
```

There appears to be surprisingly little info about this, but here's the most relevant message from the gcc discussion.

https://www.mail-archive.com/gcc-patches@gcc.gnu.org/msg203357.html

From my understanding of the clobber list, that the compiler should restore the clobbered registers, it makes sense that "sp" should not have been in the clobber list in the first place.

I've attached disassembly of compiling for PYBV11 with both gcc 8.3, and 9.1 with "sp" removed from the clobber list. I haven't spotted any relevant difference.

[machine_bootloader.9.1.txt](https://github.com/micropython/micropython/files/3177463/machine_bootloader.9.1.txt)
[machine_bootloader.8.3.txt](https://github.com/micropython/micropython/files/3177464/machine_bootloader.8.3.txt)

Not sure the best way to fix this in `lib/cmsis/inc/cmsis_gcc.h` though, seeing as there isn't a fix in upstream cmsis. Are you willing to have diffs to upstream?

dpgeorge, 2019-05-14:
I saw this issue as well.  Just to get it building it was enough to comment out the offending asm line in modmachine.c, so hopefully that means we don't need to patch CMSIS.

If removing sp from the clobber list generates the same code then that would be the easiest change.  Otherwise that asm statement could be extended to include the following call/jump (to the bootloader), then it doesn't matter what's in the clobber list.

dpgeorge, 2019-05-15:
See #4791 for a proposal to fix this.

dpgeorge, 2019-05-17:
Fixed by e1e3704aa1406f9511728f54e62569ea08762d4b

closed 2019-05-17 by dpgeorge