micropython/micropython #4786: esp32: uart deinit missing

nedoskiv, 2019-05-14:
Hello,

I was trying to start internet connection using GSM module and run into some troubles. A part from network.PPP stability there is kind a device restarts when comunicating with modem over serial after network.PPP link is deactivated, so I decided to deinit uart and init it again with hope it gonna solve that issue,but for my surprise uart.deinit is not available.

dpgeorge, 2019-05-15:
UART.deinit() was implemented in 2630d3e51f3f4ac933e556bb3df2953698ac74fc

closed 2019-05-15 by dpgeorge

nedoskiv, 2019-05-15:
Thank you, and if you can answer my next silly question, If I download tomorow latest binary from download page will it contain that implementation?

dpgeorge, 2019-05-17:
It will now, it's updated every 12 hours.