micropython/micropython #4790: ujson and dictionaries with numeric keys

peter-conalgo, 2019-05-14:
ujson module is able to serialize/deserialize dictionaries with numeric keys:

```python
>>> a=dict((i,'a') for i in range(3))
>>> import ujson
>>> ujson.dumps(a)
'{0: "a", 1: "a", 2: "a"}'
>>> b=ujson.loads(ujson.dumps(a))
>>> b==a
True
```

Of course that's useful and all, but it isn't JSON. Javascript can't handle numeric keys in it's mapping objects (AFAIK).

This bit me on a recent project, but I don't think it should be changed, however, a list of differences from typical JSON in the docs would be really helpful.

dpgeorge, 2019-05-15:
It's mentioned near the start of `extmod/modujson.c` that the parser "allows through a superset of JSON".  I guess it's a good idea to put that paragraph in the docs for the `ujson` module.

MrSurly, 2019-07-16:
Today, I also hit this.  I think it _should_ be changed to match cpython, since what `ujson.dumps` emits is _not_ JSON [per the spec](https://www.json.org/).  In fact, it cannot be ingested by the cpython `json` module, nor ECMA script, nor by the `jq` utility.

dpgeorge, 2019-07-16:
> since what `ujson.dumps` emits is _not_ JSON [per the spec](https://www.json.org/).

Ahh, I see.  I overlooked the dump(s) behaviour here, thinking the issue was just that load(s) would accept more than the spec allowed.

Along with numeric keys, uPy also doesn't handle True/False/None correctly, which should also be quoted.  Only strings can appear in json objects as keys.

dpgeorge, 2019-07-19:
> Today, I also hit this.

@MrSurly can you give some more details about the context where you hit this?  Since all keys should be strings in JSON output it's possible to get valid JSON simply by making dicts that have only strings as keys, eg:
```python
d = {"1":1, "1.2":1.2, "true":True}
json.dumps(d)
```

MrSurly, 2019-07-25:
> @MrSurly can you give some more details about the context where you hit this? Since all keys should be strings in JSON output it's possible to get valid JSON simply by making dicts that have only strings as keys, eg

Firstly, `json.dump` and `json.dumps` should _only produce valid JSON_, or throw an exception when something can't be properly serialized to JSON.

For my use case, I have abstracted classes that return a Python data structures, that I then serialize (using the `json` class) for transmission via MQTT. With `ujson` in it's present state, then I'd either have to modify underlying classes to be aware of the limitation with `ujson`, or modify the layers above to do a deep traversal of the data structure to "fix" it so that `ujson` behaves correctly.

dpgeorge, 2019-07-30:
The case of `json.dump/dumps` quoting non-string keys is fixed by 01054f20925b92229402750979864f8d38a76916

closed 2019-07-30 by dpgeorge

MrSurly, 2019-07-30:
> Ahh, I see. I overlooked the dump(s) behaviour here, thinking the issue was just that load(s) would accept more than the spec allowed.

Nothing wrong with a little bit of Postel's Law.