micropython/micropython #4792: MICROPY_PY_BUILTINS_ROUND_INT==0 breaks round(<int>, <n>)

dhalbert, 2019-05-16:
Before https://github.com/micropython/micropython/commit/b318ebf1015ced6354f8bbaf035308214b3f5c5d, which was a rework of #3557, `round(<int>, <n>)` worked. Now, if `MICROPY_PY_BUILTINS_ROUND_INT` is set to 0, `round(2, 2)` or similar throws `NotImplementedError`. `round(2)` still works.

@dpgeorge Was it your intention to disable rounding of integers in this way completely, instead of just not handling negative values for the second arg if it was an int?

A user discovered this after ending up with an integer zero while doing some floating arithmetic.

dpgeorge, 2019-05-17:
I think the intention was to retain the existing behaviour (before that patch) as much as possible, and adding a config option to handle rounding of (big) integers, hence the naming of `MICROPY_PY_BUILTINS_ROUND_INT`.

But indeed it looks like the case of `round(x, n)` with `n>=0` was overlooked (since it worked before).

Regardless of the original intention, it might be worth revisiting how `round()` is configured at build time.  It should be that a calculation is the same regardless of whether an int or float is used, eg `1/2` is the same as `1.0/2`.  And `round(123)` is the same as `round(123.0)`.  So if round supports a second argument for float it should also support it in the same way for int.  As such maybe the configuration should be `MICROPY_PY_BUILTINS_ROUND2` which selects if a second arg is supported fully or not at all.

> A user discovered this after ending up with an integer zero while doing some floating arithmetic.

Can you explain in more detail?  With the current behaviour everything should work consistently, except a `NotImplementedError` is raised for ints with a second arg.

dhalbert, 2019-08-20:
> Can you explain in more detail? With the current behaviour everything should work consistently, except a `NotImplementedError` is raised for ints with a second arg.

(Apology for missing your question.) Here's the original report: https://forums.adafruit.com/viewtopic.php?f=60&t=151793. The user was doing `float(A-B, 2)`, where `A` and `B` were equal.