micropython/micropython #4795: ESP8266 gets stuck with uasyncio for 1h 11m

kevinkk525, 2019-05-17:
Since now a 2nd person encountered the bug I brought up on the forum months ago, it might be time to put it to github. It doesn't happen very often and apparently not on every device.

What happens: Running uasyncio code sometimes gets stuck for exactly 1h 11m and then resumes normally. Interrupts still work during this time, that's why I use a software watchdog to reset the boards when it happens.

The latest forum post is this: https://forum.micropython.org/viewtopic.php?f=16&t=6433
Jomas who started that post said: "1 hour, 11 minutes is 2^32 uSeconds, that is one cycle of the internal system_get_time() call of the Espressif SDK".

As interrupts are still working and the program resumes after the pause, the reason can't be a bad power supply or similar external reasons.

peterhinch, 2019-05-18:
**uasyncio** gets all its timing from **utime.ticks_ms**.

I can see two possibilities, depending on the nature of the failing code. If the code always yields with specific time intervals e.g.:
```python
await asyncio.sleep_ms(10)
```
then the fault may be caused by discontinuities in the values returned by **utime.ticks_ms**. However if there is any coroutine which runs an infinite loop with a zero delay
```python
while True:
    await asyncio.sleep(0)
    # do something
```
the task will remain on the runq. That coroutine will be unaffected by glitches in **ticks_ms**.

I'd therefore suggest two tests. Firstly, add a task with a zero delay and test whether it continues to run during these outages. If it does not it suggests that the whole MicroPython VM may be stalled. If the coroutine runs while others hang it suggests that there may be glitches in **ticks_ms**.

The presence of such timing discontinuities could be tested along these lines:
```python
import utime as time
import gc
COUNT, MAX, MIN =  , ,   # Fill in suitable values
a = 0
while True:
    gc.collect()
    t = time.ticks_ms()
    for x in range(COUNT):
        a = (a + 1) % 10 # anything that takes time
    delta = time.ticks_diff(time.ticks_ms(), t)
    if delta < MIN or delta > MAX:
        print('FAIL Time interval', delta)
```

kevinkk525, 2019-05-21:
We now found the bug and I'll try to summarize everything here. Thanks @pythoncoder @robert-hh and @jomas for the idea, testing and posts they made.

Problem: utime.ticks_ms() returns the wrong value when system_get_time() overflows.
Reason: system_get_time() overflows but system_time_high_word gets only incremented when the function ets_loop_iter in ets_alt_task.c is called.
However calling mp_hal_ticks_ms() in esp_mhal.c through utime.ticks_ms needs the correct system_time_high_word:

```
uint32_t mp_hal_ticks_ms(void) {
    return ((uint64_t)system_time_high_word << 32 | (uint64_t)system_get_time()) / 1000;
}
```

This means, that between the overflow in system_get_time() and the next call of ets_loop_iter the ticks_ms returned by mp_hal_ticks_ms() is wrong by an offset of almost 2^32 us. 

How to reproduce:
```
>>> while True:
...     gc.collect()
...     t=time.ticks_ms()
...     time.sleep_us(20)
...     delta = time.ticks_diff(time.ticks_ms(), t)
...     if delta < 0 or delta >5:
...         print("Fail time interval", delta, ", t:",t,"now:",time.ticks_ms())
...
...
...
Fail time interval 4294967, t: 1 now: 4294968
Fail time interval 4294967, t: 4294968 now: 8589936
Fail time interval 4294968, t: 8589935 now: 12884903
Fail time interval 4294968, t: 12884902 now: 17179870
Fail time interval 4294967, t: 17179869 now: 21474837
Fail time interval 4294968, t: 21474837 now: 25769805
```
On every overflow of system_get_time() I'm able to read one ticks_ms() value without an incremented system_time_high_word value.

Consequences: Most noticeable in uasyncio which will block everything for 1h 11m if it tries to get the current time and gets that wrong value. But could generally create problems with timings.

Possible bugfix (not tested and I don't know much about micropython C-Code):

Change mp_hal_ticks_ms in esp_mphal.c to this:
```
uint32_t mp_hal_ticks_ms(void) {
    // handle overflow of system microsecond counter
    ets_intr_lock();
    uint32_t system_time_cur = system_get_time();
    if (system_time_cur < system_time_prev) {
        system_time_high_word +=1; //record overflow of low 32-bits
    }
    system_time_prev = system_time_cur
    ets_intr_unlock();
    return ((uint64_t)system_time_high_word << 32 | (uint64_t)system_time_cur) / 1000;
}
```

In ets_loop_iter in ets_alt_task.c: 
```
bool ets_loop_iter(void) {
    if (ets_loop_iter_disable) {
        return false;
    }

    // call time function to ensure that overflow is being recognized.
    mp_hal_ticks_ms();

```

There might be better solutions, this was just what I could think of.

dpgeorge, 2019-05-22:
Thanks very much for the report and the detailed analysis and fix!  I agree with everything above.

It's possible to make it so you don't need to wait 71mins for the test to fail: write a big number to 0x3ffeaa88, which is a 32-bit delta that's added to the result of `system_get_time()`.  Eg:
```python
def test(advance=0):
    import gc, time, machine
    machine.mem32[0x3ffeaa88] = advance
    print(hex(time.ticks_us()))
    print(hex(time.ticks_ms()))
    while True:
        gc.collect()
        t=time.ticks_ms()
        time.sleep_us(20)
        delta = time.ticks_diff(time.ticks_ms(), t)
        if delta < 0 or delta >5:
            print("Fail time interval", delta, ", t:",t,"now:",time.ticks_ms())
```
Then if you run this after a hard reset with `test(0xfe00_0000)` it will only take about 30 seconds to hit the wraparound.

I've posted a fix in #4803

dpgeorge, 2019-05-24:
Fixed by 5357dad52efbe5773f4beda7ad014dc62ad8e44f

closed 2019-05-24 by dpgeorge