micropython/micropython #4797: The structure that calls `m_new()` to instantiate must contain `mp_obj_base_t base;`?

Neutree, 2019-05-20:
https://github.com/micropython/micropython/blob/e328a5d4693f9e4a03b296e7a9a7af6660d99515/py/gc.c#L270-L293

In the function `gc_sweep`  at line 275:
We got pointer by block id, and convert it to `mp_obj_base_t`, 
does it means that we must use `mp_obj_base_t base;` at the beginning of the structure to be instantiated with `m_new`? Otherwise, the `mp_load_method_maybe` function may cause a program error  ( for example, https://github.com/micropython/micropython/blob/e328a5d4693f9e4a03b296e7a9a7af6660d99515/py/runtime.c#L1055 will fail the check because `type` itself does not exist. 
)!

But we can find many places in the source code to initialize the structure without `base` members with `m_new`：[m_new search](https://github.com/micropython/micropython/search?q=m_new&unscoped_q=m_new)

Which of the following two methods is correct? Thanks for the answer!

```python
typedef struct{
    int a;
    int b;
} data_t;

data_t* p = m_new(data_t, 1);
```

or

```python
typedef struct{
    mp_obj_base_t base;
    int a;
    int b;
}
data_t* p = m_new(data_t, 1);
p->base.type = NULL;
```

dpgeorge, 2019-05-20:
If you use `m_new_obj_with_finaliser` or `m_new_obj_var_with_finaliser` to allocate memory then the first element of the struct must be `mp_obj_base_t` and must be a valid pointer.  Otherwise, using any other allocation macro/function (eg `m_new`), it doesn't need this member, the block won't be checked for a finaliser function.

closed 2019-05-20 by dpgeorge

Neutree, 2019-05-21:
Sorry i missed the `FTB_GET(block)` at line 274!

Thank you very much for your help