micropython/micropython #4801: sock.connect(socket.getaddrinfo(ip,port)[0][-1]) takes too long to respond

hetvishah08, 2019-05-21:
I'm trying to use usocket library in my application and I want to develop a software where if a I input a wrong host address it should respond quickly, reset and try again through a webserver based configuration.

sock.connect(socket.getaddrinfo(slave_ip, slave_port)[0][-1])

During testing, I intentionally input the wrong address and it takes around 19 seconds, after which it responds with this error:
[Errno 113] EHOSTUNREACH
 I want to reduce this time so that i can perform some other actions.
P.S: sock.setblocking(False) is not an option.

peterhinch, 2019-05-22:
The need for a nonblocking version of **getaddrinfo** has been discussed. As far as I know nobody has yet written one for MicroPython.

dpgeorge, 2019-05-24:
See also #3666