micropython/micropython #4804: i2c.readfrom_mem cannot receive more than 15 bytes

nedoskiv, 2019-05-22:
Hello,

I'm using an attiny programmed with arduino ide to communicate with esp32, no matter how I try i cannot receive more than 15 bytes at same time:

`>>> i2c.readfrom_mem(41, 0x09, 30)
b'1234560,9,12345\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff'`

arduino code:

`case 0x09:
      Wire.write("1234560,9,1234569");`

is there a limit using that command?

If I try to send 30 bytes for example, (reprogram attiny)  i2c command raise an error and next i2c communication also fails:

`i2c.readfrom_mem(41, 0x09, 30)

Traceback (most recent call last):

  File "<stdin>", line 1, in <module>
OSError: [Errno 110] ETIMEDOUT
i2c.readfrom_mem(41, 0x08, 30)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 19] ENODEV
i2c.readfrom_mem(41, 0x08, 30)
b'396,361,757\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff'`

dpgeorge, 2019-05-22:
As far as I know, the Arduino Wire interface has a fixed buffer for read/write data and you cannot read or write more than this fixed buffer size (from the Arduino side).  Usually it's 32 bytes but it could be less for an ATTINY, eg 15-16 bytes.

I would suggest checking what the limit is and trying to increase it (on the Arduino side).

nedoskiv, 2019-05-22:
I also tried to sent 3 x 10 butes in 3 diffetent lines of arduino IDE code, it goes down on 15th byte no matter what. So far it points to esp/micropyuthon issue since arduino got no problem like this reported. But gonna try to investigate further.
I also tried similar code on atmega8 processor, same situation, but it can still be arduino wire library, or something on micropython side, gonna try communication between 2 arduino devices to see the result

dpgeorge, 2019-05-23:
Which version of MicroPython are you using that shows the problem?

nedoskiv, 2019-05-23:
oficial binary:
MicroPython v1.10 on 2019-01-25; ESP32 module with ESP32

[esp32-20190125-v1.10_work_fine!.zip](https://github.com/micropython/micropython/files/3211038/esp32-20190125-v1.10_work_fine.zip)

perhaps a little bit old, gonna try on new releases

nedoskiv, 2019-05-23:
Just tried it on latest build, it still cut off on 15th byte, but communication did not fail when sent 30 bytes from slave device (attiny). That it better :)

dpgeorge, 2019-05-24:
I tested latest master 2762f323bf6407093dc814591e7c6d08ca82bc6f on esp32 connected over I2C to a custom I2C slave and the esp32 could read beyond 15 bytes without an issue (I tested a bit more that 100 bytes in one `readfrom_mem` call).

What Wire library is the ATTINY using?

nedoskiv, 2019-05-24:
standart wire.h library, used it on attiny and atmega8, since you have tested it at it works, problem may be somewhere in my code, but there is nothing I can do for now, for the test I do wire.write ("something to 30 bytes") not variables ...... and it fails,
I use interrupts to handle i2c communication could there be the problem?

dpgeorge, 2019-05-24:
It really depends what Wire library you use.  For example this library has a 16-byte maximum buffer size for I2C slave: https://github.com/rambo/TinyWire/blob/rollback/TinyWireS/usiTwiSlave.h#L76 .

There could also be a problem that the esp32's I2C is too fast.  Try a lower I2C frequency.

nedoskiv, 2019-05-24:
I already lowered frquency to 10000, and beside library i tried to sent in portions example:

```
wire.write("12345")
wire.write("12345")
wire.write("12345")
wire.write("12345")
```

that way there is a buffer for every command, do not work either, let froze for a while this issue until I manage to test with something else than a esp32 master

dpgeorge, 2019-05-24:
> i tried to sent in portions

That won't help, the 16 bytes is the total maximum of all `Wire.write()` calls for a given transaction (the data to output is buffered until the IRQ handler returns).

nedoskiv, 2019-05-25:
Thank you,
Here is knowledge/solution I aquired thanks to you and attiny core creator:

Wire.h for attiny is split in half, and some devices buffers are defined in different file (in my case attiny 84):

`ATTinyCore\hardware\avr\1.2.3\libraries\Wire\src\USI_TWI_Slave\USI_TWI_Slave.h`

I read buffer size of 32 bytes in wire.h but that was incorrect:)

closed 2019-05-25 by nedoskiv