micropython/micropython #4805: Flashing of the build for stm32f769 fails

stepansnigirev, 2019-05-22:
I am trying to flash the micropython build on STM32F769 discovery board:
```
make BOARD=STM32F769DISC deploy-stlink
```

It compiles successfully and creates two files `firmware0.bin` and `firmware1.bin`. During flashing everything works fine for `firmware0.bin` but flashing of `firmware1.bin` fails with an error:
```
Writing build-STM32F769DISC/firmware1.bin to the board via ST-LINK
st-flash 1.5.1
2019-05-22T21:49:35 INFO common.c: Loading device parameters....
2019-05-22T21:49:35 INFO common.c: Device connected is: F76xxx device, id 0x10016451
2019-05-22T21:49:35 INFO common.c: SRAM size: 0x80000 bytes (512 KiB), Flash: 0x200000 bytes (2048 KiB) in pages of 2048 bytes
Unknown memory region
make: *** [deploy-stlink] Error 255
```

When I try to convert it to a bin file for drag-and-drop flashing using `arm-none-eabi-objcopy -O binary build-STM32F769DISC/firmware.elf upy-f769disco.bin` it creates a 2GB file.

Previous builds without qspi work fine.

dhylands, 2019-05-22:
By default it looks like USE_MBOOT and USE_QSPI are both defaulted to 1. This means that you'd need to use mboot, since it knows how to write to the QSPI, where the deploy-stlink doesn't know how to do this.

In order to use deploy-stlink I think that you'd need to use USE_QSPI=0 and possibly USE_MBOOT=0 as well.

stepansnigirev, 2019-05-22:
Thanks! It worked!

closed 2019-05-22 by stepansnigirev

dhylands, 2019-05-22:
You can also tell whether USE_QSPI was used by running `readelf -l build-STM32F769DISC/firmware.elfreadelf -l build-STM32F769DISC/firmware.elf`

When USE_QSPI is set to 1, you'll get output like this:
```
Elf file type is EXEC (Executable file)
Entry point 0x804bd59
There are 7 program headers, starting at offset 52

Program Headers:
  Type           Offset   VirtAddr   PhysAddr   FileSiz MemSiz  Flg Align
  EXIDX          0x06315c 0x0807315c 0x0807315c 0x00008 0x00008 R   0x4
  LOAD           0x010000 0x08020000 0x08020000 0x53164 0x53164 R E 0x10000
  LOAD           0x070000 0x20020000 0x08073164 0x00034 0x00034 RW  0x10000
  LOAD           0x074000 0x20024000 0x08073198 0x00000 0x0f064 RW  0x10000
  LOAD           0x073064 0x20033064 0x08073198 0x00000 0x04000 RW  0x10000
  LOAD           0x077064 0x20037064 0x08073198 0x00000 0x00800 RW  0x10000
  LOAD           0x080000 0x90000000 0x90000000 0x08b84 0x08b84 R E 0x10000

 Section to Segment mapping:
  Segment Sections...
   00     .ARM.exidx 
   01     .isr_vector .text .ARM.exidx 
   02     .data 
   03     .bss 
   04     .heap 
   05     .stack 
   06     .text_qspi 
```
(notice the .text_qspi section and the load address of 0x90000000. If you build with USE_QSPI=0 then running readelf will show this instead:
```
Elf file type is EXEC (Executable file)
Entry point 0x8051851
There are 6 program headers, starting at offset 52

Program Headers:
  Type           Offset   VirtAddr   PhysAddr   FileSiz MemSiz  Flg Align
  EXIDX          0x06b748 0x0807b748 0x0807b748 0x00008 0x00008 R   0x4
  LOAD           0x010000 0x08020000 0x08020000 0x5b750 0x5b750 R E 0x10000
  LOAD           0x070000 0x20020000 0x0807b750 0x00034 0x00034 RW  0x10000
  LOAD           0x074000 0x20024000 0x0807b784 0x00000 0x0f064 RW  0x10000
  LOAD           0x073064 0x20033064 0x0807b784 0x00000 0x04000 RW  0x10000
  LOAD           0x077064 0x20037064 0x0807b784 0x00000 0x00800 RW  0x10000

 Section to Segment mapping:
  Segment Sections...
   00     .ARM.exidx 
   01     .isr_vector .text .ARM.exidx 
   02     .data 
   03     .bss 
   04     .heap 
   05     .stack 
```
Note that in this case there is no .text_qspi section. This .elf should flash fine using deploy-stlink. This image assumes that mboot is flashed (see the PhysAddr of 0x08020000 for the first LOAD line). If you build with both USE_MBOOT=0 and USE_QSPI=0 then you'll get something like this instead:
```
Elf file type is EXEC (Executable file)
Entry point 0x804dd91
There are 7 program headers, starting at offset 52

Program Headers:
  Type           Offset   VirtAddr   PhysAddr   FileSiz MemSiz  Flg Align
  EXIDX          0x077c88 0x08077c88 0x08077c88 0x00008 0x00008 R   0x4
  LOAD           0x010000 0x08000000 0x08000000 0x03a88 0x03a88 R E 0x10000
  LOAD           0x020000 0x08020000 0x08020000 0x57c90 0x57c90 R E 0x10000
  LOAD           0x080000 0x20020000 0x08077c90 0x00034 0x00034 RW  0x10000
  LOAD           0x084000 0x20024000 0x08077cc4 0x00000 0x0f064 RW  0x10000
  LOAD           0x083064 0x20033064 0x08077cc4 0x00000 0x04000 RW  0x10000
  LOAD           0x087064 0x20037064 0x08077cc4 0x00000 0x00800 RW  0x10000

 Section to Segment mapping:
  Segment Sections...
   00     .ARM.exidx 
   01     .isr_vector 
   02     .text .ARM.exidx 
   03     .data 
   04     .bss 
   05     .heap 
   06     .stack 
```
Note that the PhysAddr for the first LOAD section is now 0x08000000.

dpgeorge, 2019-05-23:
Some boards (like this one) are configured to use mboot/qspi as an example of how to use these features.  Perhaps it'd be helpful to emit a message when building that mboot is required for the firmware to work.

dhylands, 2019-05-23:
I created PR #4806