micropython/micropython #4814: stm32 NUCLEO boards: pin name conflicts

chrismas9, 2019-05-26:
On pyboard:
`>>> import pyb`
`>>> pin = pyb.Pin('A2')`
`>>> pin`
`Pin(Pin.cpu.A2, mode=Pin.ALT, pull=Pin.PULL_UP, af=Pin.AF7_USART2)
`

On many NUCLEO boards:
`>>> import pyb`
`>>> pin = pyb.Pin('A2')`
`>>> pin`
`Pin(Pin.cpu.A4, mode=Pin.ANALOG)`

On NUCLEO boards (eg NUCLEO_F091RC or L476RG) 'A2' maps to cpu pin A4.

This is because pins.csv has a board pin entry 'A2' for the Arduino analog pin name. Quite a few of the Arduino pin names are the same as the STM32 port names.

I can think of 4 solutions:
1. Remove the Arduino pin names from pins.csv.
2. Prepend the Arduino name with something, eg connector name - `CN8_A2`
3. Require cpu pins to be entered with a P, eg `pyb.Pin('PA2')`.
4. Require cpu pins to be entered with a` cpu.` prefix, eg `pyb.Pin('cpu.A2')`.

Options 3 & 4 are most likely to break existing code, although this could be avoided by mapping the short names in the csv file, eg `A2,PA2` or `A2,cpu.A2`, for non NUCLEO ports.

 Option 3 may not work for all MCUs as some MCUs don't have a P, just A2, etc.

I am currently using option 2, ie make board pin names unique.

Any suggestions on the best fix?

rolandvs, 2019-05-26:
Why not limit the use of the 'labels' to board only `'X12' = board.X12` and have generic pins always being referred to using `cpu` using like `pyb.Pin.cpu.A2` etc.
A lot of *.csv files also contain all generic pins and these could be removed reducing the set down to what should 'only' be part of the board file anyway as a description of your used board.

chrismas9, 2019-05-26:
I am writing hardware test scripts for new ports and I need to read a database of pins and peripherals to test. Is there a way to create a pin object from a string, eg pyb.Pin.cpu.A2 from 'A2' or 'pyb.Pin.cpu.A2'?

With regard to having CPU pin names in the csv file I would rather see pin names like CN8-1 on the NUCLEO boards. that's roughly equivalent to X1 on Pyboard.

Even if there is a workaround it's not good that code that runs on Pyboard won't run on Nucleo.

dhylands, 2019-05-26:
The order of precedence is pin object, mapper function, mapper dict, board name, CPU name.

So you could provide a dict that maps A2 to pub.Pin.cpu.A2. I think (not in front of a computer with a pyboard to test this) that pub.Pin.cpu is a dictionary, so you could try pyb.Pin.cpu['A2'] and if that's the case then you could provide a mapper function that just looks up CPU pin names. Otherwise I think you could create a string and eval it in a mapper function to get the same result.

chrismas9, 2019-05-26:
Thanks Dave, I will try that tonight. Do you have any opinion on removing CPU pin names, eg PA2, from the left column of pins.csv (the board column) and replacing them with board names like CN8-2 which is what Pyboard does with its X and Y names?

chrismas9, 2019-05-26:
Replacing CPU names in the board column won't break compatibility because PAx, etc will still be found in the CPU names list.

dhylands, 2019-05-26:
The goal of having board names was that if A2 was silk screened on the board then I should be able to use that and get the correct pin. CPU pin names don't mean much to many users.

dhylands, 2019-05-26:
It looks like the cpu and board objects don't support lookups. However I was able to get this to work:
```
pyb.Pin.mapper(lambda x: eval('pyb.Pin.cpu.' + x))
```
If you add that one line to the top of your file, then CPU names should be preferred over board names (since the mapper function gets called first).

dhylands, 2019-05-26:
You can also use getattr:
```
pyb.Pin.mapper(lambda x: getattr(pyb.Pin.cpu, x))
```

dpgeorge, 2019-05-28:
> I am writing hardware test scripts for new ports and I need to read a database of pins and peripherals to test. Is there a way to create a pin object from a string, eg pyb.Pin.cpu.A2 from 'A2' or 'pyb.Pin.cpu.A2'?

Why exactly do you need to reference the pin using the cpu name, why not use the board name?

I understand that there can be conflicts with names but that's because there are two namespaces that are searched, board and cpu.  The only robust way to get exactly the pin you want is to have a way to specify the namespace (or to have just one namespace, but I think both board name and CPU pin name are useful).

chrismas9, 2019-05-28:
> Why exactly do you need to reference the pin using the cpu name, why not use the board name?

Primarily because I am testing the port, not the board. When I do a port I want to implement and verify every alternate function, even if they are not exposed on an eval board. That is to make it easier to develop custom boards that may use all pins or features,. That is why I added support for SPI5, CAN3 and UART10 to the F413 port. If an MCU pin is not exposed on a board pin I will probe it at the chip or internal use point and these pins don't have a board pin name.

I thought about extracting the available pins and af's programatically but that wouldn't find bugs in the af file or omissions from the pins.csv file. I have to work from the data sheet pin list and af's to reliably test the whole port. Creating a list of alternate functions to test has to be done against the MCU port names as there is no cross reference between the board names and the af's.

After your comment about the F413 test code being too board specific I am trying to make them more generic. Port names are common across all MCUs in a family. The board names in the various STM ports are quite different, eg some use Arduino names, others use port names.

> The only robust way to get exactly the pin you want is to have a way to specify the namespace (or to have just one namespace, but I think both board name and CPU pin name are useful).

I agree both namespaces are useful and I think the best solution is to make them unique. I did a survey of all the brands of microcontroller at Digikey (14) with 256K Flash and all but two start their pin names with "P". there are many variations : PA0, PA00, P00, P0_0, PIO0_0, PAD1, P03 & PA0, P0_00. The exceptions are Microchip PIC which has port names (PMD0, RA0) and function names (SCL1, SDA1) and TI DSP (GPIO0 and ADCIN0).

My preference would be to add the "P" to the MCU port names. When designing boards I am always working with PA0, etc and have to remember to remove the P when programming. That's probably less disruptive than adding a long prefix (like connector name) to the board names and having them not match the the board legend.

Alternatively I will find a work around for the tests such as @dhylands suggestion to reverse the search order (thanks Dave).

I intend to run my tests on quite a few of the STM eval boards and i will submit PRs for any bugs or omissions I find. I am happy to clean up the board names at the same time, but I need guidelines or consensus on how to do it.

chrismas9, 2019-05-29:
I realise that changing MCU pin names to `PA0` format would break `mpconfigport.h` pin assignments and that would be a problem for user custom ports, as well as some user code. Would it be possible to add an optional prefix to the pin name string?

`pin_name_string = 'A0'` or `pin_name_string = 'cpu.A0'`
`pin = pyb.Pin(pin_name_string)`

this would not require any changes to configuration files or user code as the MCU pin would still be defined as `A0` but if the optional `cpu.` prefix was found in the string it would force a lookup from the cpu pins. This is the same format as used in the pin object `pyb.Pin.cpu.A0` but with a string `pyb.Pin('cpu.A0')`.