micropython/micropython #4816: `AttributeError: 'Pin' object has no attribute 'on'`

cool-RR, 2019-05-27:
On my ESP32, this happens:

```
>>> import machine                                
>>> p = machine.Pin(2, machine.Pin.OUT)           
>>> p.on()                                        
Traceback (most recent call last):                
  File "<stdin>", line 1, in <module>             
AttributeError: 'Pin' object has no attribute 'on'
```

Even though the docs say that the `Pin.on` method works.

robert-hh, 2019-05-27:
Maybe your firmware version is too old. Which version are you using?
Besides that, you may use p(1) and p(0) as faster method.

cool-RR, 2019-05-27:
Is there a way to find the version from the shell by typing a command?

robert-hh, 2019-05-27:
In REPL, Just Press Ctrl-B

dpgeorge, 2019-05-28:
On esp32, the `on()` and `off()` methods first appeared in release v1.10.

cool-RR, 2019-05-28:
Result: `MicroPython v1.9.4 on 2018-05-11; ESP32 module with ESP32`

closed 2019-05-31 by dpgeorge

thinhx2, 2020-10-21:
> On esp32, the `on()` and `off()` methods first appeared in release v1.10.

thank