micropython/micropython #4817: MQTT on LAN using ESP32

hetvishah08, 2019-05-28:
Can I use the umqtt.simple library on ESP32 DevkitC with LAN as my network? 
I tried using this library and it generated an error for connect():

`>>> from machine import Pin
>>> import micropython,network,time
>>> led = Pin(32, Pin.OUT, value=1)
>>> l = network.LAN(mdc = Pin(23), mdio = Pin(18), power = Pin(17), phy_type = network.PHY_LAN8720, phy_addr=1)
I (17267) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
>>> l.active(1)
I (18487) emac: emac reset done
True
>>> I (18487) ethernet: start
I (22487) ethernet: LAN cable connected
I (23237) event: eth ip: 192.168.29.37, mask: 255.255.255.0, gw: 192.168.29.1
I (23237) ethernet: Got IP

>>> from umqtt.simple import MQTTClient
I (49577) modsocket: Initializing
>>> SERVER = "broker.mqttdashboard.com"
>>> CLIENT_ID = "6pPjrbwL6h"
>>> TOPIC = "testtopic/1"
>>> c = MQTTClient(CLIENT_ID, SERVER)
>>> c.connect()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "umqtt/simple.py", line 105, in connect
IndexError: bytes index out of range
>>>
`
On checking the library i found out the error is generated on line : 
`resp = self.sock.read(4)`

dpgeorge, 2019-05-29:
Yes you should be able to use the MQTT library with LAN.  If there are problems the first thing to do would be verify the LAN connection and stability, eg by using `urequests` to download files from the internet.

hetvishah08, 2019-05-30:
LAN is working fine and I have used urequests for https post and get requests and it worked fine without any error. 
I used the MQTT library for WiFi and it worked properly but it gives me error when i use it for LAN.

dpgeorge, 2019-05-31:
Ok, so there could be a low-level problem with the LAN driver, that it's returning a short read (less than 4 bytes when 4 requested).

jonnor, 2024-09-19:
Is this issue still relevant?

closed 2025-03-03 by jonnor