micropython/micropython #4818: `machine.bootloader()` hard fault

andrewleech, 2019-05-29:
https://github.com/micropython/micropython/pull/4809 breaks `machine.bootloader()`. 

More specifically, after this change was merged `SCB_DisableDCache()` hard faults when trying to jump to bootloader. This is on a custom stm32f765 based board.

Can't really see why, but running `SCB_EnableDCache();` immediately before `SCB_DisableDCache();` fixes the issue, though it's a somewhat unsatisfying fix.

dpgeorge, 2019-05-29:
Is it 100% reproducible, happens every time?

> `HAL_MPU_Disable();` is run just before the offending `SCB_DisableDCache();`

What about `SCB_DisableICache()`?

andrewleech, 2019-05-29:
Every time, yes. I've just changed my boot.py to immediately run `machine.bootloader()` and it's an immediate fail.

`SCB_DisableICache()` works just fine.

I've just added a `if (!(SCB->CCR & (uint32_t)SCB_CCR_DC_Msk)) return;` at the start of `SCB_DisableDCache` function to check and fast-exit if it's not already enabled - it didn't exit. So the `CCR` enable flag is set, ie it is enabled.

Perhaps something about the sdram settings change puts the dcache in a different state, maybe just putting the dcache enable line after that is a good idea.

andrewleech, 2019-05-29:
Moving `SCB_DisableDCache();` above `HAL_MPU_Disable();` makes no difference, still a hard fault.

Adding 
```
    #if __DCACHE_PRESENT == 1
    SCB_EnableDCache();
    #endif
```
to the end of the MPU setting in sdram also fixes the issue (not surprisingly, it's basically the same fix). 

It's probably because the sdram is now marked as CACHEABLE so the DCache needs to be re-enabled to take full effect. 

However, running `SCB_EnableDCache()` which internally runs a cache invalidate now takes quite a long time with my 32MB flash chip, we're talking a number of seconds. 

I'd say having the sdram marked cachable quite possibly isn't worth it. I might want to revert that part of my recent change.

dpgeorge, 2019-05-29:
> However, running `SCB_EnableDCache()` which internally runs a cache invalidate now takes quite a long time with my 32MB flash chip, we're talking a number of seconds.

You mean 32MB SDRAM?

It may be that the cache must be flushed *before* the SDRAM is turned off (which is done by `HAL_DeInit()` in `machine.bootloader()`).  Currently it's the other way around.

andrewleech, 2019-05-29:
Yes, I did mean sdram. 

That's a good idea, I'll try that.

andrewleech, 2019-05-29:
Good call, moving `SCB_DisableDCache();` above `HAL_DeInit` does the trick. It doesn't seem noticeably slow at all either, so looks like a good fix.

dpgeorge, 2019-05-29:
Ok.  I'll have to think how much of a risk it is making this change now, since I was just about to do the v1.11 release.

Case 1: not a risk, make the change.
Case 2: is a risk, so have some #if's to only make the change for F7 with SDRAM.

andrewleech, 2019-05-29:
If it feels risky to you, feel free to change:
https://github.com/micropython/micropython/blob/master/ports/stm32/sdram.c#L277
`MPU_ACCESS_CACHEABLE` to `MPU_ACCESS_NOT_CACHEABLE` as that also resolves the problem, albeit with an un-measured, presumed performance regression back to the way it was a few commits ago.

commit 66bcb559 (2019-05-29): stm32/modmachine: In bootloader() disable caches before reset of periphs

Otherwise flushing and disabling the D-cache will give a hard-fault when
SDRAM is used.

Fixes #4818.
closed 2019-05-29 by dpgeorge