micropython/micropython #4821: Roadmap for next release (v1.11 -> v2.0?)

dpgeorge, 2019-05-29:
This issue is intended to give an overview, track progress and provide discussion for the roadmap of features and changes to get MicroPython to the next release.

And I'd propose that the next release be v2.0, because MicroPython is now quite mature (although definitely not finished), and also v2.0 of the software would roughly correspond to the release of pyboard v2.  (See #2486 for existing discussion about v2.0.)

The following features/changes are in scope for the next release (note this list is not a guarantee, nor exhaustive):
- [ ] Python 3.5 support? and 3.6?
- [ ] full support for PYBD-SFx; see #4669 
- [ ] initial Bluetooth support; see #4589 
- [ ] add LittleFS; see #3847 
- [ ] update CMSIS; see #4820 
- [ ] enhance pyboard.py to copy files over serial REPL
- [ ] remove mpy-cross from automatically being built (it's stable enough now)
- [ ] add some performance tests
- [ ] stm32: move `boards/startup*.s` to stm32lib
- [ ] stm32: multi LUN support on USB MSC
- [ ] add I2CSlave support (I2CSlaveFIFO, I2CSlaveMem)
- [ ] esp8266/esp32: switch to use LittleFS (saves 4k RAM, more robust filesystem)?
- [ ] esp32: support more of its features, like SDCard, OTA, mesh
- [ ] initial port to ATSAMDx MCUs?

rolandvs, 2019-06-02:
stm32: USB HOST/OTG?

dpgeorge, 2019-06-04:
> stm32: USB HOST/OTG?

I'm not sure about this, it's a nice feature to have but a fair bit of work to implement fully.

dmartauz, 2019-06-07:
esp32: hardware I2C

mattytrentini, 2019-06-12:
Maybe strive for identical - or very similar - API's across the supported ports? Where are we at with compatibility? 

Ideally everything in `machine` would be (close to) identical across ports. 

I know this is controversial, but...this should include considering deprecating parts of `pyb` - at least those that have alternatives in `machine`. This needs to be done with care and respect; as @peterhinch has correctly pointed out there is a lot of code that uses `pyb`. So maybe output a warning when a deprecated API is imported? Update documentation to guide users to the common API's and then later (or never) remove the `pyb` entries. Also need to figure out what other parts of pyb ought to remain there (genuinely specific to the PyBoard) or should be shifted so that other ports can use the same API (maybe I2C client?). 

It just seems that API stability would be a great goal of v2.0.

dpgeorge, 2019-06-12:
> It just seems that API stability would be a great goal of v2.0.

Yes I agree, and work continues to be done in that direction, but it's a hard goal to hit.  For ADC and PWM see #4213 and #4237 .  It'd be good to get at least these done, and then Pin, ADC, PWM, SPI, I2C and UART would be relatively consistent.

> Update documentation to guide users to the common API's and then later (or never) remove the `pyb` entries

Yes, that's the aim, to update the docs so pyboard uses machine for the most part.  See eg e5e472198c0cf5cbac8fdf0a9ad5037a9c995f02 which prefers `machine.I2C` over `pyb.I2C`.  In general it's hard because pyb-level classes provide a lot of detailed functionality that machine doesn't.

CaptainJackey, 2019-06-13:
Everytime we changed main.py and it need to RST and run , many uart tools such as putty will get error and need to restart . I hope we can improve this damn problem！

jimmo, 2019-06-13:
> Everytime we changed main.py and it need to RST and run , many uart tools such as putty will get error and need to restart . I hope we can improve this damn problem！

Hit Ctrl-D after the red led turns off, rather than pressing the reset button on your pyboard. Also putty has a option in the menu to quickly restart the connection (use this rather than closing the window).

CaptainJackey, 2019-06-13:
> Hit Ctrl-D after the red led turns off, rather than pressing the reset button on your pyboard. Also putty has a option in the menu to quickly restart the connection (use this rather than closing the [...]

Thank you so much!

KarlBenjaminHorn, 2019-06-13:
I would like to have DMA support for the ADC of the Pyboard.

peterhinch, 2019-06-15:
Nonblocking ADC and DAC methods would be great, especially on the Pyboard. I guess these would probably use DMA.

If this were done it would be worth considering **uasyncio** compatibility. This won't suit the highest performance applications. However there are uses for **StreamReader** and **StreamWriter** objects targeted on ADC and DAC instances. For this to work buffering would be needed. This could be implemented via double buffers or ring buffers which can signal half full/half empty status to give **uasyncio** time to schedule the **StreamX** method.

Meekdai, 2019-06-28:
Add support for SPI RAM interface On PYB?
Applications that require more RAM can run

dpgeorge, 2019-06-28:
> Add support for SPI RAM interface On PYB?
> Applications that require more RAM can run

The stm32 port already supports external SDRAM, and external QSPI flash as ROM.

Meekdai, 2019-06-28:
> > Add support for SPI RAM interface On PYB?
> > Applications that require more RAM can run
> 
> The stm32 port already supports external SDRAM, and external QSPI flash as ROM.

I learned about SDRAM extensions from #3940  , but STM32F405RGT6 does not have FMC, so I want to use SPI RAM to extend. This is my discussion in the forum [SPI FLASH 2 in PYBD_SF2](https://forum.micropython.org/viewtopic.php?f=20&t=6630)

mattytrentini, 2019-06-29:
Is it worth improving _some_ aspects of time handling?

- Timezone/DST support, even if it's preliminary or an indication of how it is to be implemented
- [MicroPython](https://docs.micropython.org/en/latest/library/utime.html#utime.localtime) and [CPython](https://docs.python.org/3/library/time.html#time.struct_time) use different structs for `localtime()` (tuple vs named tuple and DST is present in CPython) 
- [RTC](https://docs.micropython.org/en/latest/library/machine.RTC.html) uses a _completely different_ tuple again
  - Shouldn't it match the struct in `time.localtime()`?
- [ntptime](https://github.com/micropython/micropython/blob/master/ports/esp8266/modules/ntptime.py) (ESP32/ESP8266 only) could do with improvements
  - Allow different servers to be queried
  - Improve error handling
    - Retries
    - Meaningful exceptions when there's no network or the request failed. The current errors confuse beginners.
  - Add an async settime method?

Further:

- Consider adding some of [datetime](https://docs.python.org/3/library/datetime.html)?
  - The [CPython implementation](https://github.com/python/cpython/blob/master/Lib/datetime.py) is written in fairly pure Python so perhaps a subset may port easily?
- Consider reviewing [Maya](https://github.com/kennethreitz/maya), [Arrow](https://arrow.readthedocs.io/en/latest/), [Pendulum](https://pendulum.eustace.io/) and [Delorean](https://delorean.readthedocs.io/en/latest/)
  - Are any of their features feasible and valuable? Especially if they're relatively portable and lightweight!

Time is a big and complex domain but it _is_ quite important on an embedded device.

peterhinch, 2019-06-30:
Re modframebuf.c I'd like to see a solution to https://github.com/micropython/micropython/issues/2973.

Blitting between framebufs with arbitrary colour mapping would be ideal. But if it is too heavyweight, a fix for the problem as originally outlined would have practical application and would be simple and cheap. A use case is to render in arbitrary colours objects stored as 1-bit maps (such as font glyphs). This is very slow in Python.

kingsjl, 2019-07-02:
esp32:support bluetooth?

goatchurchprime, 2019-07-09:
I'd really like mDNS support in the socket.getaddrinfo() for looking up ".local" domains.  (This is just the "easy" half of mDNS spec.)  

Otherwise I've got to hard-code the ip-number when using mqtt to a local broker, which is not very robust as this number can change, which then requires taking down and reprogramming all my Micropython devices.  

I've been getting around it using [this minimal UDP sending code](https://github.com/goatchurchprime/jupyter_micropython_developer_notebooks/blob/master/basicsockets/minimalmdns.py), which just about works on an ESP32.  If this were done properly it would seriously help out with connecting embedded devices onto local networks in the real world where people don't always have access to the router to force it to use fixed ip-numbers.

jimmo, 2019-07-09:
> I've been getting around it using [this minimal UDP sending code](https://github.com/goatchurchprime/jupyter_micropython_developer_notebooks/blob/master/basicsockets/minimalmdns.py), which just [...]

@goatchurchprime I agree this is a useful feature!

What do you mean by "just about works". Do you see any reason why this can't be implemented in Python and therefore needs to be part of the base firmware. i.e. can this be an optional module written in Python that users can add to their filesystem (or add as a frozen module) if they want it? (Perhaps ultimately add it to drivers, or micropython-lib)

dpgeorge, 2019-07-09:
> I'd really like mDNS support

Note that in the stm32 port mDNS queries and the responder are enabled for boards that have network interfaces.

For esp32 I think it's just a matter of enabling the feature.

goatchurchprime, 2019-07-09:
Moved the mDNS discussion out to #4912

pidou46, 2019-07-26:
Do you think the pulse counter (pcnt) support for esp32 could fit in the "esp32: support more of its features, like OTA, mesh, I2S" item ?

Does it lives in the lowhanging fruit area ?

dpgeorge, 2019-07-29:
> Do you think the pulse counter (pcnt) support for esp32 could fit in the "esp32: support more of its features, like OTA, mesh, I2S" item ?
> 
> Does it lives in the lowhanging fruit area ?

Feel free to look into it.  There's no problem to support more of the esp32 peripherals, it's just a matter of time to work on it.

peterhinch, 2019-08-24:
Support for nonblocking DNS lookup (`usocket.getaddrinfo`).
This has cropped up in the forum, here https://github.com/micropython/micropython/issues/4801 and in my own applications.

nevercast, 2019-09-01:
My colleague @ironss has created a fork over here for internal use that adds LittleFS for ESP32 https://github.com/ironss/micropython-littlefs

I see LittleFS is on the roadmap but no mention of an ESP32 implementation. Can we get this mainlined?

mattytrentini, 2019-09-01:
@nevercast 
The description does mention an ESP32 implementation: "esp8266/esp32: switch to use LittleFS (saves 4k RAM, more robust filesystem)?" 

Looks like nice work by @ironss... :+1:

nevercast, 2019-09-01:
Ah yes, further down the page. I stopped reading too early, good spotting.

beyonlo, 2019-09-19:
OTA and Mesh is very important in my opinion to support on the ESP32.

nevercast, 2019-09-19:
> Mesh
I believe ESP32 Mesh networking requires ESP-IDF 4.0 or 4.1, its not available in our current version. Though there is progress at least with moving to the newer versions of the IDF, see this commit: https://github.com/micropython/micropython/commit/96008ff59a8af9883af17d01b951029d9d02eec9 and the PR before it: https://github.com/micropython/micropython/pull/5107

beyonlo, 2019-09-19:
> > Mesh
> 
> I believe ESP32 Mesh networking requires ESP-IDF 4.0 or 4.1, its not available in our current version. Though there is progress at least with moving to the newer versions of the IDF, see this commit: [96008ff](https://github.com/micropython/micropython/commit/96008ff59a8af9883af17d01b951029d9d02eec9) and the PR before it: #5107
> 
> Its also in the OP post, so that's neat.

@nevercast 
That's a great news!
Mesh and OTA will be amazing =D

peterhinch, 2019-09-27:
Support for SSL/TLS on nonblocking sockets.

hoihu, 2019-10-01:
i saw that littlefs is on the roadmap and just wanted to 👍👍 this. 

It would be cool to have an alternative FS around that would be more robust, resilient against power outages and a low memory footprint.

beyonlo, 2019-10-07:
> Support for SSL/TLS on non-blocking sockets.

But actually using uasyncio we already has non-blocking sockets, right? So, add SSL/TLS on that socket will change from a non-blocking sockets to a blocking sockets?

Thank you.

beyonlo, 2019-10-07:
@peterhinch 

> Support for SSL/TLS on nonblocking sockets.

But actually using uasyncio we already has non-blocking sockets, right? So, add SSL/TLS on that socket will change from a non-blocking sockets to a blocking sockets?

Thank you.

peterhinch, 2019-10-08:
The use of a blocking socket is incompatible with uasyncio because all other tasks will be suspended for the duration of the blocking. In the case of SSL/TLS on ESP8266 I gather blocking can be for long periods (tens of seconds).

beyonlo, 2019-10-08:
@peterhinch 

Sorry, that was not clear to me. Please, could you to clarify?

> The use of a blocking socket is incompatible with uasyncio because all other tasks will be suspended for the duration of the blocking. 

I checked a different affirmation, that yes, uasyncio supports blocking sockets:
**"uasyncio supports blocking sockets by using select.poll to prevent them from blocking the scheduler. In most cases it is simplest to use this mechanism. Example client and server code may be found in the client_server directory. The userver application uses select.poll explicitly to poll the server socket. The client sockets use it implicitly in that the uasyncio stream mechanism employs it."** This was get from https://github.com/peterhinch/micropython-async/blob/master/TUTORIAL.md#76-socket-programming

Well, I'm already using that client tcp socket example, referenced above (**client_server directory**) https://github.com/peterhinch/micropython-async/blob/master/client_server/uclient.py  So, I create a simple microPython app, where I use that socket example running together others **"async def function()"** and all show to running in asyncronous mode (non-blocking), I mean, that socket not seems to be are blocking. Or, are that socket (StreamReader and StreamWriter) being blocking, but is so fast that is not perceptible?

> In the case of SSL/TLS on ESP8266 I gather blocking can be for long periods (tens of seconds).

Just o to confirm, so, you say that if I use SSL/TLS on this example https://github.com/peterhinch/micropython-async/blob/master/client_server/uclient.py it will be block for long periods (tens of seconds)?

O love about uasyncio and non-blocking mode, so I would like to understand better that.

Thank you so much to clarify that.

peterhinch, 2019-10-09:
This discussion should be continued in [the forum](http://forum.micropython.org/index.php) not this bug tracker. 

You are correct in that blocking sockets can be used with the poll mechanism. The point I was trying to make - evidently not very well - is this. As I understand it SSL/TLS as implemented in MicroPython only works with blocking sockets. Further, it can block for significant periods. So SSL/TLS currently does not cooperate well with **uasyncio**.

If you look at the **uasyncio** **start_server()** method you will see that **StreamReader** and **StreamWriter** will work with nonblocking sockets; they can also use blocking sockets because the poll mechanism is used. As in my **uclient.py** example you cited.

I can't comment on likely performance on ESP32. I have done little actual testing with TLS. I suggest you experiment and report back - in the forum. :)

beyonlo, 2019-10-09:
@peterhinch 

Thank you for clarify!
All right, any new comment or report I will do at the forum :)

Anyway, I agree, the **support for SSL/TLS on nonblocking sockets** I would like to see in the next version :)

nevercast, 2019-10-15:
> esp32 supports quadrature encoder.

Using which hardware? I don't believe there is a hardware driver for that.

mcauser, 2019-10-15:
> > esp32 supports quadrature encoder.
> 
> Using which hardware? I don't believe there is a hardware driver for that.

Using the pulse counter peripheral?

nevercast, 2019-10-15:
> Using the pulse counter peripheral? 

Interesting usecase. It would still require a Python driver to decode the pulse counts in to a rotation and we should probably have a purely GPIO solution also. But yes, good point, we should have an ESP32 Pulse Counter driver.

mattytrentini, 2019-10-16:
For reference: [ESP-IDF Pulse Counter](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/pcnt.html). 

Agree, this should be a separate ticket. Note that #5214 touches on the topic (and there are some helpful links to related topics in the forum) but it isn't specific to the ESP32 hardware.

Manu-8, 2019-10-21:
I think support for **WPA2-ENTERPRISE** is essential. Many school and university networks are using this for authentication. This would certainly boost the popularity of the system.
Actually i can't use micropython with my students for this reason.

igiagante, 2019-11-05:
@dpgeorge is there any chance to see soon something as `smart config`? Or any implementation that could help us to peer an app with an esp32?

dpgeorge, 2019-12-20:
v1.12 was tagged in 1f371947309c5ea6023b6d9065415697cbc75578

There are a lot of good suggestions here, which will need to form part of the longer-term roadmap of features.

alirezaimi, 2019-12-21:
please add mqtt async .

dlech, 2019-12-21:
Suggestion: use Kconfig for configuring MicroPython

Would you like to configure MicroPython like this?

[image]

Or this:

[image]

See #5444 for details.

tve, 2020-01-10:
IMHO getting uasyncio sorted out and stable should be a major feature for the next release. I know there's a lot in progress but it's such as core facility for a lot of users (I believe) that it should be on the roadmap explicitly.

goatchurchprime, 2020-01-11:
Is there a spectrum of benchmarks to stress the timing, capacity and memory footprint of the uasyncio library in place before it is rewritten/worked on? 

I don't see any here:  https://github.com/micropython/micropython/tree/master/tests/perf_bench

I must confess that I have never actually noticed a problem with this library in my use of it, so I don't know what people are complaining about. Maybe these benchmarks will show something up.  For all I know the code might be really gnarly and be due for a rewrite, but you have to be sure that when you change it you don't accidentally lose performance that is already there.

peterhinch, 2020-01-12:
@goatchurchprime uasyncio V2.0 has bugs, some [listed here](https://github.com/peterhinch/micropython-async/blob/master/FASTPOLL.md); in particular task cancellation has a major failing. The new version has cleaner code and according to my measurements on early code was nearly as fast as V2.0. It has since been [improved](https://github.com/micropython/micropython/pull/5332#issuecomment-571133960) for better performance. I think you can be confident that, when released, it will offer a substantial improvement.

goatchurchprime, 2020-01-12:
Boy, you just blink and stuff gets already done!  Please accept my apologies for not keeping up.  

When the test/benchmarks are released, I hope I can contribute one that exercises my use-case of scriptlets -- that of running code via the exec() or execfile() functions in cancellable tasks.  Just in case the code parser is or is not async optimized.

The code snippets are received  through MQTT.  There is no line between an API and an interpreted programming language, as many websites discovered when they got taken down by sql-inject attacks.

beyonlo, 2020-01-13:
> IMHO getting uasyncio sorted out, stable, and documented should be a major feature for the next release. I know there's a lot in progress but it's such as core facility for a lot of users (I believe) that it should be on the roadmap explicitly.

I totally agree :)

nalt, 2020-04-05:
Are there any plans for better DAC support? This repo has routines for tone generation, even playing WAV files: https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/wiki/dac

nevercast, 2020-06-08:
Has the ADC changes from v1.10 been forgotten? https://github.com/micropython/micropython/pull/4213 / https://github.com/micropython/micropython/issues/4356 ? 

I would like to offer `read_mv` and `read_v` for the ESP32 port. Since this is something I'm in need of for a project (calibrated values).

nevercast, 2020-06-08:
Also, since this is v2.0, can we break some stuff and see consistency on the use of Pin vs number? https://github.com/micropython/micropython/issues/4379

PyjamasBeforeChrist, 2020-06-10:
> Feel free to look into it. There's no problem to support more of the esp32 peripherals, it's just a matter of time to work on it.

Done for us by the looks?

https://forum.lvgl.io/t/example-for-working-with-a-rotary-encoder/1509/20

pidou46, 2020-09-11:
I guess "esp8266/esp32: switch to use LittleFS (saves 4k RAM, more robust filesystem)?" should be ticked ? isn't it ?

IhorNehrutsa, 2020-12-15:
> esp32 supports quadrature encoder.

ESP32: New features PCNT() and QUAD() #6639

PyjamasBeforeChrist, 2020-12-15:
> > esp32 supports quadrature encoder.
> 
> ESP32: New features PCNT() and QUAD() #6639

> > esp32 supports quadrature encoder.
> 
> ESP32: New features PCNT() and QUAD() #6639

Yes please this is absolutely needed