micropython/micropython #4822: Make micropython sleep on javascript backend does not does not freeze the browser

alanjds, 2019-05-29:
https://github.com/micropython/micropython/blob/d1dea4f57701b9464e3d6c45a5814d58fb0795f9/ports/javascript/library.js#L44-L46

Question: Changing this function to do some `await` inside it would fix the browser freezes in case of a Python `time.sleep()` call?

Example:
```js
 mp_js_ticks_ms: async function() {
     await sleep(0.0001);
     return (new Date()).getTime() - MP_JS_EPOCH; 
 }, 
```
having:
```js
function sleep(ms) {
  return new Promise(resolve => setTimeout(resolve, ms));
}
```

pmp-p, 2019-06-05:
maybe it would be better to write fully async code instead of trying to run sync functions on a platform that will nether support them.

dpgeorge, 2019-06-19:
> Question: Changing this function to do some `await` inside it would fix the browser freezes in case of a Python `time.sleep()` call?

I doubt it, but you could try to see what happens.

For a related issue see #3619.  See also Emterpreter which can help convert sync to async behaviour.

pmp-p, 2019-06-19:
you could also try micropython as a background thread communicating with main via RPC
It's early but there are various new flags to test if you're not dealing with GLES/SDL from C code 
```-s PROXY_TO_WORKER=1  -s BUILD_AS_WORKER=1``` 
that may allow blocking.

Anton-2, 2019-06-20:
To avoid this, I'm running MicroPython in a web worker, the web worker thread can be executing python code without locking the main thread responsible of UI updates. But you need to communicate between main and  worker thru postMessage....

rafi16jan, 2019-06-27:
You guys are achieving something impossible. Use emterpreter to skip an event loop (to get a promise result) https://github.com/emscripten-core/emscripten/wiki/Emterpreter

pmp-p, 2019-06-27:
@rafi16jan i see no call to emscripten_sleep() in javascript port, how do you service OS ( browser / gpu / sockets ) when you do  input() or sleep() in repl ?

rafi16jan, 2019-06-27:
@pmp-p https://github.com/rafi16jan/micropython/blob/master/ports/javascript/js.c

pmp-p, 2019-06-27:
@rafi16jan nice ! i was told emterpreter slows things a bit did you notice bad drop in performance ?

pmp-p, 2019-07-16:
@rafi16jan, On a side note i think the way you expose
```
MicroPython.js

MicroPython transmuted into Javascript (WASM) by Emscripten.

Official Repo https://github.com/micropython/micropython/tree/master/ports/javascript

Testing and Contribution Needed, feel free to make an issue or even better, a PR
What's New on 1.1

    New Async/Await or Promise API
    New Python classes to expose JS API and Objects like DOM API, XHR, Node.JS require, and etc
    New Python promise class to wait promises with emscripten_sleep, using Emterpreter
```
on https://www.npmjs.com/package/micropython is a bit misleading  since it's not using official sources
and real sources are on the small link on the right.

i'm quite sure that with the emterpreter performance drop, it is not a good way to promote micropython on npm, you should make two packages available 1) the official from @dpgeorge/@danicampora repo  2) yours.

alanjds, 2019-07-16:
Or keep as is and maybe be more proeminent on "this is a modified code"
disclaimer?

To be honest, I thought that it was based on the official code on 1st
glance.

pmp-p, 2019-07-17:
first, to solve the freeze problem for now it's simpler to turn time.sleep(),input() and other functions that are notouriously blocking to code that does nothing and output a warning both to stdout and stderr.

next come some ideas ( apart from trashing performance with emterpreting bytecode which yet another VM beetween  JS and WASM ones):

 - replace those specific calls with generators and asyncify module body of ```__main__``` => that's almost  ```python3.8 -m asyncio```. 
PRO: pythonic.
CON: still leave problem of script compatibility with missing "await" in front of offending functions...

 - keep ```__main__``` sync, but force those special case to generate exception to force VM to treat a block as a special async loop

basically save state then exit from mp_execute_bytecode and re-enter to resume later at the same point instead of running other async loop that would change state.

PRO: transparent  usage. real impression of running sync code ( eg sleep() would block asyncio )
CON: hard to implement and in a critical place ( vm.c  ). is it even possible ?

 - other ideas ?

actually VM can easily monitor generator instance objects like 
```
# awaited_ prefix could mean we want to be awaited also on standard calls and faking block on VM flow.
async def awaited_sleep(s):  
    ms = float(s) * 1000
    for step in (0, 1, 2):
        if step == 0:
            yield "'awaited_sleep'"
            continue
        if step == 1:
            try:
                raise RTAwait("async call in sync repl")
            except:
                continue
        if step == 2:
            try:
                yield from asyncio.sleep_ms(ms)
            except:
                continue

    embed.log("should have slept %s" % s)
    return None
```

would show something like 
```
micropython.js:1 comefrom 90
micropython.js:1 vm:1135 yield
micropython.js:1 "'awaited_sleep'"
micropython.js:1 vm:1328 except
micropython.js:1 vm:1135 yield
micropython.js:1 <SleepMs object at bd560>
micropython.js:1 vm:1328 except
micropython.js:1 vm:1328 except
micropython.js:1 should have slept 5
micropython.js:1 goto 85
```

@alanjds i'll keep you informed of my progress, though meanwhile i would be glad to hear from @* on that  since VM is a very fuzzy place for me.

pmp-p, 2019-07-17:
script used to generate ouput above 
```
<script type="text/µpython">#sync is the default
print("Hello World ..",end="")
sleep(5) #should not block at all as it's overloaded as a generator
print(". 1")
</script>
```
and it's ouput on repl
```
MicroPython 1160fbc-dirty on 2019-07-17; emscripten with wasm
>>> Hello World ..<generator object 'awaited_sleep' at bf800>
. 1

>>>
```

rafi16jan, 2019-07-17:
Sorry for late response. I'm busy in real life developing my own framework using a Python to JS transpiler, as now my framework is used in production I can give you a glimpse about 'Python in JS/Browser'  in a production code.

Before I use emterpreter, I realize that emterpreter generate it's own runtime to analyze and save the WASM code state and resumes after emscripten_sleep_with_yield is called. Not just it will add some overhead but the file size is also increasing, but still less than pyodide/pypy.js. Actually before I add emterpreter to micropython I manage to make some custom logic to queue callbacks (lambda or dynamic functions) but its syntactically awful (just like JS callbacks back in the days).

But I realize that for example micropython's string is a bit [slow](https://forum.micropython.org/viewtopic.php?t=1165) because it uses UTF-8 that are suitable for microcontrollers, but not for PCs or smartphones with fast ARM cpu. Later, eventually we should implement our custom string type to micropython either with C or even Javascript. And that's where I think that emterpreter shouldn't be the bottleneck. I think my usecase with micropython on WASM will be just a Python runtime on Browser/node.js. The http calls, filesystem APIs, and any other else that will block the I/O should be implemented on Javascript, and evented with emscripten_sleep_with_yield so the micropython code behave like async/await but without CPython 3's asyncio and the event loop (emterpreter is using JS's event loop, so there will be one event loop per web-worker). Also some asynchronous calls can be called before and awaited like JS's [Promise.all](https://www.freecodecamp.org/news/promise-all-in-javascript-with-example-6c8c5aea3e32/) to enable parallel I/O calls and this, based on my experience, is the power of JS and asynchronous programming.

If you compare the raw performance of plain micropython and micropython with emterpreter, ofcourse plain micropython will be faster (although not 10x faster) but the real battle is not there. JS event loop (especially V8) is proven in many cases beat Java or even C# (and the rest of .NET) on concurrency. And I'm ready to debate full-day about this as I experience it myself on my 2 projects in production.

@pmp-p @alanjds as of the disclaimer. Yes I will put a disclaimer that it is a modified code from the official repo, but I will fork any of the forks/repo that implement asynchronous sleep the best as I know my version is far from good becayse I'm not even fluent in C. Pardon my inabilities, I'm brought to the programming world with dynamic languages and that is one of my biggest regrets haha.

@pmp-p the first option even though it is Pythonic and follows CPython implementation, but it will cost us a headache of implementing another event loop. If you see the difference betwen JS event loop and asyncio event loop is that JS event loop is managed by each browser implementation (V8, gecko) out of the box. Even though we doesn't have control over it like asyncio does, but thats make it easier for us to think of another else (implementing custom pause and resume on micropython's C) as opposing to asyncio.get_event_loop() and then run_until_complete after every task.

The second option should be the best option as it is not adding any overhead (even though it is hard to implement) and will eliminate emterpreter use. The time it is implemented I will change my package to provide the WASM built from that implementation.

alanjds, 2019-07-17:
> Pardon my inabilities...

Just to clarify: thanks @rafi16jan for the NPM package. It is a very useful proof-of-concept of uPy on browser that I can send people a link to.

> * keep __main__ sync, but force those special case to generate exception to force VM to treat a block as a special async loop

Is what I would expect. I thought that just implementing `mp_hal_delay_ms` as some _await_ -ing call to a JS _async_ setTimeout would do the trick. _May_ I was naive on this JS context... :trollface:

pmp-p, 2019-07-17:
> I thought that just implementing mp_hal_delay_ms as some await -ing call to a JS async setTimeout would do the trick. 

it could , but only for a few milliseconds : then tab will appear seriously ill/locked and a few more browser will kill it somehow.

@dpgeorge, i tried to unwrap the bycode loop in the asynchronous js loop call, but when using C generators i got hit by **an emscripten raised js exception** as in 
https://github.com/emscripten-core/emscripten/issues/4042
But i don't get the "trying to jump back is an attempt to jump into a stack that has been unwound already" thing kripken is talking about :  isn't the VM loop a static state machine in Micropython specific case ?

pmp-p, 2019-07-18:
well now i know what he meant ...
so actually i managed to pause/resume vm but only on top off call stack, 
so that's for REPL or scripts run as '__main__'

as it seems mp_execute_bytecode() is called recursively, though i guess there are PR around that could help work around that ( like https://github.com/micropython/micropython/pull/4131? ).

pmp-p, 2019-08-07:
here's my finding it's really possible to make a register based VM with actual code ( using the stack less option, some registers and unwrapping the vm loop ) but it quickly reaches limits of wasm compiler on indirect calls and some bugs pop out of nowhere.
so maybe the solution is there https://kripken.github.io/blog/wasm/2019/07/16/asyncify.html

rafi16jan, 2019-08-08:
> here's my finding it's really possible to make a register based VM with actual code ( using the stack less option, some registers and unwrapping the vm loop ) but it quickly reaches limits of wasm [...]

https://github.com/emscripten-core/emscripten/wiki/Emterpreter#comparison-to-asyncify

pmp-p, 2019-08-08:
@rafi16jan no that's not the same asyncify, this one is new , give only about 50% overhead, but requires clang 10 llvm wasm backend ( javascript port won't even build with it in its current state ).

pmp-p, 2019-08-09:
it's possible to *compile* with 1.38.40-upstream (clang 10) using the no nlr patch from https://github.com/micropython/micropython/pull/4131.
Running and getting testsuite to pass is another story before trying the new asyncfy method.

pmp-p, 2020-05-03:
Using PYSTACK and a **custom no_nlr py/vm.c or a dual vm model** that allow interrupts,  that problem is solved on baremetal wasm/wasi ( no workers , no stack snapshotting , no slowdown ).

dpgeorge, 2024-07-26:
> Wondering if there's a recommended solution now (in 2024)

Nowadays one can use asyncify, that allows the running (Emscripten compiled) program to run indefinitely and yield to the browser/JS runtime when needed.

Asyncify is enabled by default on the webassembly port, for the standard build variant.  For the pyscript variant asyncify is disabled to increase performance and reduce WASM size.

I'll close this issue because asyncify is the solution to it.

closed 2024-07-26 by dpgeorge