micropython/micropython #4823: [ESP32] OSError: 201, when trying to bind to random TCP port

devxpy, 2019-05-30:
```
>>> import usocket as socket
>>> sock = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
>>> sock.bind(('0.0.0.0', 0))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: 201
```

```
>>> uos.uname()
(sysname='esp32', nodename='esp32', release='1.11.0', version='v1.11-2-ga4f1d8275 on 2019-05-30', machine='ESP32 module with ESP32')
```

Sorry for creating a new issue. I cannot find any good resource on doing this the right way in micropython.

dpgeorge, 2019-05-31:
Thanks for the report.  It looks like the cause is lwIP's getaddrinfo which returns an error when passed in 0 for the port.  But lwIP does accept 0 as a port to bind to.  So to fix it it looks like there needs to be a special case in esp32's socket implementation when looking up an address with port=0.

dpgeorge, 2019-06-04:
See #4830 for a fix.

But I'm not sure how useful this is... @devxpy how would you use this feature?

devxpy, 2019-06-04:
Whoa! Thanks a ton for looking into it.

I'm trying to create a networking library for micropython. 

It's fairly simple -- the goal is to make network config irrelevant to communicate between devices on a network.

Involves opening up a UDP socket on a known port for peer discovery (using broadcasts) and then opening up a TCP channel or two on a random port to communicate over.

Will surely post here once [we're](https://github.com/pycampers) ready to make it open :)

dpgeorge, 2019-06-04:
> Involves opening up a UDP socket on a known port for peer discovery (using broadcasts) and then opening up a TCP channel or two on a random port to communicate over.

Ok, but if the TCP port is random how do you find out which one it is so the other peer can connect to it?

(BTW, I'm sure you already know about mDNS)

devxpy, 2019-06-04:
Well, the port is communicated over the broadcast.

[Here's a gist](https://gist.github.com/devxpy/49e2ce2cad87ef86e692cf766fcce828#file-unetwork-py-L367
).

I just never got around to using mDns. UDP broadcasts seemed simple enough for my use-cases.
I should revisit it, I guess?

dpgeorge, 2019-06-04:
> Well, the port is communicated over the broadcast.

Right, so you use `s.getsockname()` to get the randomly generated port number (along with IP).  That method is not implemented in MicroPython...

> I just never got around to using mDns. UDP broadcasts seemed simple enough for my use-cases.
> I should revisit it, I guess?

I'd suggest trying it out, it may do everything you need/want, and it already has support on PC operating systems.

devxpy, 2019-06-04:
Ah, dang! 

Thanks for the suggestion on mDns. Will be sure to try it out.

Should I close this?

dpgeorge, 2019-06-04:
Let's leave this open for a bit, it's still an error.

dpgeorge, 2019-06-25:
Ok let's close this, since there's not really any use (at this stage) for a random port number.

closed 2019-06-25 by dpgeorge