micropython/micropython #4824: non-blocking socket blocks and raises EAGAIN

c0d3z3r0, 2019-05-30:
from https://github.com/pfalcon/micropython/issues/30

Problem: a non-blocking socket seems to block and raises EAGAIN (was EWOULDBLOCK some time ago)

~~~
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== import socket as _socket
=== ai = _socket.getaddrinfo("127.0.0.1", 85, 0, _socket.SOCK_STREAM)
=== ai = ai[0]
=== s = _socket.socket(ai[0], ai[1], ai[2])
=== s.setblocking(False)
=== s.setsockopt(_socket.SOL_SOCKET, _socket.SO_REUSEADDR, 1)
=== s.bind(ai[-1])
=== s.listen(0)
=== 
>>> s2, client_addr = s.accept()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 11] EAGAIN
~~~

Catching EAGAIN works but this only circumvents that bug...

~~~
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== import socket as _socket
=== ai = _socket.getaddrinfo("127.0.0.1", 85, 0, _socket.SOCK_STREAM)
=== ai = ai[0]
=== s = _socket.socket(ai[0], ai[1], ai[2])
=== s.setblocking(False)
=== s.setsockopt(_socket.SOL_SOCKET, _socket.SO_REUSEADDR, 1)
=== s.bind(ai[-1])
=== s.listen(0)
=== 
>>> try:
>>>   s2, client_addr = s.accept()
>>> except OSError as e:
>>>   if e.args[0] == uerrno.EAGAIN:
>>>     continue
>>>

~~~

c0d3z3r0, 2019-05-31:
Update: I tried this on CPython, where I get a BlockingIOError - which seems to be correct according to the python documentation.
Maybe this is no bug but intended behaviour. Can anyone confirm this?

pfalcon, 2019-05-31:
In MicroPython (following POSIX), calling accept() on non-blocking server socket: a) should not block. Then it should either b) return a client connection socket, if such a connection was pending, or c) raise OSError(EAGAIN) otherwise. Any realistic usage would be first waiting for "read" event on such a server socket (using uselect.poll()), and if event is signaled, only then call .accept(). In such a case, this .accept() should return a pending connection presence which of was signaled, and not raise EAGAIN.

c0d3z3r0, 2019-05-31:
@pfalcon Ok, great - so there was some misunderstanding on my side... there is no bug in micropython but there is in uasyncio... preparing a PR und closing this

pfalcon, 2019-05-31:
What makes you think there's a bug in uasyncio?

c0d3z3r0, 2019-05-31:
See my very first test here https://github.com/pfalcon/micropython/issues/30
run() should not raise, should it?

pfalcon, 2019-06-01:
> See my very first test here pfalcon/micropython#30
run() should not raise, should it?

Yes, it shouldn't. "it" == socket.accept() call, in the given conditions, from modlwip module running on esp8266. And that's apparently where the culprit is.

c0d3z3r0, 2019-06-01:
@pfalcon I'm sorry, I really don't get this...
>"calling accept() on non-blocking server socket: a) should not block"

vs.

>>run() should not raise, should it?
>
>Yes, it shouldn't. "it" == socket.accept()

Instead of rasing EAGAIN, what exactly should happen on accept()? return (None, None)? That would break uasyncio.start_server here https://github.com/pfalcon/micropython-lib/pull/33/commits/9cdd4ab0f63eaac699ac1cc278cc8167e870fb8e#diff-c502b73effb6927190c86fdaab27ee84L253

> In MicroPython (following POSIX)

Uhm... POSIX:2001 says:
*If the listen() queue is empty of connection requests and O_NONBLOCK is set on the file descriptor for the socket, accept() shall fail and set errno to [EAGAIN] or [EWOULDBLOCK].*

From linux man(2) accept:
*If no pending connections are present on the queue, and the socket is not marked as nonblocking, accept() blocks the caller until a connection is present. ***If the socket is marked nonblocking and no pending connections are present on the queue, accept() fails with the error EAGAIN or EWOULDBLOCK.****

CPython (yes, I know that micropython != CPython) raises BlockingIOError in this example:
~~~
import socket 
s=socket.socket(socket.AF_INET, socket.SOCK_STREAM)                     
s.bind(('', 1234))
s.listen(1)
s.setblocking(0)                                                        
conn, addr = s.accept()
---------------------------------------------------------------------------
BlockingIOError                           Traceback (most recent call last)
<ipython-input-8-3b24cc35f468> in <module>
----> 1 conn, addr = s.accept()

/usr/lib/python3.7/socket.py in accept(self)
    210         For IP sockets, the address info is a pair (hostaddr, port).
    211         """
--> 212         fd, addr = self._accept()
    213         sock = socket(self.family, self.type, self.proto, fileno=fd)
    214         # Issue #7995: if no default timeout is set and the listening

BlockingIOError: [Errno 11] Resource temporarily unavailable
~~~

pfalcon, 2019-06-01:
> Instead of rasing EAGAIN, what exactly should happen on accept()?

You would need to look at the context of more than 1 line (say, 3 lines), and match that with the description provided above.

uasyncio/__init__.py:
~~~
        yield IORead(s)
        if DEBUG and __debug__:
            log.debug("start_server: After iowait")
        s2, client_addr = s.accept()
~~~

After `yield IORead(s)` returns, the only expected behavior of s.accept() is to return new client connection. Because the fact that `yield IORead(s)` returns, means that there's a pending client connection.

c0d3z3r0, 2019-06-01:
Ouch. Thank you, I simply overlooked that line... o.O

dpgeorge, 2019-06-02:
I'm not sure exactly what problem is reported here.  If you could provide a test case that shows the problem, and what the expected behaviour should instead be, that would be helpful.

c0d3z3r0, 2019-06-02:
Well, I‘m still a bit confused but in summary it seems, socket.accept() is raising EAGAIN where it should not, specfically when using IORead(s).
Maybe @pfalcon can give a more precise description?
See the linked issue in pfalcons repo, too.

c0d3z3r0, 2019-06-04:
Hi there. Unfortunately I sent my esp8266 to silicon heaven yesterday due to a broken voltage regulator........ I now bought an esp32 as replacement so I'm sorry I cannot help testing/debugging this any further :S

JuniorJPDJ, 2019-08-08:
I can help if you'll tell me what to do.
I'm trying to use non-blocking sockets (uasyncio especially) and I'm getting EAGAIN when trying to accept connection.
It's called after `yield IORead(s)` so there should be new client to accept.

JuniorJPDJ, 2019-08-08:
https://github.com/c0d3z3r0/micropython-lib/commit/b876ecbd297b133708c313a694f1bfa175d6eb97
There's @c0d3z3r0 's workaround for this, which clearly explains the bug imo.

JuniorJPDJ, 2019-08-10:
I've spotted (probably) the same bug.
I tried to use `select.poll`, but my bind socket got event twice, when I connected once and It just blocked on s.accept().
My sample code:
```
        ss = socket.socket()
        ss.bind((addr, port))
        ss.listen(5)

        poll = select.poll()
        poll.register(ss, select.POLLIN)

        buf = {}

        while True:
            for s, e, *_ in poll.poll():
                print("Something happened.", s, e, _)
                del _
                if s == ss and e == select.POLLIN:
                    s, _ = ss.accept()   # I'm getting blocked here in second iteration
                    del _
                    poll.register(s, select.POLLIN)
                    print("New connection!")
                    continue
```

output:
```
% micropython webserver.py
Something happened. <_socket 3> 1 []
New connection!
Something happened. <_socket 3> 1 []
```

JuniorJPDJ, 2019-08-10:
Aaalso.. when I set `ss` to non-blocking, I've infinite loop with only that socket getting events..
```
...
Something happened. <_socket 3> 1 []
Something happened. <_socket 3> 1 []
Something happened. <_socket 3> 1 []
Something happened. <_socket 3> 1 []
Something happened. <_socket 3> 1 []
Something happened. <_socket 3> 1 []
Something happened. <_socket 3> 1 []
Something happened. <_socket 3> 1 []
Something happened. <_socket 3> 1 []
Something happened. <_socket 3> 1 []
Something happened. <_socket 3> 1 []
Something happened. <_socket 3> 1 []
...
```

JuniorJPDJ, 2019-08-11:
@dpgeorge this is not port-esp8266 specific.
I'd that also on unix port.

dpgeorge, 2019-08-17:
@JuniorJPDJ I can't reproduce any error here.  I use the following test script:
```python
try:
    # MicroPython
    import usocket as socket, uselect as select
    def issock(s1, s2): return s1 == s2
except:
    # CPython
    import socket, select
    def issock(s1, s2): return s1 == s2.fileno()

def test(ai, blocking):
    ss = socket.socket()
    ss.bind(ai)
    ss.listen(5)
    ss.setblocking(blocking)

    poll = select.poll()
    poll.register(ss, select.POLLIN)

    buf = {}

    i=0
    while True:
        for s, e, *_ in poll.poll():
            i += 1
            print(i, "Something happened.", s, e, _)
            del _
            if issock(s, ss) and e == select.POLLIN:
                s, _ = ss.accept()   
                del _
                poll.register(s, select.POLLIN)
                print("New connection!")
                continue

test(socket.getaddrinfo('0.0.0.0', 8000)[0][-1], True)
```

It runs the same on CPython 3.7, and latest MicroPython on unix and stm32 (with networking via lwip).

The output is (on MicroPython unix):
```
1 Something happened. <_socket 3> 1 []
New connection!
2 Something happened. <_socket 4> 1 []
3 Something happened. <_socket 4> 1 []
4 Something happened. <_socket 4> 1 []
...
```

JuniorJPDJ, 2019-09-12:
```
import socket
import select
import re

class WebServer:
    """Only GET is supported, protocol line (including url) is limited to 1kiB"""
    def __init__(self):
        self._endpoints = {}

    def register_endpoint(self, func, uri, method="GET"):
        self._endpoints[(uri, method)] = func
        return func

    def run(self, addr="0.0.0.0", port=80):
        ss = socket.socket()
        ss.bind((addr, port))
        ss.listen(5)

        poll = select.poll()
        poll.register(ss, select.POLLIN)

        buf = {}

        while True:
            for s, e, *_ in poll.ipoll():
                print("Something happened.", s, e, _)
                del _
                if s == ss and e == select.POLLIN:
                    s, _ = ss.accept()
                    del _
                    poll.register(s, select.POLLIN)
                    print("New connection!")
                    continue

                if e != select.POLLIN:
                    poll.unregister(s)
                    s.close()
                    if s in buf:
                        del buf[s]
                    continue

                data = s.recv(1024)
                if s in buf:
                    buf[s] += data
                    data = buf[s]
                    if len(data) > 1024:
                        poll.unregister(s)
                        s.close()
                        if s in buf:
                            del buf[s]
                        continue
                else:
                    buf[s] = data

                nl = data.find(b'\n')
                if nl < 0:
                    if data.count(b' ') > 2:
                        poll.unregister(s)
                        s.close()
                        if s in buf:
                            del buf[s]
                    continue

                line = data[:nl]
                method, resource, proto = line.split(b' ', 3)
                self._handle_request(method, resource, s)

    def _handle_request(self, method, resource, socket):
        print(method, resource)

if __name__ == "__main__":
    srv = WebServer()
    srv.run(port=8888)
```
This is my whole code.
It blocks itself on `ss.accept` line when blocking. My micropython version is latest from git (ArchLinux AUR micropython-git package installed).

When setting `ss` to non-blocking I'm getting instant:
```
Something happened. <_socket 3> 1 []
Traceback (most recent call last):
  File "webserver.py", line 76, in <module>
  File "webserver.py", line 31, in run
  File "/usr/lib/micropython/socket.py", line 45, in accept
OSError: [Errno 11] EAGAIN
```

dpgeorge, 2019-09-12:
@JuniorJPDJ it looks like you're using the `socket` and `select` modules, which means you must have some additional libraries installed (they are not included by default, only `usocket` and `uselect` are).

If you run the following you can see which libraries are imported:
```
import socket, select
print(socket.__file__, select.__file__)
```
It really depends which socket/select implementation you use.  It might be better to just use:
```
import usocket as socket
import uselect as select
```

peterhinch, 2019-09-12:
The EAGAIN error tells you that the socket would block - so you should try again later.

robert-hh, 2019-09-12:
How so? Importing socket and select works on a plain vanilla ESP8266.
```
>>> import socket
>>> import select
>>> print(socket.__file__, select.__file__)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'module' object has no attribute '__file__'
```

dpgeorge, 2019-09-12:
> How so? Importing socket and select works on a plain vanilla ESP8266.

True.  I meant on the unix port, which it seems what's being used considering it was installed from ArchLinux AUR.  The aim here is to get a common baseline for the test, so I can reproduce it.

JuniorJPDJ, 2019-10-16:
> @JuniorJPDJ it looks like you're using the `socket` and `select` modules, which means you must have some additional libraries installed (they are not included by default, only `usocket` and [...]

>>> import socket, select
>>> print(socket.__file__, select.__file__)
/usr/lib/micropython/socket.py /usr/lib/micropython/select.py
>>>
:o

closed 2026-05-01 by dpgeorge