micropython/micropython #4827: OSError: [Errno 113] EHOSTUNREACH while listen for connection

nedoskiv, 2019-06-01:
Hello,
I was using so far older micropython (1.10 official ) release, but I was testing some stuff and switched to one of the latest:

MicroPython v1.10-380-g6cf4e9675 on 2019-05-23; ESP32 module with ESP32

problem is that I got a simple web server using usocket to listen for connections:

my code is large i just gonna show how I use socket:

```
s = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
s.bind(('', 80))
s.listen(3)
conn, addr = s.accept()
```

with older version of micropython (1.10 official) I never got a problem, but with this one from time to time i got:
OSError: [Errno 113] EHOSTUNREACH
failing on line with:
`conn, addr = s.accept()`
google it and it may be dns related, but there is no dns involved when you listen for connections.

dpgeorge, 2019-06-04:
Since v1.10 there weren't any significant changes to the esp32's socket module, at least none that would give these kinds of problems.

What did change was the IDF which was updated, and the system was switched to use dual core mode.  So there problems could come from there.

Do you use external SPIRAM on your board?  If so that is known to still have some issues.

nedoskiv, 2019-06-04:
nope no SPIRAM, I can confirm that on old 1.10 release these not a single error that I mention above, no socket or conn receive problem.

dpgeorge, 2019-06-04:
> OSError: [Errno 113] EHOSTUNREACH

This could mean that the wifi has disconnected (and is maybe attempting to reconnect).

> I can confirm that on old 1.10 release these not a single error that I mention above, no socket or conn receive problem.

Ok.  But, without a fully contained script that can reproduce this error reliably, or someone else who has the same problem, not much progress can be made here.

nedoskiv, 2019-06-04:
> 
> 
> > OSError: [Errno 113] EHOSTUNREACH
> 
> This could mean that the wifi has disconnected (and is maybe attempting to reconnect).

Did not see any wifi reconnect messages in console during that error. And why that error is reised when it is listening not connecting to an host? On 1.10 it start listening without wifi connection (in my case as client) gonna test it on 1.11 when switch back to it.

> 
> > I can confirm that on old 1.10 release these not a single error that I mention above, no socket or conn receive problem.
> 
> Ok. But, without a fully contained script that can reproduce this error reliably, or someone else who has the same problem, not much progress can be made here.

Well hope someone who have lot of incoming tcp connections report that too.

I remember that i have posted an issue about 1.10 where socket connection is maded to DNS unresolved hostname,  it restart system. You fixed it, not tested it yet, but can that fix made the problem here?

milkshakespoon, 2019-06-04:
Wondering if this is related to my issue... based on the fact you're suggesting disconnection of WiFi but not seeing anything in the console - #4748

nedoskiv, 2019-06-05:
from my point of view - no,  I monitor other processes in console, they do remote connections and report errors. None of them reported so it is different issue

ned-pcs, 2023-06-12:
I have seen this behavior in [my recent work on `mip`](https://github.com/micropython/micropython-lib/pull/676): in the middle of reading data from a socket I would occasionally see an `EHOSTUNREACH` error. I wrapped the whole request and data copy in an error handler, and on an `OSError` did a `gc.collect()` and a 1 second delay and retry. This has apparently cured those issues. I did test for `isconnected()` but that always returned `True`.