micropython/micropython #4828: THREADS: MemoryError: memory allocation failed

nedoskiv, 2019-06-02:
Hello,

First I want to say sorry about posting kind of issue since they are hard to reproduce.
Here is the situation, I got few threads started, in one is web server in another adc monitor, in third is remote logger. from time to time I got error:

MemoryError: memory allocation failed, allocating 1821 bytes

while trying to fix that i place some free heap monitoring and there is a lot of heap, but it still fails with memoty allocation error.
I also place gc_collect() in many places, it improves situation a bit, but it still fails with same error.
Increased _thead stack size, no effect.
How is the things working with threads, why fail to allocate memory since there is a lot? Is there a way to reserve some memory for a thread?

(sysname='esp32', nodename='esp32', release='1.10.0', version='v1.10-380-g6cf4e9675 on 2019-05-23', machine='ESP32 module with ESP32')

dpgeorge, 2019-06-04:
It's hard to say much here without knowing a lot more about the code that is running.  At the very least, know what call is raising the MemoryError.

> there is a lot of heap, but it still fails with memoty allocation error.

`micropython.mem_info(1)` will show the heap layout.

> Is there a way to reserve some memory for a thread?

You can preallocate objects like arrays, lists, dicts and keep them around to prevent the need to reallocate them every time in a loop.

nedoskiv, 2019-06-04:
well I switched back to 1.10 it works stable for me.
Problem for me is that I cannot debug where problem come from. Managed to isolate most recent lines who cause the problem, 
One is ujson.dumb () on array, 
other is composing  text  from multiple variables some coming from function returns.

I can upload my complete code if you want to run it