micropython/micropython #4829: mpy-tool.py qstr linking fails with @native on a member method

jimmo, 2019-06-04:
A simple way to repro is to add `@micropython.native` to any of the methods in dht.py (e.g. DHT22.temperature) and build for PYBV11.

I think in `read_raw_code`, `off = read_uint(f, qstr_win)` shouldn't be passing `qstr_win`.

But then this uncovered a different problem with `_link_qstr`. Depending on which other .mpy files were being passed to mpy-tool. If the last .mpy to be processed was non-native (i.e. native_arch == 0), then it hits the `assert 0` if there are any kind != 0 qstrs. Alternatively, if the last .mpy was the one with native code (e.g. dht.mpy, native_arch=5), it looks like the branch for ARM can't work (`is_obj` and `i` don't exist).

dpgeorge, 2019-06-04:
Thanks for the report, I can confirm it.

The main problem is that most `config` values should be per-mpy file, not global (as they are currently treated).

dpgeorge, 2019-06-04:
Fixed by faf3d3e9e93a71b3947bf563e4ffe13cbd053cb8

hacksterous, 2019-06-05:
Build now goes through, but I get a runtime error when accessing my function:

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: function missing required positional argument #31
```

I see the these warnings during build:

```
build-BLACK_F407VE/frozen_mpy.c:20725:83: warning: excess elements in array initializer
     0x00, 0x31, 0xfb, 0x6e, 0x98, 0x47, 0x05, 0x90, 0x00, 0xf0, 0x00, 0xb8, 0x0b, 0x98, 0x00, 0x28,
                                                                                   ^~~~
build-BLACK_F407VE/frozen_mpy.c:20725:83: note: (near initialization for 'fun_data_mpap__lt_module_gt__mpap___truediv__')
build-BLACK_F407VE/frozen_mpy.c:20725:89: warning: excess elements in array initializer
     0x00, 0x31, 0xfb, 0x6e, 0x98, 0x47, 0x05, 0x90, 0x00, 0xf0, 0x00, 0xb8, 0x0b, 0x98, 0x00, 0x28,
                                                                                         ^~~~
build-BLACK_F407VE/frozen_mpy.c:20725:89: note: (near initialization for 'fun_data_mpap__lt_module_gt__mpap___truediv__')
build-BLACK_F407VE/frozen_mpy.c:20725:95: warning: excess elements in array initializer
     0x00, 0x31, 0xfb, 0x6e, 0x98, 0x47, 0x05, 0x90, 0x00, 0xf0, 0x00, 0xb8, 0x0b, 0x98, 0x00, 0x28,
                                                                                               ^~~~
build-BLACK_F407VE/frozen_mpy.c:20725:95: note: (near initialization for 'fun_data_mpap__lt_module_gt__mpap___truediv__')
build-BLACK_F407VE/frozen_mpy.c:20726:5: warning: excess elements in array initializer
     0x00, 0xf0, 0x05, 0x80, 0x7b, 0x69, 0x98, 0x47, 0xd7, 0xf8, 0x84, 0x30, 0x98, 0x47, 0x05, 0x98,
     ^~~~
build-BLACK_F407VE/frozen_mpy.c:20726:5: note: (near initialization for 'fun_data_mpap__lt_module_gt__mpap___truediv__')
build-BLACK_F407VE/frozen_mpy.c:20726:11: warning: excess elements in array initializer
     0x00, 0xf0, 0x05, 0x80, 0x7b, 0x69, 0x98, 0x47, 0xd7, 0xf8, 0x84, 0x30, 0x98, 0x47, 0x05, 0x98,
           ^~~~
build-BLACK_F407VE/frozen_mpy.c:20726:11: note: (near initialization for 'fun_data_mpap__lt_module_gt__mpap___truediv__')
build-BLACK_F407VE/frozen_mpy.c:20726:17: warning: excess elements in array initializer
     0x00, 0xf0, 0x05, 0x80, 0x7b, 0x69, 0x98, 0x47, 0xd7, 0xf8, 0x84, 0x30, 0x98, 0x47, 0x05, 0x98,
                 ^~~~
build-BLACK_F407VE/frozen_mpy.c:20726:17: note: (near initialization for 'fun_data_mpap__lt_module_gt__mpap___truediv__')
build-BLACK_F407VE/frozen_mpy.c:20726:23: warning: excess elements in array initializer
     0x00, 0xf0, 0x05, 0x80, 0x7b, 0x69, 0x98, 0x47, 0xd7, 0xf8, 0x84, 0x30, 0x98, 0x47, 0x05, 0x98,
                       ^~~~
build-BLACK_F407VE/frozen_mpy.c:20726:23: note: (near initialization for 'fun_data_mpap__lt_module_gt__mpap___truediv__')
build-BLACK_F407VE/frozen_mpy.c:20726:29: warning: excess elements in array initializer
     0x00, 0xf0, 0x05, 0x80, 0x7b, 0x69, 0x98, 0x47, 0xd7, 0xf8, 0x84, 0x30, 0x98, 0x47, 0x05, 0x98,
                             ^~~~
build-BLACK_F407VE/frozen_mpy.c:20726:29: note: (near initialization for 'fun_data_mpap__lt_module_gt__mpap___truediv__')
build-BLACK_F407VE/frozen_mpy.c:20726:35: warning: excess elements in array initializer
     0x00, 0xf0, 0x05, 0x80, 0x7b, 0x69, 0x98, 0x47, 0xd7, 0xf8, 0x84, 0x30, 0x98, 0x47, 0x05, 0x98,
                                   ^~~~
build-BLACK_F407VE/frozen_mpy.c:20726:35: note: (near initialization for 'fun_data_mpap__lt_module_gt__mpap___truediv__')
build-BLACK_F407VE/frozen_mpy.c:20726:41: warning: excess elements in array initializer
     0x00, 0xf0, 0x05, 0x80, 0x7b, 0x69, 0x98, 0x47, 0xd7, 0xf8, 0x84, 0x30, 0x98, 0x47, 0x05, 0x98,
                                         ^~~~
build-BLACK_F407VE/frozen_mpy.c:20726:41: note: (near initialization for 'fun_data_mpap__lt_module_gt__mpap___truediv__')
build-BLACK_F407VE/frozen_mpy.c:20726:47: warning: excess elements in array initializer
     0x00, 0xf0, 0x05, 0x80, 0x7b, 0x69, 0x98, 0x47, 0xd7, 0xf8, 0x84, 0x30, 0x98, 0x47, 0x05, 0x98,
                                               ^~~~
build-BLACK_F407VE/frozen_mpy.c:20726:47: note: (near initialization for 'fun_data_mpap__lt_module_gt__mpap___truediv__')
build-BLACK_F407VE/frozen_mpy.c:20726:53: warning: excess elements in array initializer
     0x00, 0xf0, 0x05, 0x80, 0x7b, 0x69, 0x98, 0x47, 0xd7, 0xf8, 0x84, 0x30, 0x98, 0x47, 0x05, 0x98,
                                                     ^~~~
build-BLACK_F407VE/frozen_mpy.c:20726:53: note: (near initialization for 'fun_data_mpap__lt_module_gt__mpap___truediv__')
build-BLACK_F407VE/frozen_mpy.c:20726:59: warning: excess elements in array initializer
     0x00, 0xf0, 0x05, 0x80, 0x7b, 0x69, 0x98, 0x47, 0xd7, 0xf8, 0x84, 0x30, 0x98, 0x47, 0x05, 0x98,
                                                           ^~~~
build-BLACK_F407VE/frozen_mpy.c:20726:59: note: (near initialization for 'fun_data_mpap__lt_module_gt__mpap___truediv__')
build-BLACK_F407VE/frozen_mpy.c:20726:65: warning: excess elements in array initializer
     0x00, 0xf0, 0x05, 0x80, 0x7b, 0x69, 0x98, 0x47, 0xd7, 0xf8, 0x84, 0x30, 0x98, 0x47, 0x05, 0x98,
                                                                 ^~~~
build-BLACK_F407VE/frozen_mpy.c:20726:65: note: (near initialization for 'fun_data_mpap__lt_module_gt__mpap___truediv__')
build-BLACK_F407VE/frozen_mpy.c:20726:71: warning: excess elements in array initializer
[... cut ...]
build-BLACK_F407VE/frozen_mpy.c:20730:33: warning: excess elements in array initializer
     MP_QSTR_mpap_dot_py & 0xff, MP_QSTR_mpap_dot_py >> 8,
                                 ^~~~~~~~~~~~~~~~~~~
build-BLACK_F407VE/frozen_mpy.c:20730:33: note: (near initialization for 'fun_data_mpap__lt_module_gt__mpap___truediv__')
build-BLACK_F407VE/frozen_mpy.c:20731:5: warning: excess elements in array initializer
     0xff,
     ^~~~
build-BLACK_F407VE/frozen_mpy.c:20731:5: note: (near initialization for 'fun_data_mpap__lt_module_gt__mpap___truediv__')
```

dpgeorge, 2019-06-05:
@hacksterous can you share (link to) the frozen files that lead to this error, or at least provide a subset that still have the error?

hacksterous, 2019-06-05:
Here is the [link](https://github.com/hacksterous/mpap/blob/master/mpap.py)
I have one method using the native decorator.
Sorry, I could not get a smaller subset to show the warnings.

dpgeorge, 2019-06-11:
Thanks, I was able to reproduce the error by freezing mpap.py with a native decorator on the `__truediv__` method.  Should be fixed now by 9d3031cc9d888321b122d7e92491ba045727ac32

closed 2019-06-11 by dpgeorge