micropython/micropython #4832: Factory reset from code

TravisJoe, 2019-06-05:
I believe factory reset should be available as an alternative to USR. USR might not be practical or accessible for custom boards or how the device might be enclosed or accessed. 

This might also be related to ways to clear/format/erase the filesystem from code. 

Basically recovery and housecleaning.

peterhinch, 2019-06-05:
This could readily be achieved in Python.

dpgeorge, 2019-06-12:
As @peterhinch mentions this should be achievable in Python.  But because of the way a pyboard exposes its internal flash filesystem over USB MSC, with a partition table, it's not possible to format the flash in the usual way with `uos.VfsFat.mkfs(pyb.Flash())`.

TravisJoe, 2019-06-30:
Sorry, traveling and busy with work.

I do not disagree that it can and should be done using code. And it is a rare enough need that bloating Micropython is likely not justified. I think in some of these cases app notes or something to give somebody else the "ideal" method to accomplish this using python is helpful. 

Something like wanting/needing factory reset is often not at a convenient time to research and learn how to do it. 

Some of the documentation for the STM, in particular the Pyboard, rely heavily on the Pyboard specific LEDs and USR buttons to get out of trouble. On a development board these are great, but on a custom board or production situation these might not be practical to use or have.

One idea would be to just add more methods/tutorials on erasing internal and external flash and otherwise ways to reset the device to start from scratch. 
https://docs.micropython.org/en/latest/pyboard/tutorial/reset.html

dimpolo, 2020-07-23:
I would also like to see this.
I tried this https://docs.micropython.org/en/latest/reference/filesystem.html#fat
```python
import os, pyb
os.umount('/flash')
os.VfsFat.mkfs(pyb.Flash(start=0))
os.mount(pyb.Flash(start=0), '/flash')
os.chdir('/flash')
```
However this does not set the USB MSC name to PYBFLASH.

Unfortunately the file systems gets corrupted every now and then for me and having something like `pyb.factory_reset()` would be great for this.

peterhinch, 2020-07-23:
@TravisJoe A factory reset is most needed when such mayhem has been caused that there is no REPL. Then you need a hardware approach as per the Pyboard USR. On production kit you might want to consider an add-on piece of test hardware which can invoke a factory reset.

@dimpolo File system corruption on a Pyboard is almost always caused by MSC mode. If Python code modifies the contents in any way a conflict with the PC host arises which usually leads to corruption. This has been discussed at length on [the forum](http://forum.micropython.org/index.php) where this issue regularly arises.

I always disable MSC mode on Pyboards. I would describe unexplained corruption as virtually nonexistent.

TravisJoe, 2020-07-23:
@peterhinch 
I agree that this needs to be a hardware trigger. I think my ask would be one that did not rely on the LED state and USR button and was programmable when building firmware.

@dimpolo 
I had corruption issues, which made me really worried about long term stability, on a product when I realized that it was people poorly handling the USB as Peter mentioned. Once that was stopped I had essentially no issues. One idea I have played with is a command that can be called from REPL that would change turn off and on the mountable filesystem, which would require a restart, to avoid accidental miss-use of the mountable filesystem.

Since handling USB better, and also moving my flash filesystem external to the MCU using QSPI (likely better at handling frequent reading and writing), I cannot remember a corruption event in 2 year even with several hundred devices.

dimpolo, 2020-07-24:
I agree that most importantly I need to figure out why the corruption is happening.
Until then, having factory reset access would be nice :)
Are there arguments against having this feature or alternativly the ability to set a name with `os.VfsFat.mkfs(pyb.Flash(start=0))`?

Here's what's currently happening for me (probaby not relevant to this thread though):
I have a custom board connected to a test bench. The test bench runs an automated process:
* flash micropython over dfu
* reset the board
* copy code over USB
* reset the board
* check if all connected peripherals work

There's 10 seconds delay between each step, however sometimes the USB still gets corrupted.
The REPL still works. The Python code does not write to the flash.
I tried using rshell but ran into different problems.

I'll see if I can recreate a MRE where this happens and maybe post a separate issue.

robert-hh, 2020-07-24:
Did you eject the drive on the PC which maps the Pyboard between copying the code and resetting the board? That is essential to cause cached data to be written.

TravisJoe, 2020-07-24:
@robert-hh 
If you are asking me, essentially yes. The problem was more of I could not keep other people using the devices when updating them diligent on managing the USB.

peterhinch, 2020-07-25:
@dimpolo I can't explain why your corruption occurs. However you say

> The REPL still works.

In this case you have full remote access to the board and can do the equivalent of a factory reset. You can put it into DFU mode or format the filesystem using the **pyboard.py** utility. As an example the following script puts a Pyboard into DFU mode:
```python
#! /usr/bin/env python

# Called from buildpyb
# Put pyboard into DFU mode.

import sys
import os
mp = os.getenv('MPDIR')
sys.path.append(''.join((mp, '/tools')))
import pyboard

errmsg = 'pyb_boot error: usage pyb_boot device.'
def main():
    if len(sys.argv) < 2:
        print(errmsg)
        sys.exit(1)
    device = sys.argv[1]
    pyb=pyboard.Pyboard(device)
    pyb.enter_raw_repl()
    try:
        pyb.exec_raw('pyb.bootloader()')
        print('Failed to enter DFU mode')
    except Exception:  # It will throw one!
        pass

if __name__ == "__main__":
    main()
```

dimpolo, 2020-07-27:
@robert-hh I'll give this a try, thanks.
@peterhinch I'd like to reset the filesystem without going into DFU.

AeroSynth, 2021-12-30:
I would like to add my vote to allow a factory reset in software.
Here is my reason.  I am using a STM32F405 and modified V1.17 to use external serial flash rather than internal.  So far, it works great but do get an occasional corruption.  The units are in a remote location and can be serviced over BLE.  So the idea is to add a main.py file that supports BLE.  This main.py file can be defined in the main.c of the MPy kernel.  So after a factory reset, I can communicate with a phone over BLE and download the Python MPY files.

peterhinch, 2022-01-01:
I use rshell and mpremote. Both work fine with LFS, and on a wide range of platforms. Most platforms don't support MSC mode - mercifully it's easily disabled on Pyboards.