micropython/micropython #4834: nrf52 - undefined reference to 'atan2'， math method.

Afantor, 2019-06-05:
I wand to add a module to driver MPU6050 by nRF52832. and then I add mpu6050.c、mpu6050.h and DMP files(inv_mpu.c) . I need trigonometric function(asin , atan2) to calculate MPU data (pitch,roll,yaw). But when I make the code ,I find some errors :
"/tmp/cc8VvtQw.ltrans4.ltrans.o: In function `mpu_dmp_get_data':
/home/af/micropython/ports/nrf/modules/machine/inv_mpu.c:2930: undefined reference to `asin'
/home/af/micropython/ports/nrf/modules/machine/inv_mpu.c:2931: undefined reference to `atan2'
/home/af/micropython/ports/nrf/modules/machine/inv_mpu.c:2932: undefined reference to `atan2'
collect2: error: ld returned 1 exit status"

the file have been add "#include <math.h>", it is not work.
so, How to support mathematical functions sin , asin , atan2 ?

I need to help.  @dpgeorge

glennrub, 2019-06-05:
@Afantor , Have you tried to use the sinf, asinf, atan2f functions from lib/libm? They should all be included in the Makefile for nrf52 targets. I did a test run just know and i'm able to link it and print the results of the functions.

jonnor, 2024-07-03:
This issue is now 5 years old and should have a resolution as documented above. Proposing to close this issue.

closed 2024-07-05 by projectgus