micropython/micropython #4835: mpy-cross build faild in release v1.11 (Using ESP32 Windows toolchain)

ShenTengTu, 2019-06-06:
I use ESP32 Windows toolchain(MSYS2) build mpy-cross successfully before, the version is:
```
MicroPython v1.10-356-g653e1756c on 2019-05-20; mpy-cross emitting mpy v4
```

But after I update to v1.11 release, it build failed an get error `error: conflicting types for 'time_t'`. The detail as below:
```
$ make
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
mkdir -p build/genhdr
GEN build/genhdr/mpversion.h
GEN build/genhdr/moduledefs.h
GEN build/genhdr/qstr.i.last
GEN build/genhdr/qstr.split
GEN build/genhdr/qstrdefs.collected.h
QSTR updated
GEN build/genhdr/qstrdefs.generated.h
mkdir -p build/ports/windows/
mkdir -p build/py/
CC ../py/mpstate.c
CC ../py/nlr.c
CC ../py/nlrx86.c
CC ../py/nlrx64.c
CC ../py/nlrthumb.c
CC ../py/nlrxtensa.c
CC ../py/nlrsetjmp.c
CC ../py/malloc.c
In file included from D:\Espressif\msys32\usr\include/sys/select.h:15:0,
                 from D:\Espressif\msys32\usr\include/sys/types.h:52,
                 from D:\Espressif\msys32\usr\include/stdio.h:61,
                 from ../py/malloc.c:27:
D:\Espressif\msys32\usr\include/sys/_timeval.h:40:18: error: conflicting types for 'time_t'
 typedef _TIME_T_ time_t;
                  ^~~~~~
In file included from D:/Espressif/msys32/mingw32/i686-w64-mingw32/include/crtdefs.h:10:0,
                 from D:/Espressif/msys32/mingw32/i686-w64-mingw32/include/stddef.h:7,
                 from D:/Espressif/msys32/mingw32/lib/gcc/i686-w64-mingw32/7.4.0/include/stddef.h:1,
                 from D:\Espressif\msys32\usr\include/sys/cdefs.h:47,
                 from D:\Espressif\msys32\usr\include/stdio.h:35,
                 from ../py/malloc.c:27:
D:/Espressif/msys32/mingw32/i686-w64-mingw32/include/corecrt.h:141:20: note: previous declaration of 'time_t' was here
 typedef __time32_t time_t;
                    ^~~~~~
make: *** [../py/mkrules.mk:47: build/py/malloc.o] Error 1
```

In **_timeval.h**:
```h
#if !defined(__time_t_defined) && !defined(_TIME_T_DECLARED)
typedef	_TIME_T_	time_t;
#define	__time_t_defined
#define	_TIME_T_DECLARED
#endif
```

In **corecrt.h**:
```h
#ifndef _TIME_T_DEFINED
#define _TIME_T_DEFINED
#ifdef _USE_32BIT_TIME_T
typedef __time32_t time_t;
#else
typedef __time64_t time_t;
#endif
#endif /* _TIME_T_DEFINED */
```

if I remove `D:\Espressif\msys32\usr\include` from  **C_INCLUDE_PATH** , it still build faild:
```
...
CC ../py/frozenmod.c
CC main.c
CC gccollect.c
CC ../ports/windows/fmode.c
LINK mpy-cross
D:\Espressif\msys32\mingw32\bin\strip.exe: mpy-cross▒G▒L▒▒▒ɮ▒
make: *** [../py/mkrules.mk:130: mpy-cross] Error 1
make: *** Deleting file 'mpy-cross'
```

Because `alloca.h` is missing when building previously, so I add `D:\Espressif\msys32\usr\include` to **C_INCLUDE_PATH** to build successfully.

robert-hh, 2019-06-06:
```
D:\Espressif\msys32\mingw32\bin\strip.exe: mpy-cross▒G▒L▒▒▒ɮ▒
make: *** [../py/mkrules.mk:130: mpy-cross] Error 1
make: *** Deleting file 'mpy-cross'
```
This is an old bug, and it has been fixed more than once, but still creeps up again. The script tries to strip mpy-cross, but that does not exists, since the binary which was created is mpy-cross.exe. And that will still be there. You can strip that manually with:
strip mpy-cross.exe
Besides that, I could build mpy-cross.exe with the 32 bit variant of MSYS2.

stinos, 2019-06-06:
I've recently worked around this with `make STRIP=echo SIZE=echo`

ShenTengTu, 2019-06-06:
Thank @robert-hh & @stinos for reply .

According @robert-hh said, I add `.exe` extension in the script `micropython/py/mkrules.mk` as below:
```mk
ifndef DEBUG
	$(Q)$(STRIP) $(STRIPFLAGS_EXTRA) $(PROG).exe
endif
	$(Q)$(SIZE) $$(find $(BUILD) -path "$(BUILD)/build/frozen*.o") $(PROG).exe
```

I don't know if make a correct effect, but it works.

closed 2019-06-06 by ShenTengTu